MarlinFirmware/Marlin #10975: TMC2130 + MKS Sbase

Studna3D, 2018-06-09:
How can I connect TMC2130 with MKS Sbase and adjust configuration? What pins should I use for SPI? Can anybody help me?

tig33r, 2018-06-09:
I had exactly this configuration. TMC2130 was put in ramps 1.4 and connected by jump wires to en, stp, dir pins on sbase. For spi and cs I used pins from J8 header and set cs pins in pin file and tmc_sw_spi in configuration_adv.

Studna3D, 2018-06-10:
Do I have to add this to pins_MKS_SBASE?

For example:
#define X_STEP_PIN         P2_00
#define X_DIR_PIN          P0_05
#define X_ENABLE_PIN       P0_04
#ifndef X_CS_PIN
  #define X_CS_PIN         P2_11

Because the PlatformIO: build does not work without this configuration.

tig33r, 2018-06-10:
Yes if they are not defined by default (I don't remember).

Studna3D, 2018-06-10:
I tried conect and set configuration. But stepper motor not turn, only "buzzes"

M122:
```
15:33:40.804 : X driver current: 600
15:33:40.806 : Y driver current: 600
15:33:54.659 : X	Y
15:33:54.663 : Enabled		true	true
15:33:54.665 : Set current	600	600
15:33:54.672 : RMS current	581	581
15:33:54.680 : MAX current	819.209961	1479.089966
15:33:54.682 : Run current	18/31	18/31
15:33:54.684 : Hold current	9/31	9/31
15:33:54.690 : CS actual		31/31	31/31
15:33:54.697 : PWM scale	207	135
15:33:54.701 : vsense		1=.18	0=.325
15:33:54.705 : stealthChop	true	true
15:33:54.712 : msteps		0	32
15:33:54.719 : tstep		3758096383	3774627599
15:33:54.719 : pwm
15:33:54.721 : threshold		39	39
15:33:54.728 : [mm/s]		810.897461	0.000000
15:33:54.732 : OT prewarn	true	true
15:33:54.733 : OT prewarn has
15:33:54.736 : been triggered	false	false
15:33:54.742 : off time		14	15
15:33:54.749 : blank time	16	54
15:33:54.750 : hysteresis
15:33:54.755 : -end		12	10
15:33:54.762 : -start		5	8
15:33:54.764 : Stallguard thrs	0	0
15:33:54.764 : DRVSTATUS	X	Y
15:33:54.768 : stallguard			X
15:33:54.775 : sg_result		767	1008
15:33:54.778 : fsactive		X	X
15:33:54.781 : stst
15:33:54.785 : olb		X	X
15:33:54.788 : ola		X	X
15:33:54.791 : s2gb		X	X
15:33:54.794 : s2ga		X	X
15:33:54.797 : otpw		X	X
15:33:54.800 : ot		X	X
15:33:54.810 : Driver registers:	X = 0x       F       8:       7       E:       1       F:       8       7
15:33:54.818 : Y = 0x       F       F:       D       F:       F       F:       F       F
```

tig33r, 2018-06-10:
Put your conf_adv file there.

Studna3D, 2018-06-10:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/2087927/config.zip)

tig33r, 2018-06-10:
Firstly try deactivate stealthchop and raise current a bit (even to 1A maybe, I don't know which motors you have but 0.6 A is really small value). You can also activate monitor driver status to be sure communication between cpu and stepsticks is proper.

Studna3D, 2018-06-10:
I used these stepper motors. 
[(https://www.aliexpress.com/item/Free-shipping-3pcs-42HM48-1684-0-9-degree-42mm-2phase-hybrid-stepper-motor-NEMA17-bipolar-step/32549678589.html?spm=a2g0s.9042311.0.0.27424c4dIDBlez)]

tig33r, 2018-06-10:
Set 1000 or 1200 mA, turn on monitor driver status and check if they will move.

Studna3D, 2018-06-10:
When I set #define MONITOR_DRIVER_STATUS and conected printer, after a while disconnect.

tig33r, 2018-06-10:
Because you dont have connection between tmc2130 and mks sbase. Check if you put your cables correctly.

Studna3D, 2018-06-10:
My pins configuration:
```
/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 * Copyright (C) 2017 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * MKS SBASE pin assignments
 */

#ifndef TARGET_LPC1768
  #error "Oops!  Make sure you have the LPC1768 environment selected in your IDE."
#endif

#ifndef BOARD_NAME
  #define BOARD_NAME "MKS SBASE"
  #define DEFAULT_WEBSITE_URL "https://github.com/makerbase-mks/MKS-SBASE"
#endif

// unused
/*
#define PIN_P0_27          P0_27
#define PIN_P0_28          P0_28
*/

//
// Servo pin
//
#define SERVO0_PIN         P2_13   // J8-3 (low jitter)
#define SERVO1_PIN         P2_12   // J8-4
#define SERVO2_PIN         P2_11   // J8-5
#define SERVO3_PIN         P4_28   // J8-6

//
// Limit Switches
//
#define X_MIN_PIN          P1_24   // 10k pullup to 3.3V, 1K series
#define X_MAX_PIN          P1_25   // 10k pullup to 3.3V, 1K series
#define Y_MIN_PIN          P1_26   // 10k pullup to 3.3V, 1K series
#define Y_MAX_PIN          P1_27   // 10k pullup to 3.3V, 1K series
#define Z_MIN_PIN          P3_25   // The original Mks Sbase DIO19 has a 10k pullup to 3.3V or 5V, 1K series, so when using a Zprobe we must use DIO41 (J8 P1.22)
#define Z_MAX_PIN          P1_29   // 10k pullup to 3.3V, 1K series

//
// Steppers
//
#define X_STEP_PIN         P2_00
#define X_DIR_PIN          P0_05
#define X_ENABLE_PIN       P0_04
#ifndef X_CS_PIN
  #define X_CS_PIN         P4_28
[... 145 lines cut ...]
 *
 *  PWM1.1   P1_18   SERVO3_PIN       FIL_RUNOUT_PIN   5V output, PWM
 *  PWM1.1   P2_00   E0_STEP_PIN
 *  PWM1.2   P1_20   SERVO0_PIN
 *  PWM1.2   P2_01   X_STEP_PIN
 *  PWM1.3   P1_21   SERVO1_PIN       J5-1
 *  PWM1.3   P2_02   Y_STEP_PIN
 *  PWM1.4   P1_23   SDSS(SSEL0)      J3-5  AUX-3
 *  PWM1.4   P2_03   Z_STEP_PIN
 *  PWM1.5   P1_24   X_MIN_PIN        10K PULLUP TO 3.3v, 1K SERIES
 *  PWM1.5   P2_04   RAMPS_D9_PIN
 *  PWM1.6   P1_26   Y_MIN_PIN        10K PULLUP TO 3.3v, 1K SERIES
 *  PWM1.6   P2_05   RAMPS_D10_PIN
 */

 /**
  * Special pins
  *   P1_30 - not 5V tolerant
  *   P1_31 - not 5V tolerant
  *   P0_27 - open collector
  *   P0_28 - open collector
  *
  */

```

Studna3D, 2018-06-10:
When I conect printer:
```
16:49:33.335 : No start signal detected - forcing start
16:49:33.387 : FIRMWARE_NAME:Marlin bugfix-2.0.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Studna 3D Pr:2 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
16:49:33.387 : Cap:SERIAL_XON_XOFF:0
16:49:33.388 : Cap:EEPROM:0
16:49:33.388 : Cap:VOLUMETRIC:1
16:49:33.388 : Cap:AUTOREPORT_TEMP:1
16:49:33.388 : Cap:PROGRESS:0
16:49:33.388 : Cap:PRINT_JOB:1
16:49:33.388 : Cap:AUTOLEVEL:1
16:49:33.389 : Cap:Z_PROBE:1
16:49:33.389 : Cap:LEVELING_DATA:1
16:49:33.389 : Cap:BUILD_PERCENT:0
16:49:33.389 : Cap:SOFTWARE_POWER:1
16:49:33.389 : Cap:TOGGLE_LIGHTS:0
16:49:33.390 : Cap:CASE_LIGHT_BRIGHTNESS:0
16:49:33.390 : Cap:EMERGENCY_PARSER:0
16:49:33.390 : Cap:AUTOREPORT_SD_STATUS:0
16:49:33.390 : Cap:THERMAL_PROTECTION:0
16:49:33.411 : X:0.000000 Y:0.000000 Z:0.000000 E:0.000000 Count A:0 B:0 Z:0
16:49:33.411 : echo:DEBUG:INFO,ERRORS
16:49:33.411 : echo:Active Extruder: 0
16:49:33.412 : echo:Unknown command: "M20"
16:49:33.584 : echo:DEBUG:INFO,ERRORS
16:49:33.586 : echo:Active Extruder: 0
16:49:35.389 : X driver error detected:
16:49:35.390 : overtemperature
16:49:35.393 : short to ground (coil A)
16:49:35.397 : short to ground (coil B)
16:49:35.397 : X	Y
16:49:35.399 : Enabled		false	false
16:49:35.401 : Set current	1000	1000
16:49:35.408 : RMS current	994	550
16:49:35.416 : MAX current	1401.539917	775.500000
16:49:35.418 : Run current	17/31	17/31
16:49:35.420 : Hold current	8/31	8/31
16:49:35.427 : CS actual		31/31	15/31
16:49:35.433 : PWM scale	253	15
16:49:35.437 : vsense		1=.18	0=.325
16:49:35.442 : stealthChop	true	true
16:49:35.448 : msteps		0	0
16:49:35.455 : tstep		4227792863	520602104
16:49:35.456 : pwm
16:49:35.457 : threshold		0	0
16:49:35.458 : [mm/s]		-	-
16:49:35.462 : OT prewarn	false	false
16:49:35.464 : OT prewarn has
16:49:35.466 : been triggered	false	false
16:49:35.472 : off time		13	8
16:49:35.479 : blank time	54	54
16:49:35.480 : hysteresis
16:49:35.486 : -end		11	12
16:49:35.492 : -start		4	8
16:49:35.493 : Stallguard thrs	0	0
16:49:35.494 : DRVSTATUS	X	Y
16:49:35.498 : stallguard		X
16:49:35.505 : sg_result		1007	126
16:49:35.508 : fsactive		X	X
16:49:35.512 : stst			X
16:49:35.515 : olb		X	X
16:49:35.518 : ola		X	X
16:49:35.521 : s2gb		X	X
16:49:35.524 : s2ga		X	X
16:49:35.528 : otpw
16:49:35.530 : ot		X
16:49:35.540 : Driver registers:	X = 0x       1       F:       C       7:       F       1:       F       E
16:49:35.548 : Y = 0x       B       F:       E       F:       F       C:       F       F
16:49:35.555 : Error:Printer halted. kill() called!
```

Studna3D, 2018-06-10:
I had bad driver TCM2130. I must delete rezistors on a driver. Now its working. How many microsteps MKS Sbase handel it?

tig33r, 2018-06-10:
1/256 I guess?

Studna3D, 2018-06-10:
When I set 256microsteps, it is very slow to commands.

tig33r, 2018-06-10:
You have to adjust steps per mm value in conf file too.

thinkyhead, 2018-06-11:
TMC2130's are generally meant to be used with 16 micro-steps, while internally interpolating these steps to 256. See this video (RAMPS, but applies also to MKS SBASE) for information about setting the stepper driver jumpers on the board: https://youtu.be/yiHLwdAIApQ?t=2m9s

tig33r, 2018-06-11:
> I wasn't able to reply to it. If it hasn't been addressed then please open a new issue.

@Bob-the-Kuhn already created mew issue and somebody for 99% solve it. I cant check it complete because I have to temporary move my tmc2130 to another printer.

END OT

boelle, 2019-02-20:
@Studna3D is this still an issue?

closed 2019-02-20 by thinkyhead