MarlinFirmware/Marlin #10976: [bugfix-2.0.x] Z bump divisor ineffective

VanessaE, 2018-06-09:
### Description

During homing, the Z move "bumps" when my sensor triggers, like normal, but the speed of the second/bump move is definitely the same as the normal Z homing speed (600 mm/min)

Z speeds during `G29` behave in a similar manner.  My divisors:

```
#define HOMING_BUMP_DIVISOR { 4, 4, 8 }  // Re-Bump Speed Divisor (Divides the Homing Feedrate)`
```

(These are my normal settings, imported by hand from `bugfix-1.1.x`)

Steps to Reproduce?  Beats me.  Load my configuration and try it? :smiley:

#### Additional Information

[Marlin-configs-20180609.zip](https://github.com/MarlinFirmware/Marlin/files/2087251/Marlin-configs-20180609.zip)

VanessaE, 2018-06-10:
I'm putting this here to avoid cluttering the issue list, as I am not sure if these are general 2.0.x bugs:

Since I'm on bugfix 2.0.x now (I figured it was long past time), I decided to give #10922 a try, and it seems to work well - as with others, my machine seems to be smoother and a little less noisy now for the most part (especially on fast travels).  Final quality is about the same for the particular print settings I'm using at the moment (attached below).  Surprisingly, my motors are definitely running a little cooler too (by 5-10° says my thermocouple).

I've run into two oddities:

First, homing is a bit... "harsh".  That is to say, acceleration seems to not be in use, at least not to the same degree.  It's especially evident when an endstop is triggered, the axis stops cold, pauses momentarily, then begins the reverse move for the "bump".  This happens on all axes (cartesian), and affects probing during `G29` also (all Z moves therein).

Second, I now hear a whine from the extruder motor during certain kinds of print moves - about 8 kHz pitch if I had to guess, with harmonics (you can feel it when touching the motor too).  The area highlighted in blue in this screenshot (also the tiny dark green area to the left of it, middle of the slope):

[image]

The object is the excavator from https://www.thingiverse.com/thing:2719434
Sliced with Cura 3.3.1 (profile: [30mms-0.08mm.curaprofile.zip](https://github.com/MarlinFirmware/Marlin/files/2087892/30mms-0.08mm.curaprofile.zip)).
Marlin configs:  [Marlin-configs-20180610.zip](https://github.com/MarlinFirmware/Marlin/files/2087896/Marlin-configs-20180610.zip)

(fair warning:  that part is a right pain in the ass to get a perfect print, which I have yet to do.  Do not attempt unless you have cartoon-grade hair restorer on hand. :grin:)

ejtagle, 2018-06-10:
Maybe your maximum feedrate is above the maximum achievable... In such case you will think divisor is not working, but, in fact, what is actually happening is that the speed is limited by hardware... 
Some improvements have been done in #10922, if you want to try them...

VanessaE, 2018-06-10:
I highly doubt I'm hitting a speed limit, as my Z axis can run somewhat reliably at around 3000 mm/min (in theory, it can do twice that) with gentle acceleration.  However, I home at 900, and do routine moves at 1800 during a print.

VanessaE, 2018-06-10:
I note that #10922 just got merged.  *updates*

I've set `MAXIMUM_STEPPER_RATE` to 500000 to suit my A4988 driver modules.

The bump divisor seems to be working.  Maybe I misunderstood what I was seeing there.

However, the not-very-accelerated motion during homing and `G29` is unchanged.

ejtagle, 2018-06-10:
Well, i do not completely understand... When an endstop is hit, there will be a sudden and inmediate stop. No deacceleration, just a complete stop. This is expected, as the only way to deaccelerate is to have some space, and when the switch triggers, it is assumed you have reached the physical limits of your machine and any extra movement in that same direction will cause damage to it...

VanessaE, 2018-06-10:
You're right, of course, but I'm talking about (the beginning and end of) the reverse move after the end stop has been hit, and the (beginning of the) bump move that follows it (or the few that follow the initial sensor trigger for each `G29` probe move).

ejtagle, 2018-06-10:
Basically, the begin and reverse moves should be respecting the maximum acceleration configured in your configuration.h file... Maybe it is just set too high ? (I would not worry about it if the motors are not losing steps, there is no need to lower it! )

VanessaE, 2018-06-10:
The *maximum* acceleration?  Should they not respect the current normal travel acceleration instead?

If I had to graph it, it would look something like this:

[image]

Time on the horizontal, speed on the vertical.  The drop at "1" is the initial hit on the end stop.  Reverse direction and accelerate to bump speed starting at "2", start back toward the end stop at "3", hit the end stop again at "4".  Green is what I'm used to seeing, red is what it's doing now.

ejtagle, 2018-06-10:
Seems it is accelerating way faster than before... Truth to be said, i never bothered investigating that... 
Now there are a ton of changes in the movement planner and stepper code of Marlin, so i have to figure out if the new behaviour is the correct, and the previous an incorrect one, or the opposite... :S

VanessaE, 2018-06-10:
I won't worry about it as long as the speed it's moving when it hits the end stop is correct - it's just something that triggered my "this is way different, approach with caution" sense. :smile:

ejtagle, 2018-06-10:
So many changes lately... I am sure there will be things that will change. The most important thing is to note is something stops working, or stops working reliably...

VanessaE, 2018-06-10:
Right.  That's why it stood out.

thinkyhead, 2018-06-11:
> …I am sure there will be things that will change…

This issue with the bump acceleration is certainly not limited to just the one place. Marlin has always been very brittle and inter-dependent, so various things are bound to break in hard-to-understand ways whenever code stabilized over months and years (by whatever means) is supplanted by fresh untested code. The bump acceleration issue is just a clue to a more fundamental flaw.

ejtagle, 2018-06-25:
Pleaso, try PR #11098 and see if it fixes things for you...

thinkyhead, 2018-06-30:
Now would be a good time to re-test the latest bugfix branches. Motion and endstops issues have been addressed, tested, and re-tested. This is one of the last issues we need to verify before we have full confidence.

VanessaE, 2018-07-03:
Still the same behavior at e0ab8acf198648f731074f9b14f3562b55c5dd5a

VanessaE, 2018-07-03:
Still the same behavior at e0ab8acf198648f731074f9b14f3562b55c5dd5a (referring to the way it accelerates, since the speed seems about right).  It still works, though.

ejtagle, 2018-07-03:
It shouldn´t. To be honest, i have seen what @VanessaE refers. But, i did dump the movement blocks and there is no sudden acceleration... So it is a big mistery the cause...

closed 2018-11-05 by thinkyhead