MarlinFirmware/Marlin #10985: Where is TEMP_CHAMBER_PIN?

JXPA, 2018-06-10:
Hello.
TEMP_SENSOR_CHAMBER is in the Thermal Settings section.
However, the Pins.h has no setting like TEMP_CHAMBER_PIN.
It only has TEMP_ 0 - 4_PIN and TEMP_BED_PIN.
Where can I set the pin for the chamber sensor?

thinkyhead, 2018-06-10:
You can define it in your `Configuration.h` file.

closed 2018-06-10 by thinkyhead

JXPA, 2018-06-10:
@thinkyhead In Configuration.h file is TEMP_SENSOR_CHAMBER.
This is to specify type of temperature sensor to use.
There is no way to define which pin the temperature sensor is connected to.
Similarly, the Configuration_adv.h file can define the pin of cooling fan, but there is no pin definition of temperature sensor.

thinkyhead, 2018-06-10:
> There is no way to define which pin the temperature sensor is connected to.

Just add a line like this to `Configuration.h`, specifying the pin you've connected it to:
```cpp
#define TEMP_CHAMBER_PIN 12
```

JXPA, 2018-06-11:
Thank you for teaching me.
I tried and I got a compile error.
This is the result of using the latest version of Marlin 2.0.x with Arduino Due.

Test1:
Delete `#define TEMP_2_PIN 3` from pins_RAMPS_FD_V1.h.
Add pin settings to the Configuration.h.
`#define TEMP_CHAMBER_PIN 3`
```
C:\Temp\arduino_build_170220\sketch\src\gcode\control\M42.cpp: In static member function 'static void GcodeSuite::M42()':

C:\Temp\arduino_build_170220\sketch\src\gcode\control\M42.cpp:45:24: error: 'pin_number' was not declared in this scope

   if (pin_is_protected(pin_number)) return protected_pin_err();

                        ^
```

Test2:
Restore pins_RAMPS_FD_V1.h.
Add pin settings to the Configuration.h.
`#define TEMP_CHAMBER_PIN TEMP_2_PIN`
```
C:\Temp\arduino_build_170220\sketch\src\module\temperature.cpp: In static member function 'static void Temperature::isr()':

C:\Temp\arduino_build_170220\sketch\src\module\temperature.cpp:2071:32: error: invalid conversion from 'Adc*' to 'uint16_t {aka short unsigned int}' [-fpermissive]

         raw_temp_chamber_value += ADC;

                                ^
```
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2088395/Marlin.zip)

thinkyhead, 2018-06-11:
Grab the newest `bugfix-2.0.x` code which has those errors patched.

JXPA, 2018-06-11:
I compiled successfully with the latest version.
Thanks a lot.