MarlinFirmware/Marlin #10988: 2nd Filament Runoutsensor resuming print at wrong area.

plloppii, 2018-06-10:
Hi all,

I have been diving into Marlin and trying to figure out a few bugs for my machine. However I have been stuck on this issue for quite a while and I am looking to see if anyone can guide me. 
Currently, I have two filament sensors and a dual extruder. The left extruder filament runout works perfectly fine. 
When I trip the second filament sensor, the nozzle parks, changes filament, homes, then resumes the print about 55mm (which is my hotend offset) to the left. I have been targeting this section of code in Marlin_main.cpp. 

```
do_blocking_move_to_xy(resume_position[X_AXIS], resume_position[Y_AXIS], NOZZLE_PARK_XY_FEEDRATE);
	
    // Set Z_AXIS to saved position
    do_blocking_move_to_z(resume_position[Z_AXIS], NOZZLE_PARK_Z_FEEDRATE);

    // Now all extrusion positions are resumed and ready to be confirmed
    // Set extruder to saved position
    planner.set_e_position_mm((destination[E_AXIS] = current_position[E_AXIS] = resume_position[E_AXIS]));
	
	
    #if ENABLED(FILAMENT_RUNOUT_SENSOR)
      runout.reset();
    #endif
```

I tried to add the following: 
	if(active_extruder == 1){
		resume_position[X_AXIS]+= hotend_offset[X_AXIS][active_extruder];
		resume_position[Y_AXIS]+= hotend_offset[Y_AXIS][active_extruder];
	}

However, when I do so, the print lowers to the original printing location, then moves over 55mm and starts printing. 

Any input would be greatly appreciated!

plloppii, 2018-06-11:
So the bug resides in the tool_change function in Marlin_main.

I found that if you do the following gcode: 
T0
G28
T1

The tool change function tries to move the Tool1 over by the offset, hitting the X limit switch. 
I am editing that snippet of code so that it only moves if the X endstop is not hit. 

How can I do this? I have not yet found a flag that says the X endstop is triggered.
I tried: 
```
if( !(TEST(Endstops::endstop_hit_bits, X_MIN)) ){
     do_blocking_move_to(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS]);
}
```

And that doesnt seem like its giving me the correct response. 

Any input?

thinkyhead, 2018-06-11:
Try `T1 S1` to do the tool change without moving the carriage.

boelle, 2019-02-20:
@plloppii problem solved?

closed 2019-02-20 by thinkyhead

robek2525, 2020-02-01:
I have a similar problem how to solve it