MarlinFirmware/Marlin #10993: TMC2130 Active in Latest 1.1.x, M122 reporting disabled; moves cause Marlin reboot without error.

XBrav, 2018-06-11:
Downloaded 1.1.x a few hours ago, and translated my parameters that work on a previous 1.1.x onto this version. After configuring as expected, the printer boots successfully. However, on the first move, the RAMPs 1.4 board reboots.

M122 reports the following:

Send: M122
Recv: 		X	Y	Z	E0
Recv: Enabled		false	false	false	false
Recv: Set current	800	800	800	800
Recv: RMS current	795	795	795	795
Recv: MAX current	1121	1121	1121	1121
Recv: Run current	25/31	25/31	25/31	25/31
Recv: Hold current	12/31	12/31	12/31	12/31
Recv: CS actual		12/31	12/31	12/31	12/31
Recv: PWM scale	46	12	10	10
Recv: vsense		1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true
Recv: msteps		16	16	16	16
Recv: tstep		1048575	1048575	1048575	1048575
Recv: pwm
Recv: threshold		0	0	0	0
Recv: [mm/s]		-	-	-	-
Recv: OT prewarn	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false
Recv: off time		5	5	5	5
Recv: blank time	24	24	24	24
Recv: hysteresis
Recv: -end		2	2	2	2
Recv: -start		3	3	3	3
Recv: Stallguard thrs	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	E0
Recv: stallguard
Recv: sg_result		0	0	0	0
Recv: fsactive
Recv: stst		X	X	X	X
Recv: olb		X
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:	X = 0xC0:0C:00:00
Recv: 	Y = 0x80:0C:00:00
Recv: 	Z = 0x80:0C:00:00
Recv: 	E0 = 0x80:0C:00:00
Recv: 
Recv: 
Recv: ok

Notably, all axis are reporting as not enabled. As well, the Driver registers vary from my previous Marlin, which works fine:

Send: M122
Recv: 		X	Y	Z	E0
Recv: Enabled		true	true	true	false
Recv: Set current	700	700	800	800
Recv: RMS current	673	673	795	795
Recv: MAX current	949	949	1121	1121
Recv: Run current	21/31	21/31	25/31	25/31
Recv: Hold current	10/31	10/31	12/31	12/31
Recv: CS actual		10/31	10/31	12/31	12/31
Recv: PWM scale	40	36	35	13
Recv: vsense		1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true
Recv: msteps		16	16	16	16
Recv: tstep		1048575	1048575	1048575	1048575
Recv: pwm
Recv: threshold		85	85	329	8150
Recv: [mm/s]		100.02	100.02	3.00	1.00
Recv: OT prewarn	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false
Recv: off time		5	5	5	5
Recv: blank time	24	24	24	24
Recv: hysteresis
Recv: -end		2	2	2	2
Recv: -start		3	3	3	3
Recv: Stallguard thrs	8	8	0	0
Recv: DRVSTATUS	X	Y	Z	E0
Recv: stallguard		X	X	X
Recv: sg_result		0	0	3	0
Recv: fsactive
Recv: stst		X	X	X	X
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:	X = 0x81:0A:00:00
Recv: 	Y = 0x81:0A:00:00
Recv: 	Z = 0x81:0C:00:03
Recv: 	E0 = 0x80:0C:00:00
Recv: 
Recv: 
Recv: ok

Sensorless homing is only enabled for X and Y. Wiring is not changed between the revisions. If I flash the previous firmware, everything works as expected.

[CFG_06102018.zip](https://github.com/MarlinFirmware/Marlin/files/2088418/CFG_06102018.zip)

thinkyhead, 2018-06-11:
> Downloaded 1.1.x a few hours ago

`bugfix-1.1.x` ?

Does it help to do `M502`, `M500`, followed by a cold boot?

XBrav, 2018-06-11:
Correct, bugfix-1.1.x.

M502 loads the hardcoded defaults, but I currently have EEPROM disabled, so 502 throws Error:EEPROM disabled.

With the printer cold booted, here's the output including a simple 10mm X jog after booting:
<details><summary>Boot and Jog</summary>
`Changing monitoring state from "Operational" to "Offline"
Connecting to: /dev/ttyACM0
Connection closed, closing down monitor
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x6a4f03b0, open=True>(port='/dev/ttyACM0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: start
Recv: echo:Marlin bugfix-1.1.x
Recv: 
Send: N0 M110 N0*125
Recv: echo: Last Updated: 2018-01-20 | Author: (none, default config)
Recv: echo:Compiled: Jun 10 2018
Recv: echo: Free Memory: 5047  PlannerBufferBytes: 1232
Recv: echo:Hardcoded Default Settings Loaded
Recv: echo:  G21    ; (mm)
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X93.00 Y93.00 Z800.00 E97.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X300.00 Y300.00 Z5.00 E25.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X3000 Y3000 Z100 E10000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P3000.00 R3000.00 T3000.00
Recv: echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 B20000 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:PID settings:
Recv: echo:  M301 P22.20 I1.08 D114.00
Recv: echo:Stepper driver current:
Recv: echo:  M906 X800 Y800 Z800
Recv:   M906 T0 E800
Recv: 
Recv: ok
Changing monitoring state from "Connecting" to "Operational"
Send: N0 M110 N0*125
Recv: echo:Unknown command: ""
Recv: ok
Send: N1 M115*39
Recv: FIRMWARE_NAME:Marlin bugfix-1.1.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:EEPROM:0
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:0
Recv: Cap:Z_PROBE:0
Recv: Cap:LEVELING_DATA:0
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:THERMAL_PROTECTION:1
Recv: ok
Send: M21
Recv: echo:Unknown command: "M21"
Recv: ok
Send: M155 S5
Recv: ok
Send: G91
Recv: ok
Send: G1 X10 F6000
Recv: ok
Send: G90
Recv: ok
Recv: start
Printer sent 'start' while already operational. External reset? Resetting line numbers to be on the safe side
Recv: echo:Marlin bugfix-1.1.x
Recv: 
Recv: echo: Last Updated: 2018-01-20 | Author: (none, default config)
Send: N0 M110 N0*125
Recv: echo:Compiled: Jun 10 2018
Recv: echo: Free Memory: 5047  PlannerBufferBytes: 1232
Recv: echo:Hardcoded Default Settings Loaded
Recv: echo:  G21    ; (mm)
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X93.00 Y93.00 Z800.00 E97.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X300.00 Y300.00 Z5.00 E25.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X3000 Y3000 Z100 E10000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P3000.00 R3000.00 T3000.00
Recv: echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 B20000 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:PID settings:
Recv: echo:  M301 P22.20 I1.08 D114.00
Recv: echo:Stepper driver current:
Recv: echo:  M906 X800 Y800 Z800
Recv:   M906 T0 E800
Recv: 
Recv: ok
Send: M155 S5
Recv: ok
Recv:  T:20.25 /0.00 B:18.50 /0.00 @:0 B@:0`
</details>

thinkyhead, 2018-06-11:
A reboot is a very unusual thing. When this happens during a move it usually means the machine is trying to do too many steps and the watchdog causes a reboot, but all your motion parameters seem reasonable. So it's mysterious.

XBrav, 2018-06-11:
Exactly. I plan on doing a full comparison between the two releases to see if any odd parameters have changed. The X jog will move a very minor amount, and then it looks like the runtime watchdog fails, as I receive zero response on any command after the confirmation of the G1.

As the motor does slightly move, it makes me think something has changed in the TMC2130 implementation. I recompiled both versions using the latest 2.3.0 library in Arduino 1.8.5, and the old one continues to work, whereas the new one fails as shown above.

The functioning version is dated 4/24/2018 on my local machine for reference, and is also 1.1.x bugfix branch.

As noted above, the biggest red flag I can see is M122 reporting them as Enabled: false, even with the code enabled.

thinkyhead, 2018-06-11:
We'll have to do a comparison of the April 24 version compared to current. Are you able to test an interim version —such as from mid-May— to see if the issue appeared before or after that date?

XBrav, 2018-06-11:
I'll grab an interim version and see if it persists. Unfortunately, it seems the TMC2130 is always changing.

I've currently checked the difference on tmc_util.cpp/h, and they only seem to have some verbage and definition modifications.

Will report back on if it works from May 15th or so.

JXPA, 2018-06-11:
A similar problem occurred when I tried the latest version.
On my printer Marlin on 6/1/2018 was working.
However, when flashing the latest version TMC 2130 of E 0 no longer rotates.
After that, it rolled back and it worked well.
I think there is a problem after June.

teemuatlut, 2018-06-11:
> As noted above, the biggest red flag I can see is M122 reporting them as Enabled: false, even with the code enabled.

That's a check for whether or not EN pin is active and if the driver has `off_time` configured at that moment.
This is **not** a check for your configuration file. If you do not have an axis with a TMC driver, it simply doesn't appear as a column on the `M122` table.

forkoz, 2018-06-11:
I just loaded the new 1.1 and I don't have TMC2130. It moves a step then marlin reboots. I guess perfect time to try 2.0

DarkCaster, 2018-06-11:
Same problem here when trying the latest bugfix-1.1.x branch and A4988 drivers. No problems with earlier bugfix-1.1.x branch snapshot (1-st june)

thinkyhead, 2018-06-12:
Sounds like probably a typo fell into 1.1.x while trying to merge recent changes from 2.0.x. I'll go through and make sure everything aligns in `stepper.h` and `stepper.cpp`.

thinkyhead, 2018-06-12:
I can't see any differences between 1.1.x and 2.0.x in the core motion, endstop, or temperature code, so it's still mysterious. Can you test `bugfix-1.1.x` from June 8 to see if it still reboots on movement? That will tell us if the issue is due to the stepper and/or serial changes introduced in the last couple of days.

ejtagle, 2018-06-12:
@thinkyhead : Check that the stepper pulse width is using the stepper timer, not the temperature timer... ;)

thinkyhead, 2018-06-12:
Thanks for pointing that out. I was supposed to have done that already…
So I've added that patch to #11002

DarkCaster, 2018-06-12:
@ejtagle, @thinkyhead: Steppers now working as intended. Thanks!

closed 2018-06-13 by thinkyhead