MarlinFirmware/Marlin #10994: Teensy 3.6

boelle, 2018-06-11:
@teemuatlut 

hi i saw that you have started on a pin file for teensy 3.5/3.6

i wonder if you have ever printed with it and if it's an "ok" board for marlin?

i used an issue since i did not know any other way to get in contact. will close this one as soon its posted to it does not take up space on the list

closed 2018-06-11 by boelle

teemuatlut, 2018-06-11:
Never printed. It was just a rough idea where everything could go.

boelle, 2018-06-11:
oki, not sure if its the correct question to ask, but how would you rate the teensy to say a Due?

if one if better than the other etc.

I thought that is better as its runs faster and if i remember correct it also have a FPU

teemuatlut, 2018-06-11:
I'd say the Teensy is better in almost every way.
It's faster, it has an FPU and better support from the developer.
The downside is that there are no commercial boards running either Teensies or even the chip itself. Teensy also has fewer available pins but depending on the implementation that may not be a limitation.

https://github.com/MarlinFirmware/Marlin/issues/7076#issue-236576233

boelle, 2018-06-11:
aha :-) 

so it would not be a bad thing if we got a shield for it, 

what i noted is that the pin map you made in top of the pin file did not have all the internal pins, but i guess that is due to that you only made a rough start?

will thinker about the shield idea, i guess a shield that has same features as Ramps FD 2.2 (yes there is a non released version with this number) 

i exclude RADDS here as its not open source in any way etc.

boelle, 2018-06-11:
i think with some time i could mod the schematic for ramps FD in kicad, but doing tracks and placing components i would need help with.

but then again would it be worth the trouble?

boelle, 2018-06-11:
tried to play arround with the thought of using Ramps FD 2.2 as a base for a Teensy shield, the kicad schematics is made of of several schematics and i had a go for the power part and came up with this

[image]

thinkyhead, 2018-06-11:
A few months ago Paul Stoffregen expressed an interest in making a 3D printer board based around it, but we haven't received any updates on that.

boelle, 2018-06-11:
my problem is that i dont know kicad well and not 100% in to the schematic of Ramps FD

i got hold of V2.2 which is many version past the unsafe ones and its 3.3V level. there are just a few things that i'm unsure off and then i suck at routing tracks

but i figured that it would be a good starting point. My plan was to make it a shield just like for arduino due but for a teensy instead and then get 3 or 10 boards made and put components on them

boelle, 2018-06-12:
@teemuatlut & @thinkyhead do we know any that are good at electronics design in kicad that might be able to adjust an Ramps like board to fit with a Teensy instead? i'm going to finish my hypercube next month and will need electronics for it. I would have gone with an Due and Ramps FD 2.2 but if i could get the Teensy instead i think i would be better off.

as an alternative i could go just hardwiring the teensy to some FET modules and Stepper expanders instead. But i would rather have a real pcb than that

boelle, 2018-06-12:
dont bother..... it does not have enough pins

lets put this no to rest, no need to reply

boelle, 2018-06-17:
trying again 

@teemuatlut in the teensy pins file heater 0 is connected to pin 30... but so is E0_DIR_PIN, was that a typo?

teemuatlut, 2018-06-17:
It's an unfinished plan that has never had any relation to actual hardware.

boelle, 2018-06-17:
oki... will try and connect as many i can with the plan you did and see where i end up

i have been told that the teensy do not have enough pins but i want to see that for myself before i belive that