MarlinFirmware/Marlin #10995: When compiling i get crazy message

madsrinker, 2018-06-11:
https://pastebin.com/Cm8UArBP

In the end it says its done and all that normal stuff! I copied about 1/5 of all the messages i get! 
is this safe to upload like this? Because when i compile skynet all i get is the last 2 lines of that pastebin and everything is good! 

I dont understand why it gives me such a CRAZY big compile message! But it says compile sucsessfull in the end! Will i brick my anet card when uploading this?

I put in the example cfg for anet a8 and changed auto_bed_leveling_bilnear and fix_probe! Thats all i changed

madsrinker, 2018-06-11:
Skissen bruker 116066 bytes (91%) av program lagrings minnet. Maks er 126976 bytes
Globale variabler bruker 4321 bytes av dynamisk minne.

this is the last message so i guess its possible to upload but i dont want to brick it :)

madsrinker, 2018-06-11:
please help i really need to get this working :(

thinkyhead, 2018-06-11:
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as requested in the Issue template) and drop them into your next reply so that we can try compiling with them.

madsrinker, 2018-06-12:
here you go

madsrinker, 2018-06-12:
[example_configurations.zip](https://github.com/MarlinFirmware/Marlin/files/2092976/example_configurations.zip)

madsrinker, 2018-06-12:
Here is all changes i made:

- downloaded the 1.1 version from official website
- copied the anet a8 example configurations from example folder
- uncomment //$auto_bed_leveling_bilnear
- uncomment //$fix_mounted_probe

This is ALL the changes i made! Everything else is 100% stock

and it says its ok to upload but as shown in the pastebin i get a crazy long message that says stuff is changed! I dont know if this is normal since its my first time

madsrinker, 2018-06-12:
ALL of the warnings is about #define TIMER2  5 ETC

--------------------
In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:257:0,

                 from sketch\MarlinConfig.h:38,

                 from sketch\G26_Mesh_Validation_Tool.cpp:27:

C:\Program Files (x86)\Arduino\hardware\anet\avr\variants\sanguino/pins_arduino.h:43:0: warning: "TIMER2" redefined

 #define TIMER2  5

 ^
---------------

It's like a 1000 different messages and they all say the same! But its Timer2, Timer3c ETC ETC

just warning about timer[something] variables

thinkyhead, 2018-06-12:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue.

madsrinker, 2018-06-13:
the same thing happens! There is no difference in any of them!

but did you open the pastebin link i added in the main post?? all i need to know is if i can send it to the anet board with those warning messages....

thinkyhead, 2018-06-13:
> did you open the pastebin link i added in the main post?

Yes I did.

> if i can send it to the anet board with those warning messages

I doubt it. No warnings would be better.

tonypnode, 2018-12-26:
Bump
I'm getting the same

tonypnode, 2018-12-26:
@madsrinker any luck with this?

boelle, 2019-02-20:
@madsrinker still an issue?

carlosgalvezp, 2019-03-04:
@boelle I do have the same issue.
I am trying to update to Marlin on my new Anet A8. I have the latest of everything as of today:
* Arduino 1.8.8
* Marlin 1.1.9
* SkyNet / anet-board commit 51abbca02a754c89f5f5823fb5d7552ea547b7f5

When building Marlin, I get the same compilation warnings as OP. This is because Arduino automatically #includes Arduino.h when building Marlin.ino, which contains pin definitions that are re-defined in the "anet" hardware that we get from the Skynet repository.

This looks like a very dangerous warning, since you are redefining the pins for the TIMERs. Now, which definitions are correct - the ones from Arduino or the ones from SkyNet?

boelle, 2019-03-04:
use arduino ide 1.9
try marlin bugfix 2.0

skynet???

carlosgalvezp, 2019-03-04:
You do need a board definition to be able to compile and upload for Anet A8 in Arduino, don't you? Everybody mentions this in the tutorials:

https://github.com/SkyNet3D/anet-board/tree/master/hardware/anet/avr

Where can I get Arduino 1.9? I can only see 1.8.8

Thanks!

boelle, 2019-03-04:
arduino ide look under BETA BUILDS

and then again i dont think skynet has anything to do with marlin so we cant support that

boelle, 2019-03-04:
example config for A8: https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/config/examples/Anet/A8

carlosgalvezp, 2019-03-05:
Marlin source code provides the following instructions for Anet printers:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/pins_ANET_10.h#L36

```
 * The standard Arduino IDE extension (board manager) for this board
 * is located at https://github.com/SkyNet3D/anet-board.
 *
 * Installation instructions are on that page.
```

So yes, Marlin *does* refer to SkyNet, there should be no surprise about this.

On the other hand, I understand that the problem comes from SkyNet and I'll open an issue in their repo. Will keep this thread updated!

closed 2019-03-05 by thinkyhead

cdelrey, 2019-03-13:
I'm having the same issue across all the suggested versions of Arduino and Marlin made above. I am constantly getting these TIMER messages;

Arduino: 1.9.0-beta (Linux), Board: "Anet V1.0"

In file included from /home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
                 from sketch/src/gcode/feature/camera/../../../inc/../HAL/HAL_AVR/HAL.h:27,
                 from sketch/src/gcode/feature/camera/../../../inc/MarlinConfig.h:30,
                 from sketch/src/gcode/feature/camera/M240.cpp:23:
/home/cdelrey/Arduino/hardware/anet/avr/variants/sanguino/pins_arduino.h:43:0: warning: "TIMER2" redefined
 #define TIMER2  5
 ^
In file included from sketch/src/gcode/feature/camera/../../../inc/../HAL/HAL_AVR/HAL.h:27:0,
                 from sketch/src/gcode/feature/camera/../../../inc/MarlinConfig.h:30,
                 from sketch/src/gcode/feature/camera/M240.cpp:23:
/home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:210:0: note: this is the location of the previous definition
 #define TIMER2  6
 ^
In file included from /home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
                 from sketch/src/gcode/feature/camera/../../../inc/../HAL/HAL_AVR/HAL.h:27,
                 from sketch/src/gcode/feature/camera/../../../inc/MarlinConfig.h:30,
                 from sketch/src/gcode/feature/camera/M240.cpp:23:
/home/cdelrey/Arduino/hardware/anet/avr/variants/sanguino/pins_arduino.h:44:0: warning: "TIMER2A" redefined
 #define TIMER2A 6
 ^
In file included from sketch/src/gcode/feature/camera/../../../inc/../HAL/HAL_AVR/HAL.h:27:0,
                 from sketch/src/gcode/feature/camera/../../../inc/MarlinConfig.h:30,
                 from sketch/src/gcode/feature/camera/M240.cpp:23:
/home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:211:0: note: this is the location of the previous definition
 #define TIMER2A 7
 ^
In file included from /home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
                 from sketch/src/gcode/feature/camera/../../../inc/../HAL/HAL_AVR/HAL.h:27,
                 from sketch/src/gcode/feature/camera/../../../inc/MarlinConfig.h:30,
                 from sketch/src/gcode/feature/camera/M240.cpp:23:
/home/cdelrey/Arduino/hardware/anet/avr/variants/sanguino/pins_arduino.h:45:0: warning: "TIMER2B" redefined
 #define TIMER2B 7
 ^
In file included from sketch/src/gcode/feature/camera/../../../inc/../HAL/HAL_AVR/HAL.h:27:0,
                 from sketch/src/gcode/feature/camera/../../../inc/MarlinConfig.h:30,
                 from sketch/src/gcode/feature/camera/M240.cpp:23:
/home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:212:0: note: this is the location of the previous definition
 #define TIMER2B 8
 ^
In file included from /home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
                 from sketch/src/gcode/feature/camera/../../../inc/../HAL/HAL_AVR/HAL.h:27,
                 from sketch/src/gcode/feature/camera/../../../inc/MarlinConfig.h:30,
                 from sketch/src/gcode/feature/camera/M240.cpp:23:
/home/cdelrey/Arduino/hardware/anet/avr/variants/sanguino/pins_arduino.h:47:0: warning: "TIMER3A" redefined
 #define TIMER3A 8
 ^
In file included from sketch/src/gcode/feature/camera/../../../inc/../HAL/HAL_AVR/HAL.h:27:0,
                 from sketch/src/gcode/feature/camera/../../../inc/MarlinConfig.h:30,
                 from sketch/src/gcode/feature/camera/M240.cpp:23:
/home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:214:0: note: this is the location of the previous definition
 #define TIMER3A 9
 ^
In file included from /home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
                 from sketch/src/gcode/feature/camera/../../../inc/../HAL/HAL_AVR/HAL.h:27,
                 from sketch/src/gcode/feature/camera/../../../inc/MarlinConfig.h:30,
                 from sketch/src/gcode/feature/camera/M240.cpp:23:
/home/cdelrey/Arduino/hardware/anet/avr/variants/sanguino/pins_arduino.h:48:0: warning: "TIMER3B" redefined
 #define TIMER3B 9
 ^
In file included from sketch/src/gcode/feature/camera/../../../inc/../HAL/HAL_AVR/HAL.h:27:0,
                 from sketch/src/gcode/feature/camera/../../../inc/MarlinConfig.h:30,
                 from sketch/src/gcode/feature/camera/M240.cpp:23:
/home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:215:0: note: this is the location of the previous definition
 #define TIMER3B 10
 ^
In file included from /home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
[... 1309 lines cut ...]
 ^
In file included from sketch/src/HAL/HAL_AVR/../../inc/../HAL/HAL_AVR/HAL.h:27:0,
                 from sketch/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:30,
[... cut ...]
                 from sketch/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:30,
                 from sketch/src/HAL/HAL_AVR/persistent_store_eeprom.cpp:24:
/home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:223:0: note: this is the location of the previous definition
 #define TIMER5C 18
 ^
Error compiling for board Anet V1.0.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

dzamosenchuk, 2019-11-12:
Hello! I have the same bugs about error. I'm using Anet a2+ configuration, Marlin 1.1.x or bugfix version and Arduino IDE 1.8.10 or 1.9 beta. And board definitions from this repo: https://github.com/SkyNet3D/anet-board .
Could You help with this issue?

qberlemont, 2020-02-17:
> I'm having the same issue across all the suggested versions of Arduino and Marlin made above. I am constantly getting these TIMER messages;
> 
> _****Update, after updating "Arduino AVR Boards" and "Sanguino Boards" verify compile completed without error and was able to Flash without any issues so far****_**
> 
> Arduino: 1.9.0-beta (Linux), Board: "Anet V1.0"
> 
> In file included from /home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
> from sketch/src/gcode/feature/camera/../../../inc/../HAL/HAL_AVR/HAL.h:27,
> from sketch/src/gcode/feature/camera/../../../inc/MarlinConfig.h:30,
> from sketch/src/gcode/feature/camera/M240.cpp:23:
> /home/cdelrey/Arduino/hardware/anet/avr/variants/sanguino/pins_arduino.h:43:0: warning: "TIMER2" redefined
> #define TIMER2 5
> ^
> In file included from sketch/src/gcode/feature/camera/../../../inc/../HAL/HAL_AVR/HAL.h:27:0,
> from sketch/src/gcode/feature/camera/../../../inc/MarlinConfig.h:30,
> from sketch/src/gcode/feature/camera/M240.cpp:23:
> /home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:210:0: note: this is the location of the previous definition
> #define TIMER2 6
> ^
> In file included from /home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
> from sketch/src/gcode/feature/camera/../../../inc/../HAL/HAL_AVR/HAL.h:27,
> from sketch/src/gcode/feature/camera/../../../inc/MarlinConfig.h:30,
> from sketch/src/gcode/feature/camera/M240.cpp:23:
> /home/cdelrey/Arduino/hardware/anet/avr/variants/sanguino/pins_arduino.h:44:0: warning: "TIMER2A" redefined
> #define TIMER2A 6
> ^
> In file included from sketch/src/gcode/feature/camera/../../../inc/../HAL/HAL_AVR/HAL.h:27:0,
> from sketch/src/gcode/feature/camera/../../../inc/MarlinConfig.h:30,
> from sketch/src/gcode/feature/camera/M240.cpp:23:
> /home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:211:0: note: this is the location of the previous definition
> #define TIMER2A 7
> ^
> In file included from /home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
> from sketch/src/gcode/feature/camera/../../../inc/../HAL/HAL_AVR/HAL.h:27,
> from sketch/src/gcode/feature/camera/../../../inc/MarlinConfig.h:30,
> from sketch/src/gcode/feature/camera/M240.cpp:23:
> /home/cdelrey/Arduino/hardware/anet/avr/variants/sanguino/pins_arduino.h:45:0: warning: "TIMER2B" redefined
> #define TIMER2B 7
> ^
> In file included from sketch/src/gcode/feature/camera/../../../inc/../HAL/HAL_AVR/HAL.h:27:0,
> from sketch/src/gcode/feature/camera/../../../inc/MarlinConfig.h:30,
> from sketch/src/gcode/feature/camera/M240.cpp:23:
> /home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:212:0: note: this is the location of the previous definition
> #define TIMER2B 8
> ^
> In file included from /home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
> from sketch/src/gcode/feature/camera/../../../inc/../HAL/HAL_AVR/HAL.h:27,
> from sketch/src/gcode/feature/camera/../../../inc/MarlinConfig.h:30,
> from sketch/src/gcode/feature/camera/M240.cpp:23:
> /home/cdelrey/Arduino/hardware/anet/avr/variants/sanguino/pins_arduino.h:47:0: warning: "TIMER3A" redefined
> #define TIMER3A 8
> ^
> In file included from sketch/src/gcode/feature/camera/../../../inc/../HAL/HAL_AVR/HAL.h:27:0,
> from sketch/src/gcode/feature/camera/../../../inc/MarlinConfig.h:30,
> from sketch/src/gcode/feature/camera/M240.cpp:23:
> /home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:214:0: note: this is the location of the previous definition
> #define TIMER3A 9
> ^
> In file included from /home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
> from sketch/src/gcode/feature/camera/../../../inc/../HAL/HAL_AVR/HAL.h:27,
> from sketch/src/gcode/feature/camera/../../../inc/MarlinConfig.h:30,
> from sketch/src/gcode/feature/camera/M240.cpp:23:
> /home/cdelrey/Arduino/hardware/anet/avr/variants/sanguino/pins_arduino.h:48:0: warning: "TIMER3B" redefined
> #define TIMER3B 9
> ^
> In file included from sketch/src/gcode/feature/camera/../../../inc/../HAL/HAL_AVR/HAL.h:27:0,
> from sketch/src/gcode/feature/camera/../../../inc/MarlinConfig.h:30,
> from sketch/src/gcode/feature/camera/M240.cpp:23:
> /home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:215:0: note: this is the location of the previous definition
> #define TIMER3B 10
[... 1317 lines cut ...]
> ^
> In file included from /home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
> from sketch/src/HAL/HAL_AVR/../../inc/../HAL/HAL_AVR/HAL.h:27,
> from sketch/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:30,
> from sketch/src/HAL/HAL_AVR/persistent_store_eeprom.cpp:24:
> /home/cdelrey/Arduino/hardware/anet/avr/variants/sanguino/pins_arduino.h:55:0: warning: "TIMER5C" redefined
> #define TIMER5C 16
> ^
[... cut ...]
> from sketch/src/HAL/HAL_AVR/persistent_store_eeprom.cpp:24:
> /home/cdelrey/Downloads/arduino-1.9.0-beta-linux64/arduino-PR-beta1.9-BUILD-104/hardware/arduino/avr/cores/arduino/Arduino.h:223:0: note: this is the location of the previous definition
> #define TIMER5C 18
> ^
> Error compiling for board Anet V1.0.
> 
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.

Hello,

Could you explain how do you update "Arduino AVR Boards" and "Sanguino Boards" ?

Thx

jcc10, 2020-04-17:
Solution for future Peeps' is to use [this guy's](https://github.com/benlye/anet-board/) hardware profile.