MarlinFirmware/Marlin #10997: TMC2130 Xaxis not enabled when printing

landodragon141, 2018-06-11:
All right guys here's a weird one. With the latest 2.x version my tmc2130 somehow doesn't get enabled when printing. The weird part is I can manually move it and it works fine for homing and g29 ubl. Any thoughts?

landodragon141, 2018-06-12:
Let me clarify my previous post. It's only the X axis driver that doesn't get enabled. Y Z and E are also 2130's and they run just fine. I originally thought it was a gcode issue but the same issue occurs with both latest Slic3r PE and Cura.

landodragon141, 2018-06-23:
Sad no love for this issue.... I'll try 1.1.x to see if that makes a difference.

thinkyhead, 2018-06-30:
It's just weird. No one else has reported anything similar. Try swapping your stepper drivers around and see if it makes any difference.

mbuc, 2018-07-16:
Can you post debug output for the TMC's? TMC_DEBUG has to be enabled in Configuration_adv.h.  Send an M122 command and paste what you get here.

I've had similar issues in the past (December, 1.1.7) where my chip select (CS) pin for my X motor didn't have any activity on it.  I ended up reassigning the chip select to 59 for X_CS_PIN and 63 for Y_CS_PIN and that solved the issue then.

Recently, on a build from about a week ago, I have been having an axis or two stop working after pausing and then stopping a print in Simplify3D.  I have to do a full shutdown (PSU and USB disconnect) then power on PSU, then connect USB power.  In that order, or the TMC's just won't do anything.  I haven't dug into this much yet so I haven't posted an issue on it yet.

landodragon141, 2018-08-25:
I got my printer working and I'm not sure what changed. I did swap out the E0 driver for a spare A4988 I had laying around. I also moved the pins around and went back to HW SPI instead of SW SPI although that specific change didn't fix it. I double checked to make sure I didn't have any duplicate pins assigned and that was good. Basically I'm going to close this because I have no idea how I fixed it.

closed 2018-08-25 by landodragon141