MarlinFirmware/Marlin #11000: STM32F4 HAL and PlatformIO

alexxy, 2018-06-11:
Is it possible to build with PiO?

Using deafult platformio.ini seems not.
```
 ~/.local/bin/pio run -e STM32F4 

[Tue Jun 12 00:34:49 2018] Processing STM32F4 (platform: ststm32; board: disco_f407vg; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: ST STM32 > ST STM32F4DISCOVERY
SYSTEM: STM32F407VGT6 168MHz 128KB RAM (1MB Flash)
DEBUG: CURRENT(stlink) ON-BOARD(stlink) EXTERNAL(blackmagic, jlink)
Converting Marlin.ino
Error: This board doesn't support arduino framework!

```
If I change board to genericSTM32F407VET6 it will try but

```
$ ~/.local/bin/pio run -e STM32F4      

[Tue Jun 12 00:35:54 2018] Processing STM32F4 (platform: ststm32; board: genericSTM32F407VET6; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: ST STM32 > STM32F407VE (192k RAM. 512k Flash)
SYSTEM: STM32F407VET6 168MHz 128KB RAM (502.23KB Flash)
DEBUG: CURRENT(stlink) EXTERNAL(stlink)
Converting Marlin.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 11 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> v0.4
|-- <LiquidCrystal_I2C>
|-- <TMC2208Stepper> v0.1.1
|-- <30aa480>
|-- <Arduino-L6470> v0.6.1
|-- <SPI> v1.0
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_AVR/HAL.cpp.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_AVR/HAL_spi_AVR.cpp.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_AVR/MarlinSerial.cpp.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_AVR/persistent_store_impl.cpp.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_AVR/servo_AVR.cpp.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_AVR/watchdog_AVR.cpp.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/DebugMonitor_Due.cpp.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/EepromEmulation_Due.cpp.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/HAL.cpp.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/HAL_spi_Due.cpp.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/HAL_timers_Due.cpp.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/InterruptVectors_Due.cpp.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/MarlinSerialUSB_Due.cpp.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/MarlinSerial_Due.cpp.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/Servo_Due.cpp.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/Tone.cpp.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/persistent_store_impl.cpp.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/u8g_com_HAL_DUE_shared_hw_spi.cpp.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/u8g_com_HAL_DUE_st7920_sw_spi.cpp.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/usb/ctrl_access.c.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/usb/sd_mmc_spi_mem.cpp.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/usb/sysclk.c.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/usb/udc.c.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/usb/udi_cdc.c.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/usb/udi_cdc_desc.c.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/usb/udi_composite_desc.c.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/usb/udi_msc.c.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/usb/uotghs_device_due.c.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/usb/usb_task.c.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_DUE/watchdog_Due.cpp.o
Compiling .pioenvs/STM32F4/src/src/HAL/HAL_LPC1768/HAL.cpp.o
[... 138 lines cut ...]
compilation terminated due to -fmax-errors=5.
*** [.pioenvs/STM32F4/src/src/HAL/HAL_LPC1768/SoftwareSPI.cpp.o] Error 1
===================================================== [ERROR] Took 4.22 seconds =====================================================

============================================================= [SUMMARY] =============================================================
Environment megaatmega2560              [SKIP]
Environment megaatmega1280              [SKIP]
Environment at90USB1286_CDC             [SKIP]
Environment at90USB1286_DFU             [SKIP]
Environment DUE                         [SKIP]
Environment DUE_USB                     [SKIP]
Environment DUE_debug                   [SKIP]
Environment LPC1768                     [SKIP]
Environment melzi                       [SKIP]
Environment melzi_optiboot              [SKIP]
Environment rambo                       [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment sanguino_atmega1284p        [SKIP]
Environment STM32F1                     [SKIP]
Environment STM32F4                     [ERROR]
Environment teensy20                    [SKIP]
Environment teensy35                    [SKIP]
Environment malyanm200                  [SKIP]
===================================================== [ERROR] Took 4.22 seconds =====================================================
```

thinkyhead, 2018-06-11:
> Is it possible to build with PiO?

That's the only build method for the 32-bit platforms we officially support, so hopefully.

> If I change board to genericSTM32F407VET6 it will try but…

I'm not sure how supported this platform is currently. Anyway, the compiler complaints mean…

- The definition of `DEBUG_NONE` in `.platformio/packages/framework-arduinoststm32/STM32F4/cores/maple/libmaple/util.h` conflicts with `DEBUG_NONE` defined by Marlin. We would need to change our define to eliminate the conflict.

- Our definition of `CYCLES_PER_MICROSECOND` in `macros.h` assumes that the platform files will be included before `macros.h` is included. We'll need to gain more control over the order of inclusion to prevent `macros.h` being included before `.platformio/packages/framework-arduinoststm32/STM32F4/variants/generic_f407v/generic_f407v.h`

- The types `IWDG_HandleTypeDef`, `TIM_HandleTypeDef`, and `IRQn_Type` apparently don't exist (possibly as a result of the previous compile errors). If fixing the previous errors doesn't resolve this, we'll need to find out what the proper types are.

ktand, 2018-06-12:
The STM32F4/F7 HALs currently requires the STM32GENERIC Arduino core (
https://github.com/danieleff/STM32GENERIC). To be able to use STM32GENERIC
in PlatformIO it is necessary to use a modified STM32 development platform
for PlatformIO, available here:
https://github.com/maichaell/platform-ststm32.

I'm currently working on modifying the STM32F4 HAL to also support the
official STM32 Arduino Core (
https://github.com/stm32duino/Arduino_Core_STM32). I will create a PR later
today. Unfortunately the official STM32 Arduino core is not yet working
with the PIO STM32 dev platform so currently you have to build using
Arduino, either in Arduino IDE or Arduino for VS Code.

xC0000005, 2018-06-13:
If you want to build using PIO and the STM32GENERIC core, in your platformio.ini file, set:
platform = https://github.com/xC0000005/platform-ststm32.git#develop
and framework to arduinostm32generic.

It should automatically download everything and compile, but I'm seriously looking at switching to the official STM32 HAL because STM32GENERIC has a number of problems I'd rather not spend time chasing down.

mr-miky, 2018-06-17:
Stupid question , but using official STM32 is there any way to use the serial over USB port or do they have to implement it yet?

ktand, 2018-06-18:
In my fork of the official STM32 core I have implemented USB CDC support
for my STM32F407 "Arm'ed" board. It is based on the work in progress for
USB support in the original repository.

Please have a look here:

https://github.com/ktand/Arduino_Core_STM32/tree/armed_board_support

Karl

mr-miky, 2018-06-22:
This weekend I see to try it.
However, I have a question about how to handle the DTR. In cards with FT232 / CH340 the dtr is used to reset the microcontroller of the printer.
At first, I thought it was ok, just hook a function to the DTR handler that stays in loop and the hardware watchdog will reset everything accordingly.

The problem that follows is that the USB serial emulation is also resetted.

Pprograms like Pronterface act on the DTR but do not close ttyACM0 so in addition to blocking they prevent the registration of ttyACM0 and then lunux creates ttyACM1.
I do not know how to do a partial reset of the only part related to the print, if I call setup () what happens to the stack, the heap and the variables already initialized ?? !!

RobertFach, 2018-10-30:
Hi,
any progress here? I'm running in exactly the same problems as described by @alexxy 

I'm using:
platform = https://github.com/platformio/platform-ststm32.git
framework     = arduino
board         = disco_f407vg (or using board = genericSTM32F407VET6)

#define SERIAL_PORT 1
#define MOTHERBOARD BOARD_STM32F4

Any help would be awesome. Thx alot.

boelle, 2019-02-20:
@alexxy problem solved?

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-13 by thinkyhead