MarlinFirmware/Marlin #11005: Problem Z offset with UBL

ryan34side, 2018-06-12:
Hello

I use the UBL for leveling the bed but I can not find and the technology to do the offset Z With the ubl to achieve a beautiful first layer.
Can you help me ?

thank you very much

thinkyhead, 2018-06-13:
Use `BABYSTEPPING` to adjust the first layer at the start of a print. If `BABYSTEP_ZPROBE_OFFSET` is enabled, follow with `M500` to adjust the probe Z offset for future use. If you're using the Z probe for Z homing, adjusting the Z probe Z offset will help to produce a better height for all subsequent print jobs.

LVD-AC, 2018-06-23:
If you have a detachable probe or a delta this will obviously not work. PR #9310 fixes this issue for all printers (home with probe, detachable probe, delta's,...). See also #11046, #10961, ...

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-13 by thinkyhead