MarlinFirmware/Marlin #11007: Compilation Error USE_CONTROLLER_FAN

th33xitus, 2018-06-12:
This is my board:
```
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS4DUE_EFB
#endif
```
I try to enable a controller fan. I want pin 6 to be the signal pin. From that pin i go into a small mosfet circuit which switches the fan.
```
#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  #define CONTROLLER_FAN_PIN 6        // Set a custom pin for the controller fan
  #define CONTROLLERFAN_SECS 300          // Duration in seconds for the fan to run after all motors are disabled
  #define CONTROLLERFAN_SPEED 255        // 255 == full speed
#endif
```
The error-message i get:
```
In file included from C:\arduino_builds\sketch\src\inc/../HAL/HAL_DUE/HAL.h:40:0,

                 from C:\arduino_builds\sketch\src\inc/MarlinConfig.h:28,

                 from C:\arduino_builds\sketch\src\Marlin.h:25,

                 from C:\arduino_builds\sketch\src\Marlin.cpp:31:

Configuration_adv.h:201:39: error: pasting "/* Set a custom pin for the controller fan*/" and "_WPORT" does not give a valid preprocessing token

   #define CONTROLLER_FAN_PIN 6        // Set a custom pin for the controller fan

                                       ^

C:\arduino_builds\sketch\src\inc/../HAL/HAL_DUE/fastio_Due.h:63:31: note: in definition of macro '_READ'

 #define _READ(IO) bool(DIO ## IO ## _WPORT -> PIO_PDSR & MASK(DIO ## IO ## _PIN))

                               ^

C:\arduino_builds\sketch\src\inc/../HAL/HAL_DUE/fastio_Due.h:123:24: note: in expansion of macro '_SET_OUTPUT'

 #define SET_OUTPUT(IO) _SET_OUTPUT(IO)

                        ^

C:\arduino_builds\sketch\src\Marlin.cpp:800:5: note: in expansion of macro 'SET_OUTPUT'

     SET_OUTPUT(CONTROLLER_FAN_PIN); //Set pin used for driver cooling fan

     ^

C:\arduino_builds\sketch\src\Marlin.cpp:800:16: note: in expansion of macro 'CONTROLLER_FAN_PIN'

     SET_OUTPUT(CONTROLLER_FAN_PIN); //Set pin used for driver cooling fan

                ^

Configuration_adv.h:201:39: error: pasting "/* Set a custom pin for the controller fan*/" and "_PIN" does not give a valid preprocessing token

   #define CONTROLLER_FAN_PIN 6        // Set a custom pin for the controller fan

                                       ^

C:\arduino_builds\sketch\src\inc/../HAL/HAL_DUE/fastio_Due.h:63:70: note: in definition of macro '_READ'

 #define _READ(IO) bool(DIO ## IO ## _WPORT -> PIO_PDSR & MASK(DIO ## IO ## _PIN))

                                                                      ^

C:\arduino_builds\sketch\src\inc/../HAL/HAL_DUE/fastio_Due.h:123:24: note: in expansion of macro '_SET_OUTPUT'

 #define SET_OUTPUT(IO) _SET_OUTPUT(IO)

                        ^

C:\arduino_builds\sketch\src\Marlin.cpp:800:5: note: in expansion of macro 'SET_OUTPUT'

     SET_OUTPUT(CONTROLLER_FAN_PIN); //Set pin used for driver cooling fan

     ^

C:\arduino_builds\sketch\src\Marlin.cpp:800:16: note: in expansion of macro 'CONTROLLER_FAN_PIN'

     SET_OUTPUT(CONTROLLER_FAN_PIN); //Set pin used for driver cooling fan

                ^

Bibliothek Wire in Version 1.0 im Ordner: C:\Users\Admin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\libraries\Wire  wird verwendet
Bibliothek U8glib in Version 1.19.1 im Ordner: C:\Users\Admin\Documents\Arduino\libraries\U8glib  wird verwendet
Bibliothek TMC2130Stepper in Version 2.3.0 im Ordner: C:\Users\Admin\Documents\Arduino\libraries\TMC2130Stepper  wird verwendet
exit status 1
pasting "/* Set a custom pin for the controller fan*/" and "_WPORT" does not give a valid preprocessing token
```

Am i missing something or what did i do wrong?
It's not working with one of the 4 servo pins on the RAMPS nor with any other digital pin.
I don't understand the error-log :(

GMagician, 2018-06-12:
Not sure but try removing comment after pin definition, some arduino compiler bug comes in my mind...

th33xitus, 2018-06-12:
That actually worked!
Thank you!

@thinkyhead maybe worth considering to change the comments? Put the comments into the comment section above where the feature is explained maybe?

closed 2018-06-12 by th33xitus

thinkyhead, 2018-06-12:
Comments on pin definitions are helpful. We hope the compiler developers will eventually work out their issues and master their domain.

```cpp
#define CONTROLLER_FAN_PIN 6        // Set a custom pin for the controller fan
```