MarlinFirmware/Marlin #11008: MKS Sbase building broken with latest platformio in 2.0 branch

CageFox, 2018-06-12:
### Description

After upgrading platformio for latest version 3.6.0a2 and with latest source mks sbase hangs with fresh compiled firmware. I using atom on windows 8

### Steps to Reproduce

1. git clone latest marlin 2.0 branch
2. Copy mks sbase config example from src\config\examples\Mks\Sbase
3. Update platform io and packages:

platformio upgrade (3.6.0a2 newest)

platformio update

Updating nxplpc                          @ 3.3.0          [Up-to-date]
Updating toolchain-gccarmnoneeabi        @ 1.70201.0      [Up-to-date]
...

Build the firmware, upload to board and flash it usually way

**Expected behavior:** [What you expect to happen]

Board connected to PC visible in device manager and have com port and flash card as windows removable drive

**Actual behavior:** [What actually happens]

Board hangs almost immediately while flashing, after reset all leds on and no communication possible

#### Additional Information

changing SERIAL_PORT from 0 (in example) to -1 changes nothing

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

Files from example dir src\config\examples\Mks\Sbase

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

The board itself works with with smoothieware and also with marlin firmware I built about month ago. The problem as I see the image created by new platformio broken, even if I try to build rather old sources i can not make the board to work

thinkyhead, 2018-06-13:
You may want to report this to the PlatformIO project. Are you signed up for their message board?

CageFox, 2018-06-13:
Before reporting in PlatformIO I must be 100% sure the problem not in my particular system.

It will be very helpful if someone with mks sbase check it

jRam84, 2018-06-13:
Using Visual Studio Code and latest PlatformIO here, my MKS SBASE also bricks on flashing.

p3p, 2018-06-13:
Can confirm there is an issue, it's all LPC176x not just SBASE as they share the same code and build, I'm unsure the cause at the moment but looking into it.

p3p, 2018-06-13:
Not sure why the linker options are not being used from our CMSIS lib
Temporary fix:

platformio.ini (line 153)
```diff
build_flags       = !python Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py
  ${common.build_flags}
  -DU8G_HAL_LINKS
+  -Wl,-Tframeworks/CMSIS/LPC1768/Re-ARM/LPC1768.ld
```

CageFox, 2018-06-13:
"-Wl,-Tframeworks/CMSIS/LPC1768/Re-ARM/LPC1768.ld"

With this line added the firmware built half of it's normal size (76 kb instead of 120kb)  and the board still don't boot

jRam84, 2018-06-13:
I tried it and it worked on my board, 193kb size

p3p, 2018-06-13:
It will link smaller, this is caused by the correct options being used to build the binary. if you still cant get the board to boot (it should, this is the cause) can you link your binary and the config used to build it.

CageFox, 2018-06-13:
You just added line
-Wl,-Tframeworks/CMSIS/LPC1768/Re-ARM/LPC1768.ld
under
-DU8G_HAL_LINKS
?

@p3p 

[cagefox_marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2099564/cagefox_marlin.zip)

Files linked

CageFox, 2018-06-13:
@p3p "Re-ARM"? I trying to resurrect mks sbase, with this firmware it's not very active. After reset all leds turning on and that's all

p3p, 2018-06-13:
They both use LPC1768 microcontrollers so the platofrmio build issue from the update effect both the same. The fix above worked on my Re-ARM (was broken the same way) and seems to have fixed @jRam84's SBASE, The binary you supplied in the zip is functional in the sense that it will boot via the smoothieware bootloader

jRam84, 2018-06-13:
Yes I copy pasted the temporary fix under  -DU8G_HAL_LINKS. Compiled, flashed it and my mks sbase is alive and printing again.

CageFox, 2018-06-13:
Lucky guys. Are you printing from pc thru usb?

p3p, 2018-06-13:
Doesn't matter, the fault caused the binary to be 'corrupt', The problem you are having now is either a consequence of flashing the corrupt binary, if done via other means than just putting the firmware on the sd-card, or possibly unrelated. If the binary is trying to start but fails you will always get the 3 led flashes.. if they never flash you are having a boot-loader issue still.

jRam84, 2018-06-13:
No, if I use USB my board randomly disconnects and re-connects to my PC. Saving things to the EEPROM causes this too. Not sure if that is just my board or my PC but I only print from a SD card connected to the MKS TFT and I flash by using a microSD in the board.

CageFox, 2018-06-13:
@p3p The board have no problems working under smoothieware, so i think bootloader is ok. I have no screen connected to board so I decide the board alive or not by connecting to PC - and right now the board seems not alive

I'ts not very urgent, if the problem confirmed I can wait for more permanent solution or better workarounds

p3p, 2018-06-13:
I'l keep looking but unfortunately I don't have a SBASE to test on and as far as I can tell the issue reported is fixed with the workaround. @jRam84 could you try a build with @CageFox's configs or see if his binary works on your board?

CageFox, 2018-06-13:
@p3p Workaround is workaround, @jRam84 not using pc connectivity, and I using the board only this way. There is no pc connectivity with current workaround and I very very dislike the thought about returning to smoothieware, so I hope there will be **The Fix** wich brings me back full power of marlin.

For now I found no way to downgrade platformio, may be there is other way to build working firmware?

CageFox, 2018-06-13:
@p3p I can check new workarounds upon my mks board if you don't have one

p3p, 2018-06-13:
> Workaround is workaround

In this case it is **The Fix** until I can figure out why platformio decided to start ignoring the option when specified in the library descriptor, and I can say that with 'some' confidence given I designed the mess that is the LPC176x platform for Marlin,

As a side note, the binary you linked in the zip actually runs fine on my Re-ARM, .. can send serial commands and get the expected responses, so platformio is building correctly for you now it seems.

CageFox, 2018-06-13:
With this binary my board for very short time creates COM4 and two seconds later it became COM5 with "Marlin USB" and "device not working"

I think workaround became The Fix when it commited to main branch and the issue can be closed completely

"and I can say that with 'some' confidence given I designed the mess that is the LPC176x platform for Marlin"

Interesting phrase, I hope you know da way ;)

jRam84, 2018-06-13:
Compiled and flashed CageFox's config files, I had to edit baudrate to 115200 and add a second serial port to 0 for my MKS TFT to function and it does work. I was able to move all axis around, so something else is happening to your board.

CageFox, 2018-06-13:
@jRam84 Strange, very strange, are you using windows? What version? Maybe it's simple some drivers issue... But smoothieware works like there is nothing happens

CageFox, 2018-06-13:
@jRam84 Have you set first com port to -1 and second to 0?

CageFox, 2018-06-13:
@jRam84 Wait a moment, after edits you were able to connect to pc or to tft? Or to both?

jRam84, 2018-06-13:
I do not have the PC near the Printer so I can't verify if it connects to both the PC and MKS TFT but that is how I used to do it before moving the Printer away from the PC.

I'm using latest Windows 10 with Visual Studio Code, all libraries and extensions up to date.

CageFox, 2018-06-13:
So, for now we can be sure the workaround let's the board to connect to mks tft, but my preferred way to print from pc not working for me and didn't tested by others

It's pity I have no backup of firmware before last troubles. If someone have the working mks sbase built before platformio update firmware please link it

p3p, 2018-06-13:
@CageFox The problem that the workaround fixes is not something that can cause random issues, either the binary works or it doesn't, it's a linker error. I have the LPC176x firmware working here (built with the SBASE config) with usb serial and disk drive working correctly.

Here are 2 binaries that I have tested and execute correctly, allowing control over usb serial.
built with platformio: [binary-pio.zip](https://github.com/MarlinFirmware/Marlin/files/2100250/binary-pio.zip)
built with my dev environment: [binary-eclipse.zip](https://github.com/MarlinFirmware/Marlin/files/2100244/binary-eclipse.zip)

CageFox, 2018-06-13:
@p3p No, both binaries flashing all leds and no connect to pc

If there anyone have binaries built for mks sbase week or so ago link t please

PS. Flashed smoothieware and all connects, leds flashing, sd card accessible as drive.

p3p, 2018-06-13:
> No, both binaries flashing all leds and no connect to pc

One of those binaries does not use PlatformIO at all during the build process, so we can assume that you no longer have the issue you opened, its something else.

CageFox, 2018-06-13:
@p3p My thoughts ended for now. I only know I printed till yesterday update and i can still print with smoothieware. If it's only me who have problems the issue may be closed

closed 2018-06-14 by thinkyhead

thinkyhead, 2018-06-15:
#11026 is now merged.

J-charles-C, 2018-06-20:
I, 
Same Problem form me.

There are a new version of VSCODE and the fix is out.

best
`

thinkyhead, 2018-06-21:
@J-charles-C — So is there still a problem or is the problem fixed?

J-charles-C, 2018-06-21:
I,

No. I have downloaded a day marlin and erase vscode. I have an error message. With my hold firmware, another default message with no explain. 

I chearch.

best

CageFox, 2018-06-21:
Building with Atom and Platformio - all ok, problem fixed. Firmware binary became half size of it were before fixes, nice

J-charles-C, 2018-06-21:
Ok, thanks.

I have a problem with my vscode.  I test tomorrow

best

thinkyhead, 2018-06-21:
I good luck, I.

hermitcrabslab, 2018-07-14:
`
....
_LPC1768\u8g Marlin\src\feature\bedlevel\mbl\mesh_bed_leveling.cpp
aException in thread Thread-2:
Traceback (most recent call last):
  File "C:\Python27\Lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "c:\users\xyz\.platformio\penv\lib\site-packages\platformio\util.py", line 100, in run
    self.outcallback(line)

  File "c:\users\xyz\.platformio\penv\lib\site-packages\platformio\managers\platform.py", line 399, in on_run_out
    self._echo_line(line, level=1)
  File "c:\users\xyz\.platformio\penv\lib\site-packages\platformio\managers\platform.py", line 420, in _echo_line
    click.secho(line, fg=fg, err=level > 1)
  File "c:\users\xyz\.platformio\penv\lib\site-packages\platformio\__main__.py", line 114, in <lambda>
    click.secho = lambda *args, **kwargs: _safe_echo(1, *args, **kwargs)
  File "c:\users\xyz\.platformio\penv\lib\site-packages\platformio\__main__.py", line 111, in _safe_echo
    sys.stdout.write)("%s\n" % (args[0] if args else ""))
IOError: [Errno 0] Error`
For some reason it dosen't work on my windows 10 machine :(
Also I failed to build on OSX/Linux too.
[config.zip](https://github.com/MarlinFirmware/Marlin/files/2195059/config.zip)
I have to admit this platformio thing is new to me, didn't expect to be easy pie but still, please help I have no idea what I do wrong.
MKS Sbase 1.3 + TFT32, I also tried with default LPC1768 configs untouched, it fails too.

CageFox, 2018-07-15:
Just tried latest commit with atom and platrofmio - nothing changed, position spammint and halts
[image]

CageFox, 2018-07-15:
I noticed the spamming of position happens in g-code like this:
G1 X60.222 Y63.079 E2.6718
G1 X60.538 Y63.326 F5700
G92 E0
G1 X59.751 Y64.283 E0.0206 F4560
G1 X58.920 Y65.200 E0.0412

The whole g-code generated by s3d

I attaching the g-code file 
[marlin_YetAnotherUtahTeapot-lid.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/2196134/marlin_YetAnotherUtahTeapot-lid.gcode.zip) may be it will help to find problem

RavenWarrior, 2020-05-09:
What board should I select for mks robin mini?