MarlinFirmware/Marlin #11016: Capacitance Sensor Z-Level Failure

Grant-fm, 2018-06-13:
### Description

I have a capacitance sensor, recently installed, on my printer that stops working when I unplug my computer USB cable to the controller board.  The sensor accuracy is impacted by the heat bed (when it is heating the sensor calibration alters), the hot end, and fan.  It appears that the electrical fields around the sensor are problematic.  Having said that, these issues I can somewhat deal with, however, having to keep my laptop connected to the board is more perplexing.  

### Steps to Reproduce

1. Connect the sensor to the 24V power terminals and the signal wire, with an apposing diode, to the Z-min signal pin
2. Plug a printer in to the board (the sensor should work)
3. Now unplug the USB cable from the computer (now the sensor will stop working)
4. By stop working, what I mean is the sensor light goes out and the sensor no longer operates
5. By plugging in the computer USB cable to the controller board again, the sensor light comes back on and the sensor works perfectly.  (I don't see why this would happen?)

When you send an M119 command you get a Sensor Triggered flag back (while Z Min is Open.  

#### Additional Information
Board - MKS -GEN L 
Sensor - LJC18A3-B Z/AX
Marlin 1.1.x

Print surface Aluminium / silicon heating pad / glass top

thinkyhead, 2018-06-13:
It doesn't sound like a firmware bug. For help with electronics / wiring / hardware issues, try the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)

houseofbugs, 2018-06-22:
@Grant-fm you have a grounding issue on the machine.

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-13 by thinkyhead