MarlinFirmware/Marlin #11017: Filament jamming

Lyr3x, 2018-06-13:
I'm fighting this issue since a week now and i dont know what else i can do.
Marlin: latest bugfix-1.1.x
Board: MKS Gen 1.4 with TMC2130
E3DV6

Printed thousand of hours withput bigger problems but since a week i am getting continuous  filament jams in nozzle/heatbreak. This happens with PLA(215°) and with PETG (230°). I need to pull out the filament cut it off and can insert it again. After a few seconds or sometimes minutes it happens again.

I swapped out the Heatbreak (fullmetal), nozzle, heatblock, ptfe tube. Cleaned the extruder gear, checked the thermistor and changed the noctua fan (with a spare new one). No positive results.

I really dont know whats the problem here. Hopefully somebody can help me out here

oysteinkrog, 2018-06-13:
Maybe your environment has changed, e.g. if there is summer you may have higher ambient temperature, which can affect things a lot. In my experience using a Noctua fan on the E3DV6 can be very borderline.

Lyr3x, 2018-06-13:
That came into my mind aswell, so i waited a few days and now it is about 20° in the room like most of the time.. Do you have any recommendation for a powerful silent fan?

oysteinkrog, 2018-06-13:
Oh, if it's the 20mm version it should be enough, as I think that is even more powerful than the stock fan.
I suggest you look at the E3D forum, as this is not really a Marlin firmware problem.

Lyr3x, 2018-06-13:
Maybe i leave this opened and will provide a solution if i found one too help others ?

forkoz, 2018-06-13:
adda AD0412HB-C50 is nice. But I use 10mm deep fans so I have sunon maglevs.  Actually, 9.4 cubic meters is middle to slightly above average for a 10mm fan but weak for 20mm. With PTFE heat break you could get away with it, but with a metal hot end no way.

Lyr3x, 2018-06-13:
I dont think this is really the problem. I printed honestly more than 1000 hours with this setup and never had any clogs or jams with normal PLA or PETG. 

And i never had a PTFE heatbreak...

forkoz, 2018-06-13:
Could be... but you replaced everything else. I had the same issue until I upgraded the fan when I installed a metal break. It would stop extruding but pulling it out and cutting off the end would get it working again for a while. Only other thing I can think of is check heat break alignment, ie how far its into the cooler/heat block and also temp sensor. Did you get jams with retraction turned off.

Lyr3x, 2018-06-13:
Allright changing the Fan is not a big Deal. I changed the noctua with an more powerful 40x10 Fan. I also reassambeled the hotend once more. We will see how the benchy comes out

Lyr3x, 2018-06-13:
So the benchy failed again. After about 3/4 the nozzle jammed. I really dont know where the problem is..

thinkyhead, 2018-06-13:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)

closed 2018-06-13 by thinkyhead