MarlinFirmware/Marlin #11021: [FR] Timeout while printing via USB

Vabe7, 2018-06-14:
Two days ago I printed from Octoprint via USB. For some reason Octoprint crashed midprint and the printer just remained at the last position and also the temperature was kept at the same level. 
To make Marlin more secure I would wish for a timeout that sets the temperature to zero after a in Configuration.h specified time passed without a received G-Code-command. Especially in times where reports of burning printers I see a need for such a function.

forkoz, 2018-06-14:
I've had the same thing happen when my PC blue screened mid print. Just sat there cooking a blob on the hot end. It would really only printing things like nylon and PC.

comps, 2018-06-14:
http://marlinfw.org/docs/gcode/M085.html is probably what you're looking for.

I set it to 10 seconds in my "post gcode" in a slicer just before it starts executing gcode for the first layer (after wiping, leveling and heating up).

AcHub, 2018-06-14:
Personally I would feel better if this protection would be hardcoded into the firmware, instead of just a Gcode. Start and End-gcode can be altered way to easy and I can very well imagine a situation where I could accidentially delete such a command from the script without noticing. Disabling a safety feature should not be possible by accident, because we all know - bad things will eventually happen, if they can happen.

Maybe thats a stupid question, as I never used M85, so I don't exactly know how it works and what it considers to be "activity" regarding the timeout-timer, but if this command halts the whole printer after timeout, wouldn't that also be an issue with for example filament change? If the printer waits for the user to interact, and the user doesn't change filament fast enough, the whole print would be lost. However if the heaters are shut down only, the fire hazzard is probably banned but you would still be able to resume your print after reheating the nozzle. Sure - you could think ahead and add M104 S0 to your filament change script to cool down your nozzle while it waits, but again, that is unsafe and might depend on an external controller for sending the command if not printing from SD card.

AnHardt, 2018-06-14:
If myprinters couldn't stand "print temperature" for a "unlimited time", i wouldn't dare to use it at all.

Vabe7, 2018-06-14:
With Octoprint this shouldn't be a problem, it sends a command (not sure which right now) every second or so to get the temperature reading. M85 seems to be exactly what I need, I simply wasn't aware of this command.

thinkyhead, 2018-06-14:
I think that `M85` pretty much covers it. The one thing we could do to augment it is to add a default `MAX_INACTIVE_TIME` configuration option. Then the machine would, by default, call `kill` whenever it was idle for a period of time.

gege2b, 2018-06-14:
Damn, I got this problem too, and I was thinking of a similar FR while going back home, glad to see someone did it :)
I was unaware of M85 and, for me, it's just perfect and I'll insert it on my start GCode as soon as I finish this post (with more than 10s btw, maybe something like 5 minutes)

@AnHardt , safety is not the only concern... Leaving a printer with heaters elements on for nothing is not really a good thing, even if the printer is perfectly safe.  this feature would avoid a big mess of filament, and reduce useless power consumption too

forkoz, 2018-06-14:
Did not know about this either, I'll definitely add it to the start/end code. To me its enough. I don't think my printer would start a fire but it could very well melt the nozzle into the printed item, burn the PEI sheet, destroy the heating element, etc.

thinkyhead, 2018-06-14:
It would be more comprehensive to have the machine time out after some period of inactivity, then (if it has an LCD) put the machine into a "paused" state with an on-screen option to reheat to the last-set temperature and resume. Essentially this is what the parking function does now, so it could be done as an add-on to the `ADVANCED_PAUSE_FEATURE` in future.

bruce356, 2018-06-15:
I was also not aware of M85
On a dual x carriage dual extrusion printer there is also inactivity while one hot end is cooled at park and wait temperature and the other heated to extrusion temperature, this time can vary from job to job.
So I guess one needs to decide on a max time.
Also leaving filament sitting in the heated hot end without extrusion over a long period of time also has consequences.
regards

gege2b, 2018-06-15:
@bruce356 Idem, but since a few weeks, there are different reports in octoprint's repo about printers that stops mid-print with heaters on (it happened to me, too), foosel is still investigating on what could cause this 

Actually, M85 is a good way to mitigate potential damages when this happens, but I agree that with some setup it could be more tricky

comps, 2018-06-15:
I still think this is a valid FR for a keepalive protocol, perhaps as a new concept, perhaps extending [M113](http://marlinfw.org/docs/gcode/M113.html) with a flag that requires hosts to report back once M113 is called, perhaps simply as expecting [M105](http://marlinfw.org/docs/gcode/M105.html) in regular intervals after its first use, for hosts that don't use [M155](http://marlinfw.org/docs/gcode/M155.html), like printrun.

Having it as a separate feature would allow the host software to use this transparently (wouldn't be used on sd card print) and gcodes wouldn't have to embed [M85](http://marlinfw.org/docs/gcode/M085.html). Also, `FILAMENT_RUNOUT_SCRIPT` wouldn't have to work around it, etc.

thinkyhead, 2018-06-15:
> it could be interesting then to move the head a little bit to avoid direct contact with the object underneath the nozzle.

If it were done as an extension to `ADVANCED_PAUSE_FEATURE` the nozzle would raise up, move to the park position, and cool down after a short time. This would only occur when the nozzle was heated and the machine were "idle" for a long enough period. If the idea was then to "wake up" in response to a resumption of serial-printing activity, then the nozzle would wait till it was re-heated, then move back to the previous position and lower back down before processing the next serial command. The host would wait while Marlin output "busy" messages, and so `HOST_KEEPALIVE` would also be required for this to be viable.

boelle, 2019-02-20:
@Vabe7 is this still relavant?

Vabe7, 2019-02-23:
To be honest I didn't follow the development but I think M85 fits the problem nicely. On the other hand the mentioned ideas would improve Marlin further, especially from a safety point of view... So to me it isn't relevant personally, but for the wider audience I still think that a timeout feature would be handy.
Maybe also the others could share their opinion?

boelle, 2019-02-23:
for me it's a bit of mjeeh, i figured out by a lot of testing that i get better prints when printing from sd card, 

since i upgraded to re-arm i 'm now testing printing from a raspberry pi 3 so there are a potential now that the pi might stop sending data and the printer just cooks whats in the nozzle

so if there should be a feature i would vote for a feature in the firmware so its independant from the gcode

filipgoc, 2019-03-14:
If some of the thinking is guided by basically an extension of thermal protection, some sort of flag could be for killing the printer not after idler for a time, but after being at certain temperature for a time. That's the more dangerous case.

boelle, 2019-06-22:
missing a label

Legionofone, 2020-02-19:
Had this happen to me last night, PC that was sending gcode to a printer crashed and left the nozzle over a print. This definitely seems like a nice piece of mind safety feature. 

It looks like this was implemented into prusa's firmware as the M86 command.

thisiskeithb, 2021-07-05:
Added in #16362 with `HOTEND_IDLE_TIMEOUT`.

closed 2021-07-05 by thisiskeithb