MarlinFirmware/Marlin #11023: [bugfix-2.0.X] Cannot leave sub-menu on printer status display while printing

AcHub, 2018-06-14:
### Description
On my Printer, while I was printing, when I went to the menu [Settings] -> [Filament] and scrolled down to the last menu entry "load mm:", I got stuck on this entry and couldn't move the menu selector by turning the rotary switch and therefore also couldn't leave the menu no more. (I don't use English as language on my printer, so maybe the menu names are a bit different). Also I couldn't select "load mm:", which was highlighted at that time, by pressing the rotary switch.
Even after 15 minutes the display did not return to the home-display and the rotary switch stayed unresponsive (timeout in configuration_adv is 60000). Display also didn't change when the print was finished (print was successfull).

Interestingly, after the print finished and my End-Gcode was executed, the communication with my printer failed and Octoprint terminated the session. Octoprint terminal log from the time when the End-Gcode starts untill the connection dies:

Send: N98456 G91\*23
Recv: ok
Send: N98457 G1 Z3\*102
Recv: ok
Send: N98458 G92 E0\*79
Recv: X:87.31 Y:121.94 Z:17.00 E:0.00 Count X:6988 Y:9760 Z:6439
Recv: ok
Send: N98459 G1 E-1.5 F500\*16
Recv: ok
Send: N98460 M104 S0\*102
Recv: ok
Send: N98461 M140 S0\*103
Recv: ok
Send: N98462 M106 S0\*102
Recv: ok
Send: N98463 G90\*16
Recv: ok
Send: N98464 G28 X0 Y0\*21
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: X:0.00 Y:0.00 Z:17.00 E:-1.50 Count X:0 Y:0 Z:6808
Recv: ok
Send: N98465 G1 Z150 F9000\*15
Recv: ok
Send: N98466 M84\*26
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: ok
Changing monitoring state from "Printing" to "Operational"
No response from printer after 3 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from "Operational" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
Connection closed, closing down monitor

Although there is a M106 S0 sent and confirmed by the printer to shut down the parts cooling fan, it actually is still running even after the connection fails. Hotend and Printbed in contrary were both turned off successfully.

### Steps to Reproduce

1. Start a Print
2. Go to the filament settings menu
3. Scroll down
4. Get stuck

**Expected behavior:** 
Don't get stuck in the menu.

**Actual behavior:** 
Getting stuck in the menu.

#### Additional Information
I'm running on Firmware 4c8d6df (June 13).

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2102617/Configuration.zip)

I went to that menu to verify the K-value beeing used by the printer, as I had accidentially set a wrong K-value in my start Gcode and then changed that K-value while it was already printing by sending M900 K0.1 via Terminal from Octoprint.

thinkyhead, 2018-06-14:
Is this 100% reproducible?

AcHub, 2018-06-15:
It is reproducable, maybe not 100% right off, but it will happen after a few seconds if I scroll up and down within the sub-menu (the faster I scroll around, the faster it crashes). 

As I was just testing right now I noticed that the whole menu is kind of lagging. It is not as responsive as it used to be and when scrolling around it jumps over some menus.
When the menu freezed, I aborted the print and while it was lowering the build plate (part of my end Gcode) I was turning the rotary switch and the menu selector happend to move once, but then freezed again. Finally, Octoprint would report a connection failure, just as described in my previous post. Also the parts cooling fan would again stay on.

I don't use my printer menu very often, so I can't tell with certainty if this is something introduced with the latest code changes, or if it has been there before unnoticed. I _feel_ like it hasn't been there with 2ba198a (June 11), but I would have to do some tests to check, if this info is required. The only thing I remember to have changed between the configs from June 11 to 13 is, that I increased the number of grid points per dimension for Auto bed leveling bilinear from 4 to 8. Could this be too much to handle?

thinkyhead, 2018-06-15:
> Could this be too much to handle?

Anything up to 16 should be ok.

boelle, 2019-02-20:
@AcHub is this still an issue?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-03-17 by p3p