MarlinFirmware/Marlin #11025: Changing the TMC2130 current on E0 also affect other axis.

JoeyCo, 2018-06-15:
HI,

I changed my old MK8 extruder for a printed geared extruder and had problem with retraction. The stepper seemed too weak to spin the gear fast enough so I thought I should try to increase current on E0 axis from 900 to 1000.
I then noticed that every axis where affected by this change. I used M906 to be sure I only changed E0 and it does state that X,Y,Z axis are still at 900 and E0 is at 1000. Yet, steps/mm and motors sound changed on all axis.
When I first installed those TMC2130 I noticed that changing the current to something above 950 also change the steps/mm ratio. (I don't know if this is normal)

I made some search on google but can't seem to find any one having the same issue.

Thank you for your time.

### Description

Changing the TMC2130 current on E0 also affect other axis.

### Steps to Reproduce

1. Change the current on E0 from 900 to 1000.
2. Try to print.
3. Notice that every axis motors sound has changed and steps per mm are now wrong

**Expected behavior:** Only E0 current should change

**Actual behavior:** Every axis is affected

#### Additional Information

I use marlin 1.1.8.
This is not a new setup. I print with this machine for a while now. I just changed the current on the extruder to see if the motor was stronger this way.
The drivers are Fystec TMC2130 in SPI mode
The motors are 17HS4401 (Nema17 rated to 1.7A)

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/2104376/Configurations.zip)

teemuatlut, 2018-06-15:
Try the bugfix branches.

JoeyCo, 2018-06-15:
Sorry for the troubles. The bugfix branche is working properly.

Thanks.

closed 2018-06-15 by JoeyCo