MarlinFirmware/Marlin #11027: AZSMZ Board Hangs on Startup

alexyu132, 2018-06-15:
I've recently run into a strange issue with my AZSMZ board+AZSMZ12864 LCD setup. After compiling the latest bugfix-2.0.x code and flashing the bin via SD card, the board never starts up properly - it just sits there with the extruder fan at full speed and blank LCD. What's even stranger is that when I tried to revert to an earlier commit I had working before, the new compile of that commit also resulted in a hang on startup. When I upload a Smoothieware binary, it works fine, and an old firmware.bin compiled in April also works fine.

Is there something that changed in one of the dependencies that could be causing this issue? I've tried reverting the configs to almost stock (only changing the motherboard and enabling sd/lcd) and redownloading zips of various recent commits, but this doesn't seem to fix the issue. I've verified that the build target is LPC1768, which should be correct for my board. Very confused as to what the cause could be.

kAdonis, 2018-06-15:
There is a problem with the latest platformIO 
Issue #11008. 
you can use the Workaround described, or wait for PR #11026 to be merged

thinkyhead, 2018-06-15:
#11026 is now merged.

closed 2018-06-15 by thinkyhead

CageFox, 2018-06-16:
Thanx, this commit fixed my issue as well