MarlinFirmware/Marlin #11029: Compilation Error - UBL with RAMPS4DUE

th33xitus, 2018-06-15:
Commit `fdd97e6`

Board `#define MOTHERBOARD BOARD_RAMPS4DUE_EFB`

Function i activated `#define AUTO_BED_LEVELING_UBL`

Error Log:
```
In file included from C:\arduino_builds\sketch\src\feature\bedlevel\ubl\ubl.cpp:30:0:

C:\arduino_builds\sketch\src\feature\bedlevel\ubl\../../../module/configuration_store.h:101:47: error: 'E2END' was not declared in this scope

         static constexpr int16_t meshes_end = E2END - 128; // 128 is a placeholder for the size of the MAT; the MAT will always

                                               ^

In file included from C:\arduino_builds\sketch\src\feature\bedlevel\ubl\ubl_G29.cpp:33:0:

C:\arduino_builds\sketch\src\feature\bedlevel\ubl\../../../module/configuration_store.h:101:47: error: 'E2END' was not declared in this scope

         static constexpr int16_t meshes_end = E2END - 128; // 128 is a placeholder for the size of the MAT; the MAT will always

                                               ^

C:\arduino_builds\sketch\src\feature\bedlevel\ubl\ubl_G29.cpp: In static member function 'static void unified_bed_leveling::g29_eeprom_dump()':

C:\arduino_builds\sketch\src\feature\bedlevel\ubl\ubl_G29.cpp:1169:31: error: 'E2END' was not declared in this scope

     for (uint16_t i = 0; i <= E2END; i += 16) {

                               ^

Bibliothek Wire in Version 1.0 im Ordner: C:\Users\Admin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\libraries\Wire  wird verwendet
Bibliothek U8glib in Version 1.19.1 im Ordner: C:\Users\Admin\Documents\Arduino\libraries\U8glib  wird verwendet
Bibliothek TMC2130Stepper in Version 2.3.0 im Ordner: C:\Users\Admin\Documents\Arduino\libraries\TMC2130Stepper  wird verwendet
Bibliothek SPI in Version 1.0 im Ordner: C:\Users\Admin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\libraries\SPI  wird verwendet
exit status 1
```

I don't understand the error message so i don't know how to solve the issue?
Just wanted to try UBL again. It works fine with bilinear ABL but i wanted to play around with UBL :D

th33xitus, 2018-06-15:
Okay and how can i fix this?
Usually the Due has no EEPROM but i connected one via I2C which works fine for all other settings i can save to the EEPROM.

ejtagle, 2018-06-15:
Add to your configuration_adv.h
```
#define E2END 0x1FFF
```

Assuming you have an 8kbyte eeprom

The value is just the last valid EEPROM address

thinkyhead, 2018-06-15:
You might need both:
```cpp
#define I2C_EEPROM
#define E2END 0x1FFF
```

th33xitus, 2018-06-15:
I have an 256kbyte EEPROM.
What value do i have to put there instead?

thinkyhead, 2018-06-15:
Any value up to 0x3FFFF should work.

thinkyhead, 2018-06-15:
~~Any value up to 0x3FFFF should work.~~

Marlin only supports up to 32K, so use 0x7FFF.

th33xitus, 2018-06-15:
Seems like it compiled fine!
Will flash and test it tomorrow, but i am sure it will work now :)
Thanks for the fast support!

closed 2018-06-15 by thinkyhead