MarlinFirmware/Marlin #11030: BabyStepping double_click not working - Marlin 2.0.0

italocjs, 2018-06-15:
I'm trying to activate the baby stepping through double click on the lcd, but its not working, 
Here are my settings:

Expected action: press the button for 1+ seconds and show the babystepping screen.
Actual action : press the button for 1+ and get redirected forward and back to the menu screen

`#define BABYSTEPPING
#if ENABLED(BABYSTEPPING)
  //#define BABYSTEP_XY              // Also enable X/Y Babystepping. Not supported on DELTA!
  #define BABYSTEP_INVERT_Z false    // Change if Z babysteps should go the other way
  #define BABYSTEP_MULTIPLICATOR 5   // Babysteps are very small. Increase for faster motion.
  //#define BABYSTEP_ZPROBE_OFFSET   // Enable to combine M851 and Babystepping
  #define DOUBLECLICK_FOR_Z_BABYSTEPPING // Double-click on the Status Screen for Z Babystepping.
  #define DOUBLECLICK_MAX_INTERVAL 4000 // Maximum interval between clicks, in milliseconds.
                                        // Note: Extra time may be added to mitigate controller latency.
  //#define BABYSTEP_ZPROBE_GFX_OVERLAY // Enable graphical overlay on Z-offset editor
#endif
`
Also i'm usint the #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

This function is working on Marlin 1.1.8

oysteinkrog, 2018-06-15:
From your description it sounds like you are pressing and holding the button, is that correct?
You need to double click the button.

italocjs, 2018-06-15:
Tried both, neither work :(

italocjs, 2018-06-15:
Sorry everyone, this is not an error nor bug, its just a change (and a good one)

Previously it was possible to access this menu when not printing, now it only works when there is some movement.

Closing the issue

closed 2018-06-15 by italocjs