MarlinFirmware/Marlin #11031: De-skewing Z axis

mikeytoo, 2018-06-15:
### Description

### Steps to Reproduce

I don't know whether you can drive the Z-axis motors independently, but if you can it would be possible to de-skew the Z-axis to the bed during auto homing.  Once auto homing is complete, average the left-side Z values and the right-side Z values separately then subtract one average from the other.  The resulting value is the amount of skew on the platform.  Drive one motor this amount and your platform is parallel to the bed.

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

boelle, 2018-06-16:
problem: what if both the bed and X axis is skewed?

when you say Z axis i assume you do mean that X is not parallel to bed

bjbeare, 2019-02-11:
I've been thinking about this problem for a while... and looked at the dual z stepper code a while ago. I need to look at it again. IIRC, the code needs to first be reworked to allow the steppers to be handled independently.

My concept would then be to tweak one of the Z steppers after an autolevel. Depending on the orientation of the dual z steppers (front/back or right/left), this would correct the leveling of either Y or X. The other axis can't be corrected... but it could significantly the amount of autolevel correction required.

What do you think of the concept? I could invest some time in this... since I'd like the feature for my printer (homebrew Vulcanus large-bed variant).

kAdonis, 2019-02-11:
@bjbeare  What about G34 Z_STEPPER_AUTO_ALIGN?

bjbeare, 2019-02-11:
@kAdonis Wow... you guys are amazing. You put that in about a month after I researched the issue. :o). Can't wait to try it out.

boelle, 2019-02-20:
@bjbeare did it do the trick?

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-14 by thinkyhead