MarlinFirmware/Marlin #11035: One of two TMC2130 does not communicate anymore after replacing and reflashing Arduino

HWiese1980, 2018-06-16:
### Description

Due to a severe hardware defect I had to replace the Arduino underneath my RAMPS 1.4. All I did was removing the old Arduino, plugging in the new one and re-flashing Marlin with the very same configuration I've had on the old Arduino. Now one of two TMC2130 does not communicate anymore. 

I have TMC2130 on my X and Y axis. The Z axis is a TMC2100 and the E0 is an old A4899. The TMC2100 and the A4899 work like a charm. 

I've already double and triple checked the wiring of the TMC2130 and the pin settings (which I haven't changed). It all seems OK. The wiring is based on this diagram: https://cdn.instructables.com/FJM/7VHE/JEEUHFSL/FJM7VHEJEEUHFSL.LARGE.jpg

I've already tried to  rule out the bread board which I used before for splitting the SPI lines for the two TMCs. I've replaced it by some soldered pinheads. The wires have an overall length of about 30cm from the RAMPS to the TMCs... which is kinda long, maybe too long for SPI, but I don't have any shorter cables around to rule that out. Also I wonder why one of the TMCs at least communicates while the other does not. They both have the same cable length. 

One kinda funny thing I noticed is that the CS pins were swapped before. That still worked because the two TMCs are configured identically. So it doesn't matter whether they get their own or the config for the respective other TMC. But what I actually noticed is that after correcting this small wiring error the TMC that stopped communicating also switched. Before that it was the TMC of the X axis, now it's that of the Y axis that won't communicate. So it seems the communication lines are OK. 

Any help would be greatly appreciated

### Steps to Reproduce

It's a rather individual problem. Don't know whether this can be reproduced. 

Unfortunately, I don't own a scope, otherwise I would have checked already what's going on on the lines, especially the CS lines. 

**Expected behavior:** [What you expect to happen]

Both TMCs should communicate

**Actual behavior:** [What actually happens]

As mentioned above, only one TMC communicates, depending on the wiring of the CS line. 

#### Additional Information

* my configuration.h: https://pastebin.com/VyMLteSF
* my configuration_adv.h: https://pastebin.com/Vb64xEuu
* the corresponding M122 output: https://pastebin.com/QJeNAJqJ

ManuelMcLure, 2018-06-16:
Given that the problem moves with the CS line it sounds like whatever pin is used for the CS line for the driver that doesn't communicate is not working properly - possibly there's a bad solder joint on the new Arduino. If you can try another pin for that CS line it might work. From the wiring diagram you provided it looks like D44 might work for you.

teemuatlut, 2018-06-16:
Looks like both your drivers have an issue.

Try testing only one TMC2130 driver at a time, and preferably with shorter, unspliced wires.

HWiese1980, 2018-06-16:
@teemuatlut What makes you conclude that both my drivers have issues?

@ManuelMcLure I've tried several pins for CS now, none of them seems to work.

teemuatlut, 2018-06-16:
From the M122 output

> READ: Driver registers: X = 0x00:00:00:00
> READ:   Y = 0xFF:FF:FF:FF

HWiese1980, 2018-06-16:
I haven't figured out what these numbers mean in detail. All I found was that 0xFF:FF:FF:FF indicates a communication error. 

The Y driver also reports an overtemp error right after switching on. The X driver does not.

HWiese1980, 2018-06-16:
Okay, this is totally weird. 

@teemuatlut I've tested the TMCs separately with shorter, unspliced wires (about 15cm each). I've tried the first TMC which is on my X driver socket, and it gives registers 0x00:00:00:00 while Y gives 0xFF:FF:FF:FF (correct, it's not connected). Then I moved the wires 1:1 to the second TMC which is in the Y driver socket. I plugged them to the socket exactly as they were plugged to the other TMC. Now X gives 0xFF:FF:FF:FF (okay, not connected) but Y still gives 0xFF:FF:FF:FF as if there was no connection. 

I'm still hoping that this is a software issue.

hamster65, 2018-06-16:
I had issues with my ramps board right after purchase because of bad solder connections. I resoldered all pins and it has been fine since.But it took ours figuring out where the problem was.

HWiese1980, 2018-06-16:
Well, my RAMPS has been working fine until I grilled the Arduino and swapped it for a new one. 

I've decided now to move on and replace the whole controller with a new Due/RADDS. Wanted to go for 32bit for quite a while now anyway and this is as good an opportunity as any other, I suppose.

HWiese1980, 2018-06-19:
@melvinisken Yeah, likely. I've ordered a RADDS and a Arduino Due in order to upgrade to 32bits and rule out Arduino and RAMPS as cause for the issue. I hope that the TMCs are still alive. Maybe if I find the time I'll temporarily wire them up to a working Arduino and see if I can still communicate with them in that setup with some very basic code... keep your fingers crossed! :fearful:

boelle, 2019-02-20:
@HWiese1980 problem solved?

HWiese1980, 2019-02-25:
@boelle As mentioned in #10320, in a way, yes... whereas "solved" is imho not the correct term. I'd rather call it "circumvented" by switching hard- and software. The TMC2130s somehow died in the process. I don't know why. It's weird and sad, but it's the outcome. 

I've got my printer running with a Due/RADDS/TMC2208 setup and Klipper as the new software, 'cos I like the concept of the minimalistic firmware footprint.

closed 2019-05-16 by thinkyhead