MarlinFirmware/Marlin #11036: Turn on fan when thermal runaway (like the Prusa MK3)

Ivomola, 2018-06-16:
Hi. I have a Tevo Tarantula with a blower fan and a duct so the air goes to the nozzle, and when it's at full blast it can't even reach more than 220 degrees. The thing is that my thermistor alredy fall of the hotend 2 or 3 times and yeah, the thermal runaway protection kicked in, but my hotend was alredy smoking when it did, so I thought: If the Prusa i3 MK3 wich uses Marlin can do it, why I can't? If this function is alredy implemented I'll love to hear how to activate it, especially since i'm using the 2nd extruder terminals to control my fan cause' I shorted out the fan ones and now if I plug it there I can't control it (If you ask, no, I didn't change any of the code, this function was alredy implemented in Marlin Tarantula Easy config by @JimBrown)

italocjs, 2018-06-17:
Interesting, i'd like to have this option too.

For the thermistor falling off, id suggest you fixing it better, maybe with high temperature silicone.  for the thermal runaway, try lowering the histeresis and detection time, so it can detect faster, but this could also create some false-positives.

Ivomola, 2018-06-21:
Hi. Sorry for not responding, I was busy configuring autolevel with a 2 mm sensor (bad idea). About the thermistor, I've alredy fixed it, but as for the thermal runaway, it alredy gave me some false positives cause of my fan, so maybe it's something about the PID values, I don't know, but I think it will be a good idea to have it so if the software doesn't act as fast as it should, at least it can save your printer.

melvinisken, 2018-06-21:
No fan should cool down your nozzle. Then the fan setup is wrong. You should have one fan cooling the coldend and one fan cooling the parts but NOT the nozzle. I would suggest to print a new fan holder. Alternatively you can search for those silicon socks to put around the hotend to prevent the fan cooling it too much.

Ivomola, 2018-06-21:
Yeah, that's true, maybe I'll have to calibrate Y steps/mm and print it again, but that doesn't change my idea of turning on the fan when the printer gives you an error because, well, let's just say that Anet claims that they're not responsable of the 2 houses burned because of having the thermal runaway disabled.

AnHardt, 2018-06-21:
If it's already burning, pushing more oxygen in, by a van, seems be a not good idea, to me.

Ivomola, 2018-06-21:
It isn't about pushing more oxygen (especially since metal can't burn, at least aluminum), but cooling it down to prevent it from melting

AnHardt, 2018-06-21:
The classic horror scenario, what thermal runaway protection is made for is:
Heat element is dropping out of the heater block and powers up because the heater block and the thermistor cools down.
A red glowing heater cartridge is now dragged over a half finished plastic part and degrades it pretty fast.

Switching off the heater is about the only thing we can do under all circumstances without possibly making things worse.

AnHardt, 2018-06-22:
Can you predict what is falling out of the block - heater or thermistor? I can't.

ps.
The ideal part cooling fan is not cooling the heater block. Either the block has a insulation, or the fan a duct to avoid exactly that. If not you have to relax the 'thermal runaway protection' parameters to be able to print with a (full speed) running part cooling fan. That is extremely counter productive.

Even Prusa is only cooking with water - like us. We all do make errors.

Ivomola, 2018-06-22:
So what you're saying is that my idea doesn't make sense and that I need to
throw away all the time I spent trying to add that to the code? Well, I
suppose I'm on my own then.

italocjs, 2018-06-22:
You didnt get the idea AnHardt.  this is a safety feature not just against fire, but accidental burning.   When my printers finish the printing, they do enable both fans at maximum speed so the nozzle will cool faster, this way there is less chance that the user will acidentally burn itself when removing the part.  Today i do it on the end gcode script, but it will be nice to have this feature if something else happened, i.e fatal errors. 

I'm with you ivomola, this can be a useful feature

siana, 2018-06-22:
Not implemented, and Prusa doesn't run Marlin, it runs Prusafirmware which was forked from Marlin years ago and diverged significantly.

I don't have an opinion on whether it should or shouldn't be done, just clarifying.

Ivomola, 2018-06-22:
Ok, thanks for clarifying it, I didn't know that, I thought that it used a
modified version of marlin cause in an interview with Josef Prusa of Thom3d
he said "Yeah, we use a modified version of marlin..."

AnHardt, 2018-06-22:
@italocjs 
There is a difference between 'end code', where cooling makes mostly perfect sense, and the automatisms of a safety feature.

@Ivomola 
If i can't convince you, feel free to implement it and try to convince someone to merge it. For sure i will kink back to here.

Ivomola, 2018-06-22:
Well, let's see if @thinkyhead wants to add a comment here (cause normally
he appears on long threads like this one)

dot-bob, 2018-06-23:
I have an idea for you... Make an option to run a custom gcode script upon thermal runaway before it halts.  Each machine is different and you may want to respond differently to a run away.  Then someone could run a script to turn off the power supply, toggle a gpio, or bump your fan to full on.

marcio-ao, 2018-06-29:
> I have an idea for you... Make an option to run a custom gcode script upon thermal runaway before it halts.

@dot-bob: I just imagined someone putting a glass of water on a corner of their print bed and having a thermal runaway script that dunks the print head to put out the fire... :)

thinkyhead, 2018-06-29:
I agree with @AnHardt's point that it's a bad idea to always turn on the fan in all cases.

Note that there are a couple of distinct features related to thermal errors. There's MINTEMP / MAXTEMP which are always-on features to catch the case of a thermistor becoming disconnected or failing. Then there's the "Thermal Runaway" error, which is used to catch both the case when a heater cartridge has failed or fallen out, or when there's simply unusual thermal behavior.

In future we might add options to define G-code for all kinds of events. The different thermal runaway errors may be included in that list of events.

Ivomola, 2018-07-03:
@thinkyhead Actually, I think that's a far better idea than mine was,
because I think it is more customizable and thus, it will protect much more
printers than simply turning on the fan. The thing was, I wanted to keep it
as simple as possible so it's simpler to implement because, well, I'm new
to coding and even more to 3D printing, but if someone can implement that
option it will be very nice, and plus, referencing to @AnHardt, turning on
the fan won't harm anything if it turns on when an error like that happens,
it will just protect the hotend in the cases it can, and when for example,
the heater falls off, like I said it won't harm anything.

Ivomola, 2018-07-10:
And adding to the fan setup, I just received a new heater and, well, let's just say that the problem wasn't the fan (well, it wasn't the MAIN problem), and that I was using the stock heater... Yeah.

closed 2018-07-10 by Ivomola