MarlinFirmware/Marlin #11039: I2C communications fouling

jaysonkelly, 2018-06-16:
I'm using the i2c encoders and getting strange results with the most recent revisions of Marlin. Is it possible that something in the main body of the code could be taking priority over sending the clock signals for i2c?

When looking at the communications with a logic analyzer at the point when marlin freezes and sometimes resets, I see abnormal clock activity on the i2c bus. 

Any thoughts as to what could be causing it?

jaysonkelly, 2018-06-17:
[image]

The top output of the logic analyzer is the failed communication after which Marlin reported an incomplete response. The one in the snipping tool below that is a successful communication. Note the shortened final pulse at the end of the communication and the abnormal termination as the clock and data lines go high later than usual. 

Since the clock is controlled by the Master in an i2c bus, it is Marlin that seems to be the cause of the failed communications. 

@thinkyhead ? The issue exists in 1.1.8 and 1.1.x bugfix

jaysonkelly, 2018-06-19:
There may be some other causes for the I2C communication issues I'm seeing. I'll try shielded cables first and different frequencies on the heated bed. 

Thanks!

closed 2018-06-19 by jaysonkelly

Grogyan, 2018-06-19:
@jaysonkelly if the I2C cables aren't shielded, the high frequencies of the stepper would disrupt I2C communication.

Grogyan, 2018-06-19:
Altneratively you can look into using a CAN bus stepper/encoder system. then use a SPI to CAN bus interface board to talk with the firmware.
Though at this stage, CAN Bus has not yet been implented into Marlin firmware

jaysonkelly, 2018-06-19:
I think shielded is the next thing to try. Thanks!