MarlinFirmware/Marlin #11040: Sensorless Z-homing don't work [tmc2130] [Marlin-bugfix-1.1.x - 16.06.2018]

alakond, 2018-06-16:
Hi.
I have 3 TMC2130 and I'm trying to make a sensorless home.
Marlin settings in the attachment.
[Z-axis.zip](https://github.com/MarlinFirmware/Marlin/files/2108498/Z-axis.zip)

The problem is that the X and Y axes work correctly, and the Z axis does not stop when it stops in the nozzle. 

In addition, M914 why is this (see Stallguard thrs	7	6	**0**): 
```
Recv: M122
Recv: 		X	Y	Z
Recv: Enabled		false	false	false
Recv: Set current	900	900	500
Recv: RMS current	887	887	489
Recv: MAX current	1251	1251	689
Recv: Run current	28/31	28/31	15/31
Recv: Hold current	14/31	14/31	7/31
Recv: CS actual		14/31	14/31	7/31
Recv: PWM scale		0	0	0
Recv: vsense		1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true
Recv: msteps		16	16	16
Recv: tstep		1048575	1048575	1048575
Recv: pwm
Recv: threshold		98	98	329
Recv: [mm/s]		100.85	100.85	3.00
Recv: OT prewarn	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false
Recv: off time		5	5	5
Recv: blank time	24	24	24
Recv: hysteresis
Recv: -end		2	2	2
Recv: -start		3	3	3
Recv: Stallguard thrs	7	6	0
Recv: DRVSTATUS	X	Y	Z
Recv: stallguard
Recv: sg_result		0	0	0
Recv: fsactive
Recv: stst		X	X	X
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:	X = 0x80:0E:00:00
Recv: 	Y = 0x80:0E:00:00
Recv: 	Z = 0x80:07:00:00
```
or:
```
Recv: echo:Sensorless homing threshold:
Recv: echo:  M914 X7 Y6
```
The firmware states:

```
#define SENSORLESS_HOMING // TMC2130 only
   #if ENABLED (SENSORLESS_HOMING)
     #define X_HOMING_SENSITIVITY 8
     #define Y_HOMING_SENSITIVITY 7
     #define Z_HOMING_SENSITIVITY 8
   #endif
```

teemuatlut, 2018-06-17:
You're using a probe to home Z axis, this also disables sensorless homing on Z.

Also the option should **ONLY** be used with delta printers or machines where Z is driven with belts.
A typical cartesian with lead screws has way too much torque on Z to reliably detect motor load and accurately trigger an endstop pulse.

tobyak, 2018-06-20:
Would sensor less homing work on a core XY, given it's unique motion system

pgScorpio, 2018-06-21:
@teemuatlut Sensorless Z homing on machines where Z is driven with lead screws can work fine, but you need to tune the motor current and homing speed correctly....

See chapter 2 of https://www.trinamic.com/fileadmin/assets/Support/Appnotes/AN002-stallGuard2.pdf 

(Systems with two motors on one driver however can be a pain in the ass, no matter if belt or lead screw driven.)

alakond, 2018-06-22:
Hi all. Thnx all for the help. I'll make "a pain in the ass" with a piezo Z-sensor.

thinkyhead, 2018-06-30:
> Would sensor less homing work on a core XY, given its unique motion system.

Yes.