MarlinFirmware/Marlin #11041: [2.0] Can't flash MKS SBASE

manutenfruits, 2018-06-16:
I have been running my MKS SBASE with Marlin 2.0 for a few months now, and merging changes every once in a while. I updated my code today and tried to flash it, but won't work. I [recorded a video](https://photos.app.goo.gl/stk7wCJUtyZuxXtS7) of the light sequence when flashing, where it stops halfway through. I tried going back to the version I had working before, as well as fresh out of `bugfix-2-0-x` HEAD and still won't flash. In case the board was fried I tried flashing Smoothieware, which worked perfectly.

I am trying to find a way to debug this issue, I am compiling using PlatformIO, as far as I know using the exact same method I did last time I flashed the firmware. Is there anything I can do to debug the situation?

manutenfruits, 2018-06-16:
Yes! I am up to date on `bugfix-2.0.x`, I see the changes on my workspace.

p3p, 2018-06-16:
Ok, make sure to do a clean build, and delete all PlatformIOs tempory folders (.pioenvs, .piolibdeps) in the Marlin folder, The fix should be working and had for others with (inclusing myself) with this issue.

Phr3d13, 2018-06-16:
Could it have been a platformio board related update that broke things for you? Maybe a bad rate or something?

manutenfruits, 2018-06-16:
I gave it another try after doing clean and removing the dotfiles. It stopped at a different point, unsure if that's any progress.

> Maybe a baud rate or something?
AFAIK no baud rates are involved here, it's just compiling, moving to the SD and powering up the board. I know it's not fried because I can flash Smoothie without problem.

p3p, 2018-06-16:
Marlin doesn't actually control the LEDs, that's the bootloader so they will stay in whatever state they were in when Marlin takes control after the flashing, I'm fixing that now so you actually get the status LED like you are supposed to.

I have to assume its a caching issue with PlatformIO, I have multiple LPC176x boards all booting OK (including a MKS SBASE), unless you are having a different issue but it seems unlikely.

p3p, 2018-06-16:
I combined the MKS SBASE LED fix into my current PR #11032, so you could try that, 
The boot sequence is 3 flashes indicates Marlin is booting then fast flashes indicate USB is trying to connect, followed by a steady heartbeat (if it is erratic it indicates the mcu can't handle the load or something is not allowing the watchdog to be reset)

Phr3d13, 2018-06-17:
I should stop guessing on boards I have no clue about. 🤐

manutenfruits, 2018-06-19:
I got somewhere. It flashed (too quick almost) and restarted, my computer detected a TTY port and mounted the SD card when connecting the board through USB. However, I don't seem to be able to talk to it using Pronterface or Octoprint. I don't think having nothing connected to it (while I debug) should affect that. I can connect to the TTY using atom but I see nothing being returned, and I can't send any commands either.

manutenfruits, 2018-06-19:
Yes thanks! It does blink, then blink faster for a second or two, then back to normal blinking. So I think I'm past my original problem. However, I am unable to connect to it using pronterface. I checked baud rate as 250000 and the TTY port, but it just says "Connecting..." and the board keeps on blinking and doing nothing.

p3p, 2018-06-19:
I can only guess without your configs, tis why I asked for them :wink:, Make sure you are setting the Serial port correctly in your config, the default changed to use Hardware port 1 rather then both USB (-1) and Hardware, baud rate actually has no purpose with the USB serial as it's a native port it runs at USB speed.

manutenfruits, 2018-06-19:
ooops I totally missed that. [Here is the config I'm trying right now](https://github.com/MarlinFirmware/Marlin/compare/bugfix-2.0.x...manutenfruits:testing-mks-sbase), to discard any config mistake I could've done.

p3p, 2018-06-19:
Look like your problem is as I said above, you have the wrong Serial port selected, 
https://github.com/manutenfruits/Marlin/blob/b2c34946ab9e567b032fefdfc97640f07564e819/Marlin/Configuration.h#L108

Needs to be set to -1 for USB.

manutenfruits, 2018-06-19:
Thanks! I totally missed that. Got it all working!

closed 2018-06-19 by manutenfruits

projectoblio, 2019-09-30:
Can you please mention what the board / controller  settings you're using in Arduino are for the MKS Sbase?