MarlinFirmware/Marlin #11043: [bugfix-2.0.x] BLTouch works only once after power-on cycle

yodor, 2018-06-16:
### Description

Hello
During z-probing with BLTouch (antclabs) the probing works only the first time after power-on cycle - soft reset does not resolve the issue only power on.
In 1.1.8 release this works very well.

### Steps to Reproduce

1. Power on printer
2. Home 
3. Call gcode G33 P1
4. Wait for successful z-height calibration and homing.
5. Store with M500
7. Call gcode G33 P1 again.
8. Probe touches the bed then lifts some and just before final homing stops and error message is shown

**Expected behavior:** [What you expect to happen]
G33 P1 to succeed each time.

**Actual behavior:** [What actually happens]
G33 P1 have error on each probing after first.

#### Additional Information

Log with first probe ok then second failing.

02:06:53.126 : echo:enqueueing "G33 P1"
02:06:53.140 : G33 Auto Calibrate
02:06:53.140 : Checking... AC
02:06:53.145 : .Height:284.37                                Offset:-0.82
02:07:29.089 : Calibration OK                                std dev:0.000
02:07:29.092 : .Height:284.36                                Offset:-0.82
02:07:29.094 : Save with M500 and/or copy to Configuration.h
02:16:43.351 : X:0.00 Y:0.00 Z:284.36 E:0.00 Count X:75614 Y:75614 Z:75614
02:16:54.028 : echo:enqueueing "G33 P1"
02:16:54.043 : G33 Auto Calibrate
02:16:54.043 : Checking... AC
02:16:54.045 : .Height:284.36                                Offset:-0.82
02:17:17.833 : Error:Probing failed
02:17:17.835 : Correct delta settings with M665 and M666

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2108575/Configuration.zip)

yodor, 2018-06-17:
I've done more testing and also enabled DEBUG_LEVELING_FEATURE.
It seems after powerup m502 m500 m501 probing works many times until m500 and m501 is done. After that probing fails as above.
Attaching two logs after M111 S32

[log_probing_ok_before_m500.txt](https://github.com/MarlinFirmware/Marlin/files/2108926/log_probing_ok_before_m500.txt)
[log_probing_fail_after_m500.txt](https://github.com/MarlinFirmware/Marlin/files/2108927/log_probing_fail_after_m500.txt)

italocjs, 2018-06-17:
I'm using 2.0.0 bugfix and its working fine here. The only big difference i can see is that you are using a delta printer, and i'm on corexy.  Maybe there is something else off on delta specific configurations?

yodor, 2018-06-17:
Not sure I'm prepared to dig deeper into this :) But atleast it works before doing m500, so it might be something to do with storing delta settings ? Also on Marlin 1.1.8 the issue is not existing - it works very well I've done hundreds of probings.

thinkyhead, 2018-06-30:
Some changes have recently been made to endstop and motion code. Please re-test the latest `bugfix-x.x.x` code to see if it's any better.

yodor, 2018-09-15:
I think this is related to #11827[https://github.com/MarlinFirmware/Marlin/issues/11103#issuecomment-421387611]

closed 2018-09-15 by yodor