MarlinFirmware/Marlin #11044: G29_RETRY_AND_RECOVER off by one

hexane360, 2018-06-17:
### Description
When using the `G29_RETRY_AND_RECOVER` feature, the retry loop is incorrectly set up. 

### Steps to Reproduce

1. Enable G29_RETRY_AND_RECOVER
2. Set G29_MAX_RETRIES
3. Bypass probe signal
4. Run G29

**Expected behavior:**
Failed bed probes are retried `G29_MAX_RETRIES` times, before failing.

**Actual behavior:**
Failed bed probes are retried `G29_MAX_RETRIES - 1` times, with one extra `G29_RECOVER_COMMANDS` followed immediately by `G29_FAILURE_COMMANDS`.

#### Additional Information
The relevant code is in [GcodeSuite::G29_with_retry()](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/gcode.cpp#L139):
```
for (uint8_t i = G29_MAX_RETRIES; i--;) {
      G29();
      if (planner.leveling_active) break;
      #ifdef G29_ACTION_ON_RECOVER
        SERIAL_ECHOLNPGM("//action:" G29_ACTION_ON_RECOVER);
      #endif
      #ifdef G29_RECOVER_COMMANDS
        process_subcommands_now_P(PSTR(G29_RECOVER_COMMANDS));
      #endif
}
```

marcio-ao, 2018-06-29:
@hexane360: Good catch. Thank you.

thinkyhead, 2018-06-29:
A `G29_MAX_RETRIES` of 1 means, "only try once." Perhaps it should have been named `G29_MAX_TRIES` instead. For now we'll just add one to it to alleviate some confusion.

marcio-ao, 2018-06-29:
@thinkyhead: However @hexane360 made a good point that it is running the recovery routine one last time needlessly.

I think the code needs to be like this:

```
for (uint8_t tries_left = G29_MAX_RETRIES;;) {
      G29();
      if (planner.leveling_active || tries_left-- == 0) break;
      #ifdef G29_ACTION_ON_RECOVER
        SERIAL_ECHOLNPGM("//action:" G29_ACTION_ON_RECOVER);
      #endif
      #ifdef G29_RECOVER_COMMANDS
        process_subcommands_now_P(PSTR(G29_RECOVER_COMMANDS));
      #endif
}
```

thinkyhead, 2018-06-29:
So the action on recover is meant to happen before each `G29` except the first. Got it.

closed 2018-06-29 by thinkyhead

marcio-ao, 2018-07-02:
> So the action on recover and recover commands are meant to happen after each failed G29? Ok, patched.

The recovery action should do whatever is necessary to help a subsequent G29 to succeed, such as wiping nozzles, leveling X axis, etc. So it generally is a waste of time to execute it if there will be no more G29s issued. This is what was happening before. It would do:

- G29 fails
- Wipes
- G29 fails
- Wipe
- Printer Halts

Notice that the last wipe doesn't need to be there.

PR #11045 (2018-06-30): Fixes G29_with_retry behavior
### Description
As currently implemented, `G29_RETRY_AND_RECOVER` is situated outside all `G29()` code, and attempts reprobe attempts whenever leveling is disabled after running `G29()`. However, there are several reasons `G29()` can leave without enabling bed leveling that have nothing to do with a failed probe (axes not homed, dry run, report, etc.). There are two ways to solve this: to turn `G29()` into a `bool` that indicates whether a probe has failed, or to handle reprobe attempts inside `G29()`. This PR is the second approach, currently for ABL only. Implem [... cut]