MarlinFirmware/Marlin #11046: M851 z() does not change G29 bed leveling

SimonWitcher, 2018-06-17:
Hi everyone,
Im trying to put up together a delta machine, finally got some good results form calibration and bed levelling as well ( I thinko so at least)
Im using a conductive probing with aluminium foil on my bed. After I made G29 I used M500 to save my bed leveling results. 
I started with 0.5 z offset which was a little too high and I wanted to lower it down using M851 Z0.4, however it does not change antything, the nozzle is still to high when printing. 
Im using bilinear bed leveling now. I have M420 S1 command in my slicer starting G-code.

[Desktop.zip](https://github.com/MarlinFirmware/Marlin/files/2108865/Desktop.zip)

dough29, 2018-06-17:
Is M503 reporting the right offset ?

You have to do M501 after M500 to load stored config.

SimonWitcher, 2018-06-17:
Yes M503 is reporting the offset I set.

I think I was doing M501 also, but rechecked this just in case and the result is still the same

forkoz, 2018-06-17:
I thought bed leveling can be done from z =0 since its measuring the distance at which your sensor activates in a particular spot as offset from the original homed position. So afterwards Z offset doesn't matter as its applied to all g code evenly. I've just been doing it at z0 and then changing the offset however, am I wrong?

SimonWitcher, 2018-06-17:
forkoz, Im not sure what do you mean.
Bed leveling measures Z positions on different spots on XY plane. Your probe however can be higher or lower to your hotend, so in that case efery point would be too high/too low of your actual bed.
Z offset is correcting that, so your measured plane meets your bed, it does not work in my case however.

forkoz, 2018-06-17:
What I mean is that the leveling is measured from the sensor and where it first homed (ie, center at Z0). So if you set the z offset to 0 and run a bed level the created plane should be the same at all Z offsets. Shifting the "0" so your hot end is actually the right distance from the bed won't affect the values saved from the G29. 

So maybe .4 is still too much.

dough29, 2018-06-17:
Offset don't have to be a negative value ?

I dunno how conductive probe works but with BLTouch I have to set a negative value.

SimonWitcher, 2018-06-17:
forkoz, I tried different values, event went down to -0.5 with no change whatsoever. The nozzle is always a very little higher than it should so the filament does no stick. 
Im looking for a way to lower down z offset without the need to bed level again since I need to lay down aluminium foil on a bed surface to do it.

dough29, yes offset can be any value. You can actually set it in a configuration.h. By default it can vary from -20 to 20 mm.

forkoz, 2018-06-17:
So you never go down to machine 0 and stick paper or a feeler gauge under the nozzle? I don't re-level the bed either when I throw glass on top because the clips would hit the probe. So I level at 0 and then lower the nozzle with soft end stops off until it touches a feeler gauge to get z offset. Then just baby step as desired to go a little lower or higher.  My z offset without glass is like -1.6, with glass it was like -0.25.

SimonWitcher, 2018-06-17:
Can u describe the exact g-code commands u are using?
I did not heard of soft end stop offset.

forkoz, 2018-06-17:
When I first started this guide saved me many headaches. And now I never bake Z offset into firmware or the bed leveling

Z-Offset Instructions: 
1. Home 3D printer 
2. M851 Z0 - Reset Z0Offset 
3. M500 - Store setting to eeprom 
4. M501 - Set active parameters 
5. M503 - Display Active Parameters 
6. G28 Z - Home Z Axis 
7. G1 F60 Z0 - Move nozzle to true 0 offset 
8. M211 S0 - Switch off soft endstops 
9. Move nozzle towards bed slowly until the paper can barely move 
10. Take note of the Z on the printer display (take that number and add the measurment of the calibration sheet or device used) 
11. M851 Z X.XX (X.XX being your z offset achieved) 
12. M211 S1 - Enable Soft Endstops 
13. M500 - Save settings to Eeprom 
14. M501 - Set Active Parameters 
15. M503 - display current settings

SimonWitcher, 2018-06-17:
Thanks, im gonna try it tomorrow later and let u know of the results.
Can you tell me more about these software endstops disabling? What does it do exacly and why you want them off during changing M851 z?

dough29, 2018-06-17:
Disabling soft endstops let you go under Z0 tout get your offset ;)

dough29, 2018-06-17:
Take a look : https://youtu.be/y_1Kg45APko

SimonWitcher, 2018-06-17:
Ok, but delta printer homes to Z max position. The soft endstops were not the problem.
This procedure is exacly the same what I used before so it wont work.

The problem is that after i do G29 I want to change this offset using M851 **without** being have to G29 again. However it does not work.

It is not the problem to tell where the offset is. The problem is that I cant change it.

forkoz, when u mention u dont re-level u mean auto leveling or manual leveling using knobs?

forkoz, 2018-06-17:
Auto leveling. The sensor itself can't measure anything but the presence of metal. So those +/- points are all from the first 0 it gets as far as I understand. So I do the leveling first before I set the offset to avoid baking any offsets into the values (I had issues when I started at different heights on previous releases and got values like 1.3 instead of .3). The sensor is always going to activate in the same place and then the z offset is applied to the whole thing.

I think you can do the same thing afterwards with G29 P5/P6: http://marlinfw.org/docs/gcode/G029-ubl.html but this is how it's worked for me.

LVD-AC, 2018-06-23:
> The problem is that after i do G29 I want to change this offset using M851 without being have to G29 again. However it does not work.

Indeed a well know issue that keeps haunting delta's. PR #9310 fixes this. Until this gets merged, as a workaround you can do G33 P1 to set a baseline **before** running G29 and use G33 P1 **after each** M851 change.

See #10961 among others.

SimonWitcher, 2018-06-23:
I switched to UBL and was using G29 P6 C to change offset value, however it is acting strange, from time to time it just gets zeroed or a written value changes with no reason.
Other issue I had with UBL was that despite changing DELTA_PROBEABLE_RADIUS to 100. it was still probing at 130 radius.

thinkyhead, 2018-06-28:
At the moment #9310 is still not yet fully-baked because it applies some changes universally that should only apply when `DELTA` is enabled, but it's getting closer by degrees.

thinkyhead, 2018-06-28:
> I switched to UBL and was using G29 P6 C to change offset value, however it is acting strange

Definitely post a separate issue about that so it can be attended to by the author of UBL.

mylife4aiurr, 2018-07-07:
Problem 1
@SimonWitcher I have been pulling my hair out with this on my Delta as well for a few days now. I have notes I use when ever I update to new firmware - basically clear eeprom, delta config, UBL bed -> leave active g29 A, disable softend stop, find good z offset, reenable softend stop. (sometimes ill fine tune mesh afterwards)

But recently the above procedure doesn't work as far as finding a good zoffset. I tried on bugfix 1.1.x fc9f4ce The offset is not used. It is saved - checked m503. Just not enforced.

I thought it was because I switched to a bltouch. (my homemade extensions failed me, bought another bltouch before I realized)

Problem 2
I can also confirm that no matter what I put :
```
#define DELTA_CALIBRATION_RADIUS 
#define DELTA_PRINTABLE_RADIUS
#define X_PROBE_OFFSET_FROM_EXTRUDER -36.00     // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 10.00   // Y offset: -front +behind [the nozzle]
#define MIN_PROBE_EDGE 20
//==== Unified Bed Leveling ============================
#define MESH_INSET 15
```
Probe still probes out side of defined bed size.

@LVD-AC 
So before your code gets merged is the following the work around ?:
M502
M500
M503
G33 P1
M500
G28
G29 P1
G29 S1
G29 A
M500

G1 F60 Z0 
M211 S0 
find good offset
M851 Z -X.XX 
M211 S1 - Enable Soft Endstops
M500 - Save settings to Eeprom

G33 P1 

If M851 needs to be changed, run G33 P1 again???

LVD-AC, 2018-07-08:
That should work (around): after each M851 run G33 P1

PS: Maybe even better to run M851; G33 P1; M500 in that sequence so the new delta height derived from G33 P1 is saved to EEPROM as well.

mylife4aiurr, 2018-07-08:
Confirmed...... Thank you sir. Hope this get patched soon.

SimonWitcher, 2018-07-09:
Thanks for the help to all of you,

I was able to change my offset in another way:
Egzample for giving offset +0.1 mm 
G29 P5 // gives current offset, only to check
G29 P6 C+0.1  // adds 0.1 mm to offset  
G29 S1 // saves in slot 1
M500 // saves in EEPROM too
M501
G29 L1 // loads offset from slot 1
G29 P5 // Check that value has changed

It works 90% of times. Sometimes mesh zeroes itself for no reason I can see, or values change strangely (for egzample adds 0.0999999999 while adding 0.1)
I can work with it for a while but Im hoping for these issues to be repaired in the future updates.

closed 2018-11-05 by thinkyhead