MarlinFirmware/Marlin #11050: [bugfix-2.0.X] UBL error on G29 J: "Heating failed, system stopped! Heater_ID: 0"

AcHub, 2018-06-17:
### Description
I'm trying to set up UBL and followed the guide in the marlin documentation. The first part works well, I was able to do automated probing and have stored a mesh. 

Now that I set up my startup Gcode, I get some strange issues. 
I added ```G28```, ```M420 S1 L1``` and ```G29 J``` to load the mesh and validate it on startup, which, if I understand the documentation correctly, is how to get UBL ready prior to printing.

But whenever ```G29 J``` is executed, three points are verified and then I get a ```Error:Heating failed, system stopped! Heater_ID: 0``` and Octoprint loses the connection. I tried to run ```G29 J``` with disabled heaters as well as heated up to temperature and regarding the temperature graph in Octoprint I don't understand what would be wrong with heating, as reported temperatures were always as they should have been (0.5°C deviation max). Also I don't understand the connection between bed leveling and temperature failure.

One thing I noticed is, that when I compiled with the ```G26``` test pattern feature enabled to validate the mesh, my printer would reset multiple times in a row as soon as I had sent one ```G26``` via terminal, but also when executed from the printer menu itself. Also I had troubles with the mesh probing (```G29 P1```), which would not finish. I thought both issues at this time were related to memory/ressource shortage of my printer. So I commented out the ```G26``` feature, allowing me to at least successfully probe the mesh.

Am I operating UBL wrong, or is it ressource shortage, or is there a bug? I appreciate your help!

#### Additional Information
Running 4c8d6df
[Configuration_UBL.zip](https://github.com/MarlinFirmware/Marlin/files/2109464/Configuration_UBL.zip)

Roxy-3D, 2018-06-19:
It would be good to enable the M100 option in Configuration.h.    And then initialize the M100 (with M100 I).   And then do your G29's  and use M100 F to see how much free memory you have.

Because...   If you are running out of stack space, there is no telling what the symptoms will be.   What kind of processor do you have?

AcHub, 2018-06-19:
I will experiment with M100 and report back.
I have a Wanhao Duplicator 6, which (I believe) is basically an ATmega 2560

AcHub, 2018-06-21:
With my UBL config using 7 gridpoints I get the following with ```M100 I``` before sending the ```G29 J``` command:
```
Send: M100 I
Recv: 
Recv: __brkval : 0x0000
Recv: __bss_end : 0x1F01
Recv: start of free space : 0x1F01
Recv: Stack Pointer : 0x2106
Recv: Initializing free memory block.
Recv: 
Recv: 
Recv: 267 bytes of memory initialized.
Recv: 
Recv: ok
```

But as soon as I try to run G29 J, my printer resets:

```
Send: G29 J
Recv: echo:Vorher XYZ homen
Recv:  T:28.59 /0.00 B:26.25 /0.00 @:0 B@:0
Recv: echo:busy: processing
Printer seems to support the busy protocol, adjusting timeouts and setting busy interval accordingly
Recv:  T:28.52 /0.00 B:26.34 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:28.59 /0.00 B:26.19 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:28.52 /0.00 B:26.28 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: X:65.00 Y:115.00 Z:5.00 E:0.00 Count X:5203 Y:9205 Z:2002
Recv: start
Printer sent 'start' while already operational. External reset? Resetting line numbers to be on the safe side
Recv: echo:Marlin bugfix-2.0.x
Recv: 
Recv: echo: Last Updated: 2018-01-20 | Author: (AcHub, D6)
Recv: echo:Compiled: Jun 21 2018
Recv: echo: Free Memory: 713  PlannerBufferBytes: 1664
Recv: echo:V55 stored settings retrieved (655 bytes; crc 9345)
```

So I recompiled again without M100 support, which according to the startup terminal output gave me 70 bytes of additional memory (terminal now says: ```Recv: echo: Free Memory: 783  PlannerBufferBytes: 1664``` as compared to previous terminal log). Now I could run ```G29 J``` without a reset. So I suppose the answer is that I'm currently runnig the maximum configuration my board can handle and there are almost no free ressources left. I'm still not sure how to interpret the heating error though...

Roxy-3D, 2018-06-21:
Even with a 10x10 mesh, I have 2500 bytes of free memory on my ATmega2560 boards.    You might be able to turn off things like the Boot Screen or other things that eat up RAM memory.   

But the point is...   I think 750 bytes of free memory is too tight.     If you post your .elf file, somebody can do an objdump on it and see what is eating up so much RAM memory.   And then you can decide what you don't care about based on what the numbers say different things cost.

You might want to try turning:
```
BLOCK_BUFFER_SIZE 8     //defaults to 16
```
That will free up some memory.

AcHub, 2018-06-21:
> turn off things like the Boot Screen or other things that eat up RAM memory

I'd love to do that, but honestly as a layman I find it kind of hard to identify "memory wasters" and decide what could be disabled without sacrificing (too much) functionality or performance. 

I guess I'm not the only person struggling with this, so maybe you or somebody else who is familiar with the marlin code could make an extension to the marlin documentation on this topic. A chart could be an appropriate presentation, listing features, their memory usage (in bytes or in an impact scale from 1-3 if numbers cannot be determined), where to find them/how to disable/change, and the consequence of disabling/changing a feature.

Roxy-3D, 2018-06-21:
Well...   I hear you....      So let's try bumping BLOCK_BUFFER_SIZE down to 8.   You can find that in Configuration_adv.h    Most likely, you won't see any negative impact.    (I have the Max7219 debug LED's on my machines.   And the default display shows how deeply the Planner buffers are loaded with commands waiting for execution.   Pretty much, on Cartesian printers, that bar graph stays pegged during the entire print.)

Roxy-3D, 2018-06-21:
Oh!  I forgot to give an explanation for the heating failed that you were seeing.   That free memory is where the firmware's stack lives.   One routine calls another routine and that calls something else.   Each routine puts is needed information on the stack.   And when it calls a subordinate routine the 'return address' is placed on the stack.    If you run out of memory...   that return address is going to be invalid.    The routine that ran out of memory will return to some random location.   And in your case, it ended up in the heater failed code.

It might not have been a direct path to get there.   It could have been the local storage needed by a routine corrupted the current temperature global variables.   And the code did return to the right location.  But later...  when the temperature is checked by some other code it was found to be invalid.   And then you ended up in the heater failed code.

If you changed some of your options...   you would run out of memory in a different way.   And very possibly get different random behavior.

Roxy-3D, 2018-06-21:
1600 bytes should be fine.    What you really want to do is initialize the free memory watcher with M100 I
and then do a print.    At the end of the print....   Do a M100 F and verify you are over 750 bytes of free memory.     It really doesn't matter how much free memory you have, as long as you never run out.   And if you can get through an entire print and have many hundreds of bytes of free memory at the end...   You can relax.

closed 2018-06-24 by AcHub