MarlinFirmware/Marlin #11052: [FR] output eeprom settings to file on SD card

Phr3d13, 2018-06-17:
### Description

An option in the LCD menu to output eeprom settings to a file on the SD card. It would make updating to a new Marlin version much easier after a hard drive crash or other instance where the old configs are no longer available.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

dot-bob, 2018-06-18:
Have the output the file in gcode so you could just select the file just like a print to reload the eeprom settings.

Squid116, 2018-06-18:
@dot-bob that would be awesome, then you could configure the host to save the settings on each print (so that you can backtrack and work out why benchy #123,423 worked, but #123,424 didn't).

thinkyhead, 2018-06-20:
Meanwhile, you can copy the output of `M503 S0` into a .gcode file and save it to disk.

sjasonsmith, 2020-08-31:
Duplicate of #12119

closed 2020-08-31 by sjasonsmith