MarlinFirmware/Marlin #11055: [bugfix-2.0.x] LCD Display on RADDS show broken up text

AmiSMB, 2018-06-18:
### Description

Taking the current Marlin bugfix-2.0.x which shows green or good for the Due platform and configuring the minimum upon boot the display shows parts of text on the LCD display but not as expected.

### Steps to Reproduce

1. Attach Due to RADDS board and make sure wiring is correct
2. Get latest Marlin repo by cloning and change to bugfix-2.0.x branch
3. Open project in Atom using PlatformIO and change the board to BOARD_RADDS and enable RADDS_DISPLAY
3. Compile and upload to Due

**Expected behavior:** 
When the firmware loads I expected to see the display to show the bugfix-2.0.x boot screen and then goto a status screen.

**Actual behavior:** 
When the firmware boots all of the text is randomly split across the screen as shown in the zip archive of photos attached.

#### Additional Information
[MFW_DUE_ISSUE.zip](https://github.com/MarlinFirmware/Marlin/files/2112857/MFW_DUE_ISSUE.zip)

AletheianAlex, 2018-06-19:
Saw your post in the 32-bit thread: To make sure it is not an interference problem, try very short connector cables to check. The RADDS has an onboard level shifter that occasionally has problems with ringing under certain load conditions. If the shorter cable work, then it is a hardware issue, and you can switch longer cables if you damp the control pins with series resistors.

AmiSMB, 2018-06-19:
@AletheianAlex Thanks for your response I was actually thinking that maybe the issue from other things that I have read. I got the devices from Oozenest in the UK as a package and did not expect that the parts they gave me would not work straight away. I will try with much shorter cables and see what happens.

ejtagle, 2018-06-20:
Or increase LCD delays... I use very long cabling, but i am using shielded ribbon cable. Without the shielding, there is a lot of LCD graphics corruption