MarlinFirmware/Marlin #11056: bugfix-2.0.x G28 with bump does NOT execute "Home 2 Slow" move

lrpirlet, 2018-06-18:
### Description

I decided to switch to the bugfix-2.0.x, and to use the Visual Studio Code + PlatformIO…
I compiled the bugfix2.0.x with my config files (Mega 2560)… 
Seems that G28 does NOT execute the

### Steps to Reproduce

Compile using my config files (see https://github.com/lrpirlet/Marlin/blob/lrpv2/Marlin/Configuration.h and https://github.com/lrpirlet/Marlin/blob/lrpv2/Marlin/Configuration_adv.h)

**Expected behavior:** [What you expect to happen]
G28 does execute second bump...

**Actual behavior:** [What actually happens]
G28 does touch the switch once, retract and forget to bump again...

#### Additional Information
```
20:52:14.995 : Printer reset detected - initalizing
20:52:14.995 : start
20:52:14.995 : echo: External Reset
20:52:14.999 : Marlin bugfix-2.0.x
20:52:15.003 : echo: Last Updated: 2018-01-20 | Author: LRP
20:52:15.003 : echo:Compiled: Jun 17 2018
20:52:15.007 : echo: Free Memory: 2571  PlannerBufferBytes: 1488
20:52:15.031 : echo:V55 stored settings retrieved (655 bytes; crc 23719)
20:52:15.035 : Unified Bed Leveling System v1.01 inactive.
20:52:15.084 : Unified Bed Leveling initialized.
20:52:15.089 : UBL System reset()
20:52:15.093 : echo:  G21    ; Units in mm (mm)
20:52:15.093 : echo:  M149 C ; Units in Celsius
20:52:15.097 : echo:Filament settings: Disabled
20:52:15.097 : echo:  M200 D3.00
20:52:15.101 : echo:  M200 D0
20:52:15.101 : echo:Steps per unit:
20:52:15.105 : echo:  M92 X80.00 Y80.00 Z4000.00 E402.00
20:52:15.109 : echo:Maximum feedrates (units/s):
20:52:15.113 : echo:  M203 X300.00 Y300.00 Z3.00 E45.00
20:52:15.113 : echo:Maximum Acceleration (units/s2):
20:52:15.117 : echo:  M201 X3000 Y3000 Z100 E10000
20:52:15.126 : echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
20:52:15.126 : echo:  M204 P3000.00 R3000.00 T3000.00
20:52:15.134 : echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
20:52:15.138 : echo:  M205 B20000 S0.00 T0.00 J0.02
20:52:15.138 : echo:Home offset:
20:52:15.142 : echo:  M206 X0.00 Y0.00 Z0.00
20:52:15.146 : N1 M110*34
20:52:15.146 : echo:Unified Bed Leveling:
20:52:15.146 : echo:  M420 S0 Z10.00
20:52:15.150 : Unified Bed Leveling System v1.01 inactive.
20:52:15.204 : Active Mesh Slot: -1
20:52:15.204 : EEPROM can hold 9 meshes.
20:52:15.208 : echo:  G29 I99
20:52:15.208 : echo:  M421 I0 J0 Z0.00
20:52:15.286 : echo:  M421 I0 J1 Z0.00
20:52:15.364 : echo:  M421 I0 J2 Z0.00
20:52:15.437 : echo:  M421 I0 J3 Z0.00
20:52:15.515 : echo:  M421 I0 J4 Z0.00
20:52:15.593 : echo:  M421 I0 J5 Z0.00
20:52:15.671 : echo:  M421 I0 J6 Z0.00
20:52:15.749 : echo:  M421 I0 J7 Z0.00
20:52:15.822 : echo:  M421 I0 J8 Z0.00
20:52:15.900 : echo:  M421 I1 J0 Z0.00
20:52:15.978 : echo:  M421 I1 J1 Z0.00
20:52:16.056 : echo:  M421 I1 J2 Z0.00
20:52:16.134 : echo:  M421 I1 J3 Z0.00
20:52:16.207 : echo:  M421 I1 J4 Z0.00
20:52:16.285 : echo:  M421 I1 J5 Z0.00
20:52:16.363 : echo:  M421 I1 J6 Z0.00
20:52:16.441 : echo:  M421 I1 J7 Z0.00
[... 225 lines cut ...]
20:58:01.643 : home_offset = 0.00
20:58:01.643 : position_shift = 0.00
20:58:01.647 : soft_endstop_min = 0.00
20:58:01.651 : soft_endstop_max = 168.00
20:58:01.655 : *** Z HOMED WITH PROBE (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) ***
20:58:01.655 : > zprobe_zoffset = -4.79
20:58:01.659 : > home_offset[Z] = 0.00
20:58:01.663 : current_position=(90.00, 90.00, 4.79) :
20:58:01.663 : <<< set_axis_is_at_home(Z)
20:58:01.671 : current_position=(90.00, 90.00, 4.79) : sync_plan_position
20:58:01.675 : current_position=(90.00, 90.00, 4.79) : > AFTER set_axis_is_at_home
20:58:01.680 : current_position=(90.00, 90.00, 4.79) : set_probe_deployed
20:58:01.684 : deploy: 0
20:58:01.684 : do_probe_raise(3.00)
20:58:01.688 : >>> do_blocking_move_to(90.00, 90.00, 7.79)
20:58:01.692 : <<< do_blocking_move_to
20:58:03.125 : >>> do_blocking_move_to(90.00, 90.00, 7.79)
20:58:03.130 : <<< do_blocking_move_to
20:58:03.130 : <<< homeaxis(Z)
20:58:03.130 : <<< Z_SAFE_HOMING
20:58:03.138 : current_position=(90.00, 90.00, 7.79) : sync_plan_position
20:58:03.142 : current_position=(90.00, 90.00, 7.79) : bracket_probe_move
20:58:03.146 : X:90.00 Y:90.00 Z:7.79 E:0.00 Count X:7200 Y:7200 Z:31160
20:58:03.146 : <<< G28
```

autonumous, 2018-06-18:
you have `ENDSTOP_NOISE_FILTER` defined, possibly similar to the issue I posted, see thread https://github.com/MarlinFirmware/Marlin/issues/10986 (albeit bugfix-1.1.x, and bed moves the other way)

lrpirlet, 2018-06-19:
@autonumous 
Thanks, with your input it works as expected…

It also means that the routine enabled by ENDSTOP_NOISE_FILTER

```diff
$ git diff 225b2546c
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index 33fb9df8e..55847ee25 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -576,7 +576,7 @@
  * based on the Makerbot design, since they already include the 100nF capacitor.)
  */
 //#define ENDSTOP_NOISE_FILTER lrp changed... it needs to put capacitor...
-#define ENDSTOP_NOISE_FILTER
+//#define ENDSTOP_NOISE_FILTER

 //=============================================================================
 //============================== Movement Settings ============================
```

lrpirlet, 2018-06-19:
@thinkyhead 
My problem is gone, there is most probably an issue hidden with the setting of ENDSTOP_NOISE_FILTER...

I have not the skill (given the time I can spend) to debug this problem...

I have no problem testing…. If you want, I'll keep my branch open till closure of this bug report: 
https://github.com/lrpirlet/Marlin/tree/lrpv2

Just tell me.

ejtagle, 2018-06-19:
@lrpirlet : ENDSTOP_NOISE_FILTER works as expected. It just waits until the endstop line stops changing and, only at that time, it uses the endstop value as the accepted (denoised) value.
If disabling ENDSTOP_NOISE_FILTER makes it work, the problem is probably your endstop switch is broken or has a lose connection. So, it just gives an small pulse when the endstop is hit, instead of keeping the signal at its active state, as it should.

AnHardt, 2018-06-19:
The other way around.
If there is no second (slow) approach the endstop probably **still** reports triggered.

lrpirlet, 2018-06-19:
Please disregard hardware failure, it worked fine before I went to bugfix-2... 
It NOW works as expected after commenting `//#define ENDSTOP_NOISE_FILTER`

ejtagle, 2018-06-19:
The previous implementation had a 2 sample noise filter enabled at all times. The new approach, if you comment out (ENDSTOP_NOISE_FILTER) has no filter at all, and, if you enable it, has a filter that waits 7 samples with the same value before accepting the new endstop state... Sampling is done every 1mS ..
There is nothing wrong disabling the noise filter. In fact, that is what we encourage to do

autonumous, 2018-06-19:
As previously mentioned in another thread (#10986), on my setup, I do see a change in Z behaviour when enabling `ENDSTOP_NOISE_FILTER`.  Whether directly or indirectly,  setting that option caused homing issues on the Z axis where the bed moved the wrong direction.  

For me, it was easier to fix the issue that required me to use that option (by adding a capacitor to the Endstop)   which was only introduced around May 21st

ejtagle, 2018-06-20:
Teoretically, when you enable ``` ENDSTOP_NOISE_FILTER ``` then it should start calling update() from the temperature ISR as well.. At least, that was the way i implemented it

thinkyhead, 2018-06-20:
Only `endstops.poll` is called from `temperature.cpp`, not `endstops.update`.

ejtagle, 2018-06-20:
endstops.poll() calls update()  (otherwise, a horrible bug would be there... )

thinkyhead, 2018-06-20:
But `endstops.update` is only called when `endstops.enabled` is true. If endstops are disabled in-between moves towards the endstops, they will appear to still be triggered at the start of the second move.

ejtagle, 2018-06-20:
Humm.. Then we have a problem. Maybe it should always be called ?

ejtagle, 2018-06-20:
Humm... good question. Yes, this seems incorrect, but the periodic polling must be done periodically...

thinkyhead, 2018-06-20:
The best solution to avoid side-effects would seem to be testing the endstops at all times and maintaining the "trusted state" as a lowest-level value.

ejtagle, 2018-06-20:
I was thinking on the purpose of ENDSTOPS_ENABLED

ejtagle, 2018-06-20:
Yes, the idea is to always poll... or at least, when the endstop_poll_count != 7

ejtagle, 2018-06-20:
Please, correct me if i am wrong @thinkyhead : The purpose of ENDSTOPS_ENABLED is just to filter out the reporting of endstops being hit

thinkyhead, 2018-06-20:
Correct. The `ENDSTOPS_ENABLED` flag tells the stepper ISR that if an endstop or probe is triggered then the current move should abort.

lrpirlet, 2018-06-21:
Do you want me to test anything?

ejtagle, 2018-06-21:
Let me check your config file...

ejtagle, 2018-06-21:
@lrpirlet : Could you try increasing ```Z_HOME_BUMP_MM``` to 5 , and enable the ```ENDSTOP_NOISE_FILTER``` ?

The default 2mm bump for Z and the default speed of 4mm/s should give 0.5seconds for the bump, and that is 500 endstop samples ... unless the endstop code is disabled while performing the sampling... But i think it is not...

lrpirlet, 2018-06-22:
SORRY 

Seems that I cannot compile…

duplicate declaration in Marlin\src\module/endstops.h and then after correction more errror in TMC related files….

I cannot format the results in a readable form… 

find what I have done here https://github.com/lrpirlet/Marlin/blob/lrpv2/Marlin/WhatIHaveDone.txt

lrpirlet, 2018-06-22:
@ejtagle @thinkyhead 

Please note that I had to edit Marlin/src/module/endstops.h to remove a duplicate declaration that made PlatformIO to stop on error…

git diff shows

```diff
diff --git a/Marlin/src/module/endstops.h b/Marlin/src/module/endstops.h
index 830b1515b..7b41977bf 100644
--- a/Marlin/src/module/endstops.h
+++ b/Marlin/src/module/endstops.h
@@ -73,7 +73,7 @@ class Endstops {

     #if ENABLED(ENDSTOP_NOISE_FILTER)
       static esbits_t validated_live_state;
-      uint8_t Endstops::endstop_poll_count;
+      // uint8_t Endstops::endstop_poll_count;
       static uint8_t endstop_poll_count;    // Countdown from threshold for polling
     #endif
```

Tested as per request with ENDSTOP_NOISE_FILTER enabled and Z_HOME_BUMP_MM set to 5… 

**SAME BEHAVIOUR :** it fails to execute a second bump….

Also, a subsequent G30 fails as it does not touch the bed

```
20:40:40.943 : Printer reset detected - initalizing
20:40:40.943 : start
20:40:40.943 : echo: External Reset
20:40:40.947 : Marlin bugfix-2.0.x
20:40:40.951 : echo: Last Updated: 2018-01-20 | Author: LRP
20:40:40.951 : echo:Compiled: Jun 22 2018
20:40:40.955 : echo: Free Memory: 2571  PlannerBufferBytes: 1488
20:40:40.976 : echo:V55 stored settings retrieved (655 bytes; crc 23719)
20:40:40.980 : Unified Bed Leveling System v1.01 inactive.
20:40:41.033 : Unified Bed Leveling initialized.
20:40:41.037 : UBL System reset()
20:40:41.037 : echo:  G21    ; Units in mm (mm)
20:40:41.041 : echo:  M149 C ; Units in Celsius
20:40:41.046 : echo:Filament settings: Disabled
20:40:41.046 : echo:  M200 D3.00
20:40:41.046 : echo:  M200 D0
20:40:41.050 : echo:Steps per unit:
20:40:41.054 : echo:  M92 X80.00 Y80.00 Z4000.00 E402.00
20:40:41.054 : echo:Maximum feedrates (units/s):
20:40:41.058 : echo:  M203 X300.00 Y300.00 Z3.00 E45.00
20:40:41.062 : echo:Maximum Acceleration (units/s2):
20:40:41.066 : echo:  M201 X3000 Y3000 Z100 E10000
20:40:41.070 : echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
20:40:41.074 : echo:  M204 P3000.00 R3000.00 T3000.00
20:40:41.083 : echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
20:40:41.087 : echo:  M205 B20000 S0.00 T0.00 J0.02
20:40:41.087 : echo:Home offset:
20:40:41.091 : echo:  M206 X0.00 Y0.00 Z0.00
20:40:41.095 : echo:Unified Bed Leveling:
20:40:41.095 : echo:  M420 S0 Z10.00
20:40:41.099 : Unified Bed Leveling System v1.01 inactive.
20:40:41.150 : N1 M110*34
20:40:41.152 : Active Mesh Slot: -1
20:40:41.152 : EEPROM can hold 9 meshes.
20:40:41.152 : echo:  G29 I99
20:40:41.156 : echo:  M421 I0 J0 Z0.00
20:40:41.233 : echo:  M421 I0 J1 Z0.00
20:40:41.311 : echo:  M421 I0 J2 Z0.00
20:40:41.385 : echo:  M421 I0 J3 Z0.00
20:40:41.463 : echo:  M421 I0 J4 Z0.00
20:40:41.541 : echo:  M421 I0 J5 Z0.00
20:40:41.618 : echo:  M421 I0 J6 Z0.00
[... 267 lines cut ...]
20:42:06.132 : <<< do_blocking_move_to
20:42:06.136 : current_position=(10.00, 10.00, 7.79) : >>> run_z_probe
20:42:06.143 : current_position=(10.00, 10.00, 7.79) : >>> do_probe_move
20:42:06.143 : >>> do_blocking_move_to(10.00, 10.00, 2.79)
20:42:06.147 : <<< do_blocking_move_to
20:42:08.785 : current_position=(10.00, 10.00, 2.79) : sync_plan_position
20:42:08.789 : current_position=(10.00, 10.00, 2.79) : <<< do_probe_move
20:42:08.794 : SLOW Probe fail!
20:42:08.798 : current_position=(10.00, 10.00, 2.79) : <<< run_z_probe
20:42:08.802 : current_position=(10.00, 10.00, 2.79) : set_probe_deployed
20:42:08.802 : deploy: 0
20:42:08.806 : do_probe_raise(3.00)
20:42:08.810 : >>> do_blocking_move_to(10.00, 10.00, 7.79)
20:42:08.810 : <<< do_blocking_move_to
20:42:10.916 : >>> do_blocking_move_to(10.00, 10.00, 7.79)
20:42:10.916 : <<< do_blocking_move_to
20:42:10.920 : Error:Probing failed
20:42:10.921 : <<< probe_pt
20:42:10.924 : current_position=(10.00, 10.00, 7.79) : bracket_probe_move
20:42:10.932 : X:10.00 Y:10.00 Z:7.79 E:0.00 Count X:800 Y:800 Z:31160
20:42:10.932 : ok

```

[WhatIHaveDone.txt](https://github.com/MarlinFirmware/Marlin/files/2128866/WhatIHaveDone.txt)

lrpirlet, 2018-06-22:
@thinkyhead @ejtagle 
Please note that I see the problem on **_ALL 3 AXIS: X, Y and Z_**...

ejtagle, 2018-06-22:
@lrpirlet Could you try latest bugfix2.0.x with the following endstops.cpp file ?

[endstops.zip](https://github.com/MarlinFirmware/Marlin/files/2129224/endstops.zip)

ejtagle, 2018-06-22:
@thinkyhead : I could be wrong here, but i think the already merged PR #11066 is not enough. If you read endstops::update(), you will notice the endstop state is ONLY read if steppers.axis_is_moving() returns true. That will only return true if the motors are moving, Something that will not happen when the move stops due to an endstop being hit. 
That causes all endstops, after 7 samples, to be reported as open...

lrpirlet, 2018-06-23:
@ejtagle @thinkyhead

Thanks ejtagle, with the new endstops.cpp, it does work as expected…

Note that I set back the Z_HOME_BUMP_MM to 2  and it still continue working as expected…
Note that G28, G30 and M48 now works fine…

I would have closed this entry… do you want me to raise a PR for the duplicated declaration in endstops.h? (see 4 entries above) 

```
09:35:48.507 : N15 G28*39
09:35:50.515 : echo:busy: processing
09:35:52.514 : echo:busy: processing
09:35:54.534 : echo:busy: processing
09:35:56.532 : echo:busy: processing
09:35:58.535 : echo:busy: processing
09:36:00.535 : echo:busy: processing
09:36:02.538 : echo:busy: processing
09:36:04.811 : X:90.00 Y:90.00 Z:7.79 E:0.00 Count X:7200 Y:7200 Z:31160
09:36:04.812 : ok
09:36:09.586 : N16 G30*45
09:36:11.591 : echo:busy: processing
09:36:12.946 : Bed X: 90.00 Y: 90.00 Z: 0.04
09:36:12.951 : X:90.00 Y:90.00 Z:7.79 E:0.00 Count X:7200 Y:7200 Z:31160
09:36:12.954 : ok
09:36:24.371 : N17 M48 V4*107
09:36:24.375 : M48 Z-Probe Repeatability Test
09:36:24.379 : Positioning the probe...
09:36:26.374 : echo:busy: processing
09:36:27.104 : Bed X: 90.000 Y: 90.000 Z: 0.013
09:36:28.378 : echo:busy: processing
09:36:29.037 : 1 of 10: z: 0.010 mean: 0.0095 sigma: 0.000000 min: 0.010 max: 0.010 range: 0.000
09:36:30.380 : echo:busy: processing
09:36:30.975 : 2 of 10: z: -0.004 mean: 0.0025 sigma: 0.007000 min: -0.004 max: 0.010 range: 0.014
09:36:32.380 : echo:busy: processing
09:36:32.908 : 3 of 10: z: -0.006 mean: -0.0004 sigma: 0.007049 min: -0.006 max: 0.010 range: 0.016
09:36:34.383 : echo:busy: processing
09:36:34.838 : 4 of 10: z: -0.002 mean: -0.0008 sigma: 0.006143 min: -0.006 max: 0.010 range: 0.016
09:36:36.382 : echo:busy: processing
09:36:36.780 : 5 of 10: z: -0.016 mean: -0.0038 sigma: 0.008191 min: -0.016 max: 0.010 range: 0.026
09:36:38.385 : echo:busy: processing
09:36:38.725 : 6 of 10: z: -0.016 mean: -0.0058 sigma: 0.008694 min: -0.016 max: 0.010 range: 0.026
09:36:40.384 : echo:busy: processing
09:36:40.658 : 7 of 10: z: -0.016 mean: -0.0073 sigma: 0.008836 min: -0.016 max: 0.010 range: 0.026
09:36:42.387 : echo:busy: processing
09:36:42.592 : 8 of 10: z: -0.020 mean: -0.0089 sigma: 0.009268 min: -0.020 max: 0.010 range: 0.030
09:36:44.418 : echo:busy: processing
09:36:44.533 : 9 of 10: z: -0.018 mean: -0.0099 sigma: 0.009169 min: -0.020 max: 0.010 range: 0.030
09:36:46.418 : echo:busy: processing
09:36:46.463 : 10 of 10: z: -0.014 mean: -0.0103 sigma: 0.008775 min: -0.020 max: 0.010 range: 0.030
09:36:47.237 : Finished!
09:36:47.241 : Mean: -0.010275 Min: -0.020 Max: 0.010 Range: 0.030
09:36:47.245 : Standard Deviation: 0.008775
09:36:47.249 : X:90.00 Y:90.00 Z:7.79 E:0.00 Count X:7200 Y:7200 Z:31160
09:36:47.249 : ok
```

ejtagle, 2018-06-23:
@lrpirlet: Donñt worry, i am raising a PR with both fixes (and an extra one to the Planner...) in a few hours... Thanks for confirming the fix works (i was not sure, as in my case the fix was not required)

thinkyhead, 2018-06-30:
Seems we're still having some endstop issues. Please test the latest code, doing multiple `G28` commands, to see if homing is behaving the way it should 100% of the time.

lrpirlet, 2018-07-02:
@thinkyhead 
I have rebased lrpv2 to current mainstream… (lrpv2 is now "hanging from" commit 8299ac121)
I have run 12+ iteration of G28... NO PROBLEM at all (statistically that leaves a very low probability of an intermittent problem, yet still possible.) Please note that I do NOT use interrupt to detect endstop change, that I disable auto reporting of temperature and stop sending busy message while executing G28 (I have no added precision with interrupt but I risk my bed if interrupt is missed, I feel that Repetier is overdoing it by requesting a temperature reading every second, I see no use for those messages clutering the console)

Log is https://github.com/lrpirlet/Marlin/blob/lrpv2/Marlin/WhatIHaveDone.txt

sorry no time for text formating…. I need to drop now… tell me if you want more testing, I'll try my best.

thinkyhead, 2018-07-02:
`AUTO_REPORT_TEMPERATURES` is a good option to quiet down the host. Repetier should notice that it's enabled and stop sending `M105` commands. Then `M155` can also be used to set a longer interval.

boelle, 2019-02-20:
@lrpirlet has problem gone away in latest bugfix 2.0?

closed 2019-02-20 by thinkyhead