MarlinFirmware/Marlin #11059: [Question] How to speed up using Lin. Adv.

Lyr3x, 2018-06-19:
Using lin. adv. 1.5 gives me awesome print results and works fine except of the rattle problem.

I realised that the printing speed with lin. adv. turned on is much slower than without. Are there any parameters, like jerk or acceleration, which i should change to get higher speeds?

Current Settings:
`
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 100, 100, 400, 102.00 }
#define DEFAULT_MAX_FEEDRATE          { 300, 300, 5, 120 }
#define DEFAULT_MAX_ACCELERATION      { 3000, 3000, 100, 10000 }
#define DEFAULT_ACCELERATION          1000    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  1000    // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   1000    // X, Y, Z acceleration for travel (non printing) moves
#define DEFAULT_XJERK                 10.0
#define DEFAULT_YJERK                 10.0
#define DEFAULT_ZJERK                  0.3
#define DEFAULT_EJERK                  5.0
`

Sebastianv650, 2018-06-21:
It's quite likely your ejerk setting of 2.5mm/s is limiting the acceleration for print moves. Pressure adjustments need an extruder speed offset, and the offset can't be higher than the extruders allowed max. jerk or it might loose steps.
Therefore inside the planner, LA checks for the needed speed offset for the calculated move (which depends on the acceleration) and if it's higher than allowed it will reduce the acceleration to the max. possible value with ejerk not violated.

boelle, 2019-02-20:
@Lyr3x did you look at it?

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-14 by thinkyhead