MarlinFirmware/Marlin #11060: M42 gcode not executing properly

godorrobogod, 2018-06-19:
I am trying to use my RAMPS 1.4 board with an INKSHIELD(http://nicholasclewis.com/projects/inkshield/)/arduino 
 mega to make an open flat-bed printer. Ideally, I would like to use the ramps board to set a pin high at the start of a new line. To do this, I have been trying to use the gcode comman M42 P11 S255/0, however I am getting some unexpected results.

When I type M42 P11 S255 into Repetier, the pin goes high. Additionally, when I type M42 P11 S0 into Repetier, the pin goes low. However, when I use these gcodes in a file, the D11 pin goes high and then immediately low and will not go high again for the course of the program. Furthermore, the pin is not going low at the correct point in the program, just immediately. If I do not include a M42 P11 S0 code, the pin stays high.

Example:
M42 P11 S255 // at this point in the execution, the pin goes on momentarily and then stays off
G01 X20 F2000
G01 X0 Y3.0 F2000
M42 P11 S0 

Am I using an improper pin for this purpose? Is there something I am not understanding about how the firmware is reading the gcode?

AnHardt, 2018-06-19:
Try 'M400' before all the 'M42' to synchronize the planners buffer with the pin.

godorrobogod, 2018-06-19:
That worked. Thanks.

closed 2018-06-20 by thinkyhead