MarlinFirmware/Marlin #11063: min/max redefinitions bug

grownseed, 2018-06-19:
### Description

Latest Marlin bugfix-2.0.x:
When trying to include certain libraries for ESP, compiling returns errors such as:
```
Marlin/src/module/stepper.h:150:107: error: no matching function for call to 'max(long int, long unsigned int)'
```

It appears Marlin redefines its own min/max, which clashes with the std::min/max from the ESP libs.

There appears to be more details in this comment: https://github.com/esp8266/Arduino/issues/398#issuecomment-356034907

I have very little experience with this and therefore have a hard time understanding what needs to be done. That said if anybody has any suggestions I'd be happy to help with a PR (if code changes are indeed required).

thinkyhead, 2018-06-20:
Try replacing `max` with `MAX`.

grownseed, 2018-06-20:
@thinkyhead had to change all the `max(...)` to `MAX(...)` in both `module/stepper.h` and `inc/Conditionals_post.h`, and I can compile with these changes :+1:

thinkyhead, 2018-06-20:
Good news. I'll do that patch asap.

boelle, 2019-02-20:
@grownseed is problem gone away with latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead