MarlinFirmware/Marlin #11065: compile error on v2 bugfix

ruggb, 2018-06-19:
### Description

compile error
fork/exec C:\Program Files (x86)\Arduino\hardware\tools\avr**/**bin**/**avr-gcc.exe: The filename or extension is too long.

### Steps to Reproduce

I modified my config files for v2 bugfix  a65d15c and compiled.
no other issues - note the **/** instead of **\** in the path. Where did that come from?

I had no problems compiling v1.1 bugfix 7cdaf69.
both compiled with same version of arduino on same computer.
is it a problem with Arduino or Marlin?

thinkyhead, 2018-06-19:
You'll need to use Arduino IDE 1.9 (now in beta) to compile Marlin 2.0.

closed 2018-06-19 by thinkyhead

ruggb, 2018-06-19:
it compiles without that error with an unmodified configuration.h
what about my mods doesn't it like?

ruggb, 2018-06-20:
OK, but the error description makes no sense. OH well I guess you had to be there!!!!!

I changed something in RAMPS.h and added a comment to another line
It errors with that comment
`#ifndef RAMPS_D9_PIN   
  #define RAMPS_D9_PIN      9  //** model fan on CCBOT
#endif
`
If I place the comment on the line above, it is fine. Does not matter what I put in there. Even just // will make it error. What is that all about?

thinkyhead, 2018-06-20:
That is another compiler bug.