MarlinFirmware/Marlin #11069: PLatform IO compilation error

J-charles-C, 2018-06-20:
I,

With the new version of Platform IO, the P3p fix is down. I have the same problem.

https://github.com/MarlinFirmware/Marlin/issues/11008#thread-subscription-status

Have you a new fix ?

p3p, 2018-06-20:
You need to supply more information as the template requests,

I made sure my PlatformIO environment was up to date and built and flashed Marlin bugfix-2.0.x without any problems, so it doesn't seem that issue with binary linking has reappeared.

J-charles-C, 2018-06-20:
I have the same Problem as CageFox.

Compilation is an success but, after flash, boot is incomplete and i lost usb link.

My marlin is the same between one mouth.Nothing changed.  If i restore a saved one, i have the same problem.

Compilation is an success but, after flash, boot is incomplete and i lost usb link.

best

p3p, 2018-06-20:
The update solved CageFoxs problem, and the others that had that issue. I can't help more until you at least supply your configs as was requested by the issue template.

J-charles-C, 2018-06-20:
Steps to Reproduce

git clone latest marlin 2.0 branch

Update platform io and packages:
platformio upgrade (last of 18-06-20)

platformio update

...

Build the firmware, upload to board and flash it usually way

Expected behavior: [What you expect to happen]

Board connected to mac visible in device manager and have com port and flash card as removable drive

Actual behavior: [What actually happens]

Board hangs almost immediately while flashing, after reset all leds on and no communication possible

Additional Information

I have a Rearm with ramps.

best

p3p, 2018-06-20:
I can't reproduce this on either the MKS SBase or Re-ARM LPC1768 boards, perhaps you are having the caching issue with PlatformIO that others have had, if you are testing with the default config make sure you are enabling the USB serial port.

J-charles-C, 2018-06-21:
I,
USB serial port is ok with hold firmware.

I have clean my platform IO and i have new error message in terminal :

Processing LPC1768 (platform: nxplpc; board: lpc1768)
--------------------------------------------------------------------------------
Warning! Library `{'requirements': None, 'name': 'CMSIS-LPC1768'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'CMSIS-LPC1768'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: NXP LPC > NXP mbed LPC1768
SYSTEM: LPC1768 100MHz 64KB RAM (512KB Flash)
DEBUG: CURRENT(custom) ON-BOARD(cmsis-dap) EXTERNAL(blackmagic, jlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(off) COMPATIBILITY(soft)
Collected 3 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4
|-- <TMC2130Stepper> 2.4.0

Unable to find destination disk.  File must be copied manually.

Compiling .pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL_spi.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL_timers.o
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_LPC1768/HAL.h:65:0,
from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:28,
from Marlin/src/HAL/HAL_LPC1768/HAL.cpp:23:
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_LPC1768/fastio.h:38:10: fatal error: LPC17xx.h: No such file or directory

*****************************************************************
* Looking for LPC17xx.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:LPC17xx.h"
* Web  > https://platformio.org/lib/search?query=header:LPC17xx.h
*
*****************************************************************

#include <LPC17xx.h>
^~~~~~~~~~~
compilation terminated.
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HardwareSerial.o
*** [.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL.o] Error 1
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_LPC1768/HAL.h:65:0,
from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:28,
from Marlin/src/HAL/HAL_LPC1768/HAL_spi.cpp:51:
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_LPC1768/fastio.h:38:10: fatal error: LPC17xx.h: No such file or directory

*****************************************************************
* Looking for LPC17xx.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:LPC17xx.h"
* Web  > https://platformio.org/lib/search?query=header:LPC17xx.h
*
*****************************************************************

#include <LPC17xx.h>
^~~~~~~~~~~
compilation terminated.
*** [.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL_spi.o] Error 1
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_LPC1768/HAL.h:65:0,
from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:28,
from Marlin/src/HAL/HAL_LPC1768/HAL_timers.cpp:31:
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_LPC1768/fastio.h:38:10: fatal error: LPC17xx.h: No such file or directory

*****************************************************************
* Looking for LPC17xx.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:LPC17xx.h"
* Web  > https://platformio.org/lib/search?query=header:LPC17xx.h
*
*****************************************************************

#include <LPC17xx.h>
^~~~~~~~~~~
compilation terminated.
*** [.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL_timers.o] Error 1
In file included from Marlin/src/HAL/HAL_LPC1768/HardwareSerial.cpp:25:0:
Marlin/src/HAL/HAL_LPC1768/HardwareSerial.h:33:10: fatal error: Stream.h: No such file or directory

****************************************************************
* Looking for Stream.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Stream.h"
* Web  > https://platformio.org/lib/search?query=header:Stream.h
*
****************************************************************

#include <Stream.h>
^~~~~~~~~~
compilation terminated.
*** [.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HardwareSerial.o] Error 1
============================================================================================================================================= [ERROR] Took 1.54 seconds =============================================================================================================================================

===================================================================================================================================================== [SUMMARY] =====================================================================================================================================================
Environment megaatmega2560              [SKIP]
Environment megaatmega1280              [SKIP]
Environment at90USB1286_CDC             [SKIP]
Environment at90USB1286_DFU             [SKIP]
Environment DUE                         [SKIP]
Environment DUE_USB                     [SKIP]
Environment DUE_debug                   [SKIP]
Environment LPC1768                     [ERROR]
Environment melzi                       [SKIP]
Environment melzi_optiboot              [SKIP]
Environment rambo                       [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment sanguino_atmega1284p        [SKIP]
Environment STM32F1                     [SKIP]
Environment STM32F4                     [SKIP]
Environment teensy20                    [SKIP]
Environment teensy35                    [SKIP]
Environment malyanm200                  [SKIP]
============================================================================================================================================= [ERROR] Took 1.54 seconds =======

p3p, 2018-06-21:
> USB serial port is ok with hold firmware.

I was just warning you that the default selected port had changed if you were using the default configs.

This last error makes no sense, the CMSIS-LPC1768 isn't even being compiled .. that shouldn't be able to happen and I can't reproduce it.

J-charles-C, 2018-06-22:
I,
Thanks p3p. Sorry for my very bad english. (I'm French)

With the Latest commit 3fa9aeb. I change only the name of board. I have this error :

Compiling .pioenvs/LPC1768/libc38/TMC2130Stepper_ID1493/source/TMC2130Stepper_PWMCONF.o
.piolibdeps/TMC2130Stepper_ID1493/src/source/TMC2130Stepper.cpp:193:10: error: prototype for 'uint32_t TMC2130Stepper::TPOWERDOWN()' does not match any in class 'TMC2130Stepper'
uint32_t TMC2130Stepper::TPOWERDOWN() { return TPOWERDOWN_sr; }
^~~~~~~~~~~~~~
In file included from .piolibdeps/TMC2130Stepper_ID1493/src/source/TMC2130Stepper.cpp:1:0:
.piolibdeps/TMC2130Stepper_ID1493/src/TMC2130Stepper.h:91:8: error: candidates are: void TMC2130Stepper::TPOWERDOWN(uint8_t)
void TPOWERDOWN(     uint8_t input);
^~~~~~~~~~
.piolibdeps/TMC2130Stepper_ID1493/src/TMC2130Stepper.h:90:11: error:                 uint8_t TMC2130Stepper::TPOWERDOWN()
uint8_t TPOWERDOWN();
^~~~~~~~~~
.piolibdeps/TMC2130Stepper_ID1493/src/source/TMC2130Stepper.cpp:194:6: error: prototype for 'void TMC2130Stepper::TPOWERDOWN(uint32_t)' does not match any in class 'TMC2130Stepper'
void TMC2130Stepper::TPOWERDOWN(uint32_t input) {
^~~~~~~~~~~~~~
In file included from .piolibdeps/TMC2130Stepper_ID1493/src/source/TMC2130Stepper.cpp:1:0:
.piolibdeps/TMC2130Stepper_ID1493/src/TMC2130Stepper.h:91:8: error: candidates are: void TMC2130Stepper::TPOWERDOWN(uint8_t)
void TPOWERDOWN(     uint8_t input);
^~~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pioenvs/LPC1768/libc38/TMC2130Stepper_ID1493/source/TMC2130Stepper.o] Error 1

J-charles-C, 2018-06-22:
P3p,

Great !!!!! Many thanks for your help.

It's Ok, with commit 3fa9aeb.

Have a good day

cordially

closed 2018-06-22 by J-charles-C