MarlinFirmware/Marlin #11070: Setting ENCODER_PULSES_PER_STEP to >1 causes LCD Distortion

ex-egll, 2018-06-20:
I am using 2.0 bugfix downloaded on June 19th on a RUMBA board with the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

### Description

If I set  ENCODER_PULSES_PER_STEP 2  (or 3 or 4...) I get some flashing interference (about 1Hz) on the LCD, it appears that some pixels from characters or graphics are shifting left by approx 16 pixels

If I set ENCODER_PULSES_PER_STEP 1 the LCD is perfectly stable.

### Steps to Reproduce

In configuration.h set set  ENCODER_PULSES_PER_STEP 2

See above

**Expected behavior:** [What you expect to happen]
A stable LCD screen regardless of the ENCODER_PULSES_PER_STEP setting

**Actual behavior:** [What actually happens]
See above 

This problem does not exist in the latest 1.1.8

ex-egll, 2018-06-21:
A little further investigation shows that this issue started with version 1.1.8. I also checked compiling 1.1.8 with both Arduino 1.8.5 and 1.9Beta and the issue exists with both compiles.

thinkyhead, 2018-06-21:
Do you require `ENCODER_PULSES_PER_STEP 2` for your setup?

ex-egll, 2018-06-21:
Hi, yes I do otherwise the encoder will cause will produce 2 "movements" for every click, unless I manage to stop the encoder mid click.

ManuelMcLure, 2018-06-21:
Would setting  `ENCODER_PULSES_PER_STEP` to 1 and `ENCODER_STEPS_PER_MENU_ITEM` to 2 perhaps help?

ex-egll, 2018-06-21:
Hi, by ENCODER_PULSES_PER_STEP to 1 and ENCODER_STEPS_PER_MENU_ITEM to 2, the LCD is no longer distorted, I can move through menu items normally, but when I go to change a value e.g. Move X Axis it changes by 2 units instead of 1.

thinkyhead, 2018-06-22:
Can you post a picture or video of this distortion? How it appears might be a clue to the cause.

ex-egll, 2018-06-22:
Hi, I have zipped 2 videos, the configuration files and the hex file. 
Video 1 is with ENCODER_PULSES_PER_STEP set to 1 and shows a clear LCD
Video 2 is with ENCODER_PULSES_PER_STEP set to 2 and shows the interference. 
This interference is on all screens and occurs mainly on the right half of the screen. 
I will be out of town for the week end, but should be within e-Mail contact.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2128014/Marlin.zip)

marcio-ao, 2018-06-29:
@ex-egll: If I were to venture a guess, I would think the delays in `src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp` are barely long enough for your situation and using a different compiler was sufficient to cause enough of a timing difference to have an effect. I bet if you increase he delays in that file, you can get it to work no matter the compiler.

ex-egll, 2018-06-29:
Thanks for the suggestion, being as I am not too advanced in this programming game there seems to be a whole slew of delays depending on the hardware in use. I am using the rumba board with the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER. Could you point me at the lines that may need changing (and possible values) and I'll plug them in and try them.

thinkyhead, 2018-06-29:
@ex-egll —
```cpp
// If you want you can define your own set of delays in Configuration.h
//#define ST7920_DELAY_1 DELAY_NS(0)
//#define ST7920_DELAY_2 DELAY_NS(0)
//#define ST7920_DELAY_3 DELAY_NS(0)
```
The defaults for RUMBA are 0, 0, 63.

closed 2018-06-29 by thinkyhead