MarlinFirmware/Marlin #11071: Printer halted thermal runaway

tkardaridis, 2018-06-20:
I have Anet A6 printer with Marlin 1.1.8
Yesterday i installed a 100A DC Digital Multifunction Power Meter Energy Monitor Module With 50A Shunt. 
(https://www.banggood.com/50A-DC-Digital-Multifunction-Power-Meter-Energy-Monitor-Module-Voltmeter-Ammeter-With-Shunt-p-996110.html?rmmds=myorder&stayold=1&cur_warehouse=CN) 
After the installation, i tried to print and after a while the printer is stopping with the error "Printer halted thermal runaway". I thought that there might be a compatibility problem with the Power Meter and i removed it. After that the printer does not stop. Any suggestions because i would like to add the Power Meter. 

Regards

Thomas

cnering, 2018-06-21:
What was your temperature graph like when it went into Thermal Runaway?  Was the nozzle at the correct temperature or was it offset and not able to catch up in time?  Your shunt says it only drops 75mV so electrically it shouldn't be too much of a burden (assuming it's specs are accurate and you installed it properly, etc), but if you were already running close to your printer's maximum power capabilities this may have been just enough to cause it to stay outside of the acceptable envelope just long enough for thermal runaway protection to trip.

Did the thermal runaway shutdown happen after a commanded temperature change?  I used to have thermal runaways when I changed temperature mid-print, I ended up having to increase the thermal runaway window to fix the issue.  Also what temperature are you printing at, and how close is that temperature to your printer's max temp?

AnHardt, 2018-06-21:
Neither 2 nor 7 is good given.
How about a adjustable amount. Add as much filtering as you need. 2 to 9?

MasterPIC, 2018-06-22:
I'd suggest to give the power meter another try: I think the issue would have occurred also without it...
If so, as already suggested, you should relax the thermal runaway detection.

tkardaridis, 2018-06-22:
The extruder temperature was at 210°C and the bed temperature was at 80°C. The printers max temperatures are about 100°C for the bed and 260°C for the extruder. I will give to the power meter another try as MasterPIC mention, but the issue was not happening without it unfortunately. I have to ask for guidenance on how to relax  the thermal runaway detection because i realy don't know how to do it safely. I would like to mention that today i will change and the stock PSU with a MeanWell RSP-320-12 which i believe is better than the stock one with more A.

MasterPIC, 2018-06-22:
As you can see in configuration_adv.h file, to relax thermal runaway detection you can  increase THERMAL_PROTECTION_HYSTERESIS and/or THERMAL_PROTECTION_PERIOD. If the thermal runaway is referred to bed temperature then you can increase THERMAL_PROTECTION_BED_PERIOD and/or THERMAL_PROTECTION_BED_HYSTERESIS.

You can read that

> Once the temperature reaches the target, start observing. If the temperature stays too far below the target (hysteresis) for too long (period), the firmware will halt the machine as a safety precaution.

So you can give the printer more time to reach target temperature by increasing period value; or you can allow the printer to keep an higher temperature error by increasing the hysteresis value..

Increase these values gradually till satisfied with no false positives, to keep a working thermal protection.

You can see different values for the hotbed and because of its higher thermal capacity you've a more stable temperature so that you can expect to give the printer more time to reach hotbed target temperature and less hysteresis.

MasterPIC, 2018-06-22:
If the issue is referred to the extruder temperature then I'd suggest to check the cooling fan isn't cooling the extruder.
For this purpose you can set your typical nozzle temperature, wait till it's stable and then power on the fan at its max speed to see if a too high temperature drop occurs. In this case I'd suggest to improve extruder isolation, if possible.

mattbehnken, 2018-12-16:
Realize that "thermal runaway" is simply "taking too long to heat up". Did you install the power meter before the psu or after?

boelle, 2019-02-20:
@tkardaridis is this still an issue when using latest bugfix 2.0?

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-14 by thinkyhead