MarlinFirmware/Marlin #11072: Problem with SD power recovery

programer786, 2018-06-20:
Hello,

First of all, thank you very much for implementing the power recovery code from the CR-10S into marlin.
I can understand that you you just implemented it a short while ago and I don't want to complain at all, but I think it isn't working correctly somehow. I've been testing it in different stages of different prints on my anet A8 and it recognizes power outages and asks whether I want to resume the print, which is already awesome.

It also recalls the position it should be at, since the LCD is displaying the right position, but the print never continues.

Apparently the code is supposed to higher the nozzle, then to home X and Y a few times and then to come back to the old position. After that, it should lower itself and continue printing. However, the printer doesn't lower the nozzle after homing a few times on my machine, it just goes to the correct X and Y position and then stays a few millimeters over the print. The LCD reads "Continuing..." but it seems like it doesn't continue, no matter how much time I give it.
Could you please fix this issue ?

I am using 1.1.x bugfix with the example configurations for the anet A8, I've only uncommented the power recovery function in configuration_adv.h

5a2v0, 2018-07-02:
I have similar issue.
Printer: Geeetech Prusa I3 Pro B
Marlin version: 2.0 Bugfix
Configuration.h & Configuration_advanced.h are attached to this post.

When I interrupt the printer (reset it by diconnecting from Repetier Host or switching off the psu) while it's printing, at next boot it show me correctly the resume menu. When I select to resume the print, it warm up the hot bed and the extruder , then raise Z (, go home with X & Y two time and then fall down z. it never go back over the object but it say "continuing print" on the display. It remain parked at x0 y0.

I uploaded a video on youtube to show you better the situation: [LINK](https://youtu.be/-6ueMff6tLs)

[config.zip](https://github.com/MarlinFirmware/Marlin/files/2156476/config.zip)

AnHardt, 2018-07-02:
Do you print from USB/serial or from SD? Can only work when printing from SD!

thinkyhead, 2018-07-02:
The feature was recently patched to fix an issue with the data saved to the BIN file. Please re-test the latest code whenever you have a chance. It can only be used with SD printing.

5a2v0, 2018-07-02:
I tested the feature while I'm printing from sd.
I will test with latest code as soon as I can and I will report the result.

thanks in advance for the tip

5a2v0, 2018-07-07:
I'm trying to upload the latest 2.0 marlin version to my geeetech i3 pro b but i can't do with arduino IDE 1.9beta because a lot of error (see here).
Then I uploaded with IDE 1.8.5 and it goes everything ok, but:

1) I Started a print from sd
2) I unplugged power cable after a few of layer
3) So replugged in the power cable and the printer notify me that there is a print can I resume 
4) I choose to resume it
5) The printer doesn't warm up hotbed and hotend, but move first to home all axis!!! (the temperature is not yet the right one so this is a problem because a cold hotend can be attached to the incomplete print and move or to broke the print!)
6) After homing the printer warm up bed and hotend
7) The printer doesn't resume but start over the print from the first layer, and the hot end crashed with the incomplete printed object

programer786, 2018-07-08:
@5a2v0 Have you tried the bugfix-1.1.x please ?
Do you have the bltouch or no probe version of the pro b please ?
Has there been any modification to the code since the last 5 days ?

I don't have any geeetech to test it but that code should work, just choose the bltouch or no probe version:
[Geetech.zip](https://github.com/MarlinFirmware/Marlin/files/2173386/Geetech.zip)

5a2v0, 2018-07-08:
I have no probe actually. I will try 1.1.x and I Will let you know.

maybe is better to run another test with 2.0.x before switching to 1.1.x

walidsmart, 2018-11-05:
hi all
i used marlin 1.9 but i have problems with power loss 

The First Resume Survive But Any Resume After It Z axis Shifted 2 , 4, 6 mm each time
Thank You For Contributing ^^

thinkyhead, 2018-11-05:
I've heard about that in another thread here someplace. `POWER_LOSS_RECOVERY` will be getting an overhaul soon. I had hoped to work on it this weekend, but attending to issues took up all the time.

walidsmart, 2018-11-13:
hello all
any one have new news

thinkyhead, 2018-11-14:
News will be posted in this issue when there is news. Please don't nag the maintainers.

thinkyhead, 2018-11-16:
See #12440 for an updated and optimized version of Power-Loss Recovery. Needs testing. Enable the `DEBUG_POWER_LOSS_RECOVERY` option in `power_loss_recovery.h` to get extra output to the serial console when recovering. (You should be able to connect a host when the "Resume"/"Cancel" menu appears and it will go back to recovery mode when it reboots.)

engineeren, 2018-12-31:
Hi! I have a different problem with power loss recovery. When I start power loss resuming, X&Y axis go to home and LCD shows X&Y positions is "0". But for Z-axis it shows "???". Print resuming works but z-axis don't raise for next layers. I'm waiting for your helps.

boelle, 2019-02-20:
@programer786 problem solved?

programer786, 2019-02-20:
@boelle Hello,
Yes, the problem is solved thankfully.
I have discussed about the problem in #2085 and the marlin team has helped me a lot.
But after trying out this feature and seeing its inaccuracy, I do now understand that some extra hardware will be necessary. I thought about making some very simple and cheap hardware for this that will work like an endstop to recognize power shortages (it could be connected to X max for example) but I don't have the time right now to make it.

boelle, 2019-02-21:
@programer786 so maybe click the green close button below?

walidsmart, 2019-02-21:
@boelle  so how to solved thes problem plasae

boelle, 2019-02-21:
use an UPS unit i guess, i dont use the feature

programer786, 2019-02-21:
@walidsmart the problem got solved in the bugfix branch, please download a fresh copy and uncomment the auto recovery line

boelle, 2019-02-21:
@walidsmart and make sure to use bugfix 2.0

@programer786 could you click the green close button below? people can still write even if closed, but there is no need to have it open anymore

programer786, 2019-02-21:
@boelle Oh yeah sure. I'm sorry, I'm not really used to posting on github.
By the way, I wouldn't really recommend the UPS for a place where the power goes for a long time like my home. I heard people are connecting car batteries since their tension is at around 14V, which 12V electronics can handle

closed 2019-02-21 by programer786

boelle, 2019-02-21:
the ups will output mains voltage to the psu.... and i guess the psu converts it to what voltage the printer needs

og course the ups needs to have enough to run the printer for say at least 6 hours

patelmontu, 2019-05-31:
when I enable power loss recover mode in marlin 1.1.9 after starting print it just shown me power loss recovery on display so i click on resume print it just restart the main board.
I have Einsy Rambo 1.1b main board.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3241161/Marlin.zip)