MarlinFirmware/Marlin #11073: Trinamic driver SPI control

pgScorpio, 2018-06-21:
The current SPI support for the trinamic drivers require all tmc chips to be connected to miso and mosi of the processor and require a separate chip select pin for each channel.
For many printer configurations however this is a pain in the ass because there are not so many pins available/accessible (many times forcing you to use the endstop pins for cs and use tmc stallguard as endstop, which has it's own problems).

The TMC chips however can be used in cascade where miso/mosi are connected from chip to chip and they all share one common chip select.

It should be possible to use this mode in marlin, which would simplify the interconnection and reduce the number of used IO pins.

[TMC-SPI.pdf](https://github.com/MarlinFirmware/Marlin/files/2124099/TMC-SPI.pdf)

thinkyhead, 2018-06-22:
I don't think this is currently supported by the TMC2130Stepper library that we use. You should post your suggestion to https://github.com/teemuatlut/TMC2130Stepper/issues

Grogyan, 2018-06-25:
Perhaps a shift register such as the 74HC595 or 74HC164 is a better solution?