MarlinFirmware/Marlin #11077: [bugfix-2.0.X] no z raise after G29 J

AcHub, 2018-06-21:
After ```G29 J```  is done with probing the bed, my nozzle doesn't raise before it moves. I have set ```#define Z_AFTER_PROBING 5``` but it doesn't seem to bother. Is it not used by G29?

AcHub, 2018-06-21:
It is only an issue with G29 J (3 Points). When setting G29 J 2 or G29 J 4, the nozzle is raised before the last move to the center.
With G29 J, the nozzle is not raised at all at the end of probing, before it moves to the center of the bed. It only raises between the probing points.

I just tested with your latest commit and now it raises at the end of G29 J, so this is a success, thanks.
Actually it seems like it is even raising twice in a row now. But I can't tell if the first time is the distance set as ```Z_CLEARANCE_BETWEEN_PROBES``` or ```Z_CLEARANCE_DEPLOY_PROBE``` (both are 2mm) and on top of that the raise after probing, as it moves too fast to identify. 
All I can say is that the nozzle finally rests 5mm over the printbed, which is what I had set as ```Z_AFTER_PROBING```

boelle, 2019-02-20:
@AcHub is this still a problem when you use latest bugfix 2.0?

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-14 by thinkyhead