MarlinFirmware/Marlin #11078: STM32F407

vyckaks, 2018-06-21:
Hi 

Can someone help me with following error messages?

/Marlin/src/core/serial.h
	 expected declaration before '}' token,(40, 1)
/Marlin/src/HAL/HAL_STM32F4/HAL_timers_STM32F4.h",
	'TIM_HandleTypeDef' does not name a type",(86, 5)
         '__HAL_TIM_GET_COUNTER' was not declared in this scope,(110, 10)
         'stm32f4_timer_t {aka struct tTimerConfig}' has no member named 'handle',(110, 56)
         'stm32f4_timer_t {aka struct tTimerConfig}' has no member named 'handle',(114, 52)
/Marlin/src/HAL/HAL_STM32F4/watchdog_STM32F4.h",
	'IWDG_HandleTypeDef' does not name a type,(28, 8)
        identifier "IWDG_HandleTypeDef" is undefined,(28, 8)
/Marlin/src/HAL/HAL_STM32F4/fastio_STM32F4.h",
	 "_BV" redefined,(32, 1)

Thanks

mr-miky, 2018-06-22:
I do not think the part of the hardware watchdog has been updated for the official STM32 arduino. Initially HAL for STM32F4 was made for STM32_GENERIC

stefan85, 2018-06-27:
I get the same errors when trying to compile for STM32F4. Is there a work-around or solution for this problem? 

It makes me wonder if the STM32F4 is a good option for me to use or switch to an Atmel/NXP processor based board? At the moment it seems that the development for STM is a bit behind other platforms.

mr-miky, 2018-06-27:
Disabling hardware whatchdog ?
For the moment I'm using stm32generic but the board I'm making is designed to use the hardware PWMs that stm32duino implements. In practice I'm waiting for some things to be implemented in stm32duino but I do not see particular problems using ST32F4xx.

ktand, 2018-06-27:
Make sure *HAL_IWDG_MODULE_ENABLED* is not commented out in the
stm32f4xx_hal_conf.h file for you board variant.

Regards.
Karl

vyckaks, 2018-08-23:
HI did anyone successfully compile Marlin 2.0 for STM32F4?

mr-miky, 2018-09-22:
In the last few days I tried to compile MARLIN 2.x using official STM32duino 1.30.
On the control board I designed I use an STM32F405 and actually now the compilation is successful but then if I load the code on the microcontroller it does not work, it stops immediately after the reset.
If I compile using STM32Generic it works well.

I do not have so much desire to invest time with the debugger to understand at what point the whole thing is blocked (something already done with STM32Generic initializing systick after objects are initialized and constructed) so I ask if **anyone** has obtained a really executable firmware.
  
The thing that bothers me of STM32Generic is the management of the PWM and above all that, and I do not understand how and why, the state of the PA0 pin is altered when the PWM is activated on some output.

boelle, 2019-02-20:
@vyckaks have you tried latest bugfix 2.0 ?

mr-miky, 2019-02-20:
It's among the things I'd like to do but I'm missing the material time to do it. 

Currently I continue to use the 2.0 compiled with STM32Generic because now I have found the flaws and I have bypassed them.

vyckaks, 2019-02-20:
Hi I think for me issue was that i didn't had arduino library installed in VS Code after installation managed to compile and flash discovery board next I will try to flash this boards [STM32F407VGT6](https://www.ebay.co.uk/itm/STM32F407VGT6-ARM-Cortex-M4-32bit-MCU-Core-Development-Board-STM32F4-Discovery-/272405737459) and connect to ramps 1.6plus but that some time in the future.
Just shame that marlin will not support on-board USB .

boelle, 2019-03-13:
re-arm has onboard usb and works fine

but maybe you have something else in mind?

closed 2019-05-16 by thinkyhead

thinkyhead, 2019-08-08:
This is still an extant issue. I'm not sure we can look at `HAL_IWDG_MODULE_ENABLED` before we decide to compile that section, but maybe we can…?