MarlinFirmware/Marlin #11083: [Bugfix 2.0] Compile error

Bergerac56, 2018-06-22:
Starting yesterday, on 2 different machines I cannot compile Bugfix 2.0 anymore. (Atom, Platform IO)

The 2 Atom/platformIO were updated and are, normally, up to date (including libraries).

With plain vanilla compilation (no changes at all- Mega2560) I get:

Compiling .pioenvs\megaatmega2560\libfee\TMC2130Stepper_ID1493\source\TMC2130Stepper_CHOPCONF.cpp.o
Compiling .pioenvs\megaatmega2560\libfee\TMC2130Stepper_ID1493\source\TMC2130Stepper_COOLCONF.cpp.o
.piolibdeps\TMC2130Stepper_ID1493\src\source\TMC2130Stepper.cpp:193:10: error: prototype for 'uint32_t TMC2130Stepper::TPOWERDOWN()
' does not match any in class 'TMC2130Stepper'

It seems there is a problem with TMC2130 library

Did I miss something?

Thanks

p3p, 2018-06-22:
There is a issue with an external library, it's fixed but will take some time to propagate through platformio, as a temporary measure you can set the library version number in platformio.ini

```diff
 lib_deps          = CMSIS-LPC1768
   https://github.com/MarlinFirmware/U8glib-HAL/archive/dev.zip
-  TMC2130Stepper@>=2.2.1
+  TMC2130Stepper@=2.2.1
```

It may be best if we make the change to a specific version permanent so we can check libraries before using them, but then you don't get updates automatically...

Bergerac56, 2018-06-22:
@p3p Thanks Chris, that was it

closed 2018-06-22 by Bergerac56