MarlinFirmware/Marlin #11085: Changing the bootscreen STRING_SPLASH_LINE1 & 2

Splic3dBread, 2018-06-22:
### Description
I would like to modify the text that is displayed on the bootscreen but I cannot find the script where it is defined. Is it that simple, or is the text actually part of the bootscreen image?

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

AnHardt, 2018-06-22:
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration.h#L78-L80

Splic3dBread, 2018-06-22:
Yes that's what I want to change

AnHardt, 2018-06-22:
```
#define SHOW_BOOTSCREEN
#define STRING_SPLASH_LINE1 "My text"
#define STRING_SPLASH_LINE2 "An other text"
```

closed 2018-06-28 by thinkyhead

labotecno, 2018-07-20:
the string_splash_line 1 is scrolling  i can t see well what is mean 
how to change time to scrolling ? thank you