MarlinFirmware/Marlin #11086: z-motor running same direction

bfrakes256, 2018-06-22:
Installed BLTouch into Marlin RC8 and added the BLTouch suggested items.

Upon running this sketch, the z-motor moves the z axis in the up direction for all selections in the Marlin 3d Software application.  Proterface does the same thing.

I've reloaded a known good sketch with the physical z switch connected and everything works as it should.

Printer is a Wanhao IIIP (Monoprice).  The BLTouch appears to work correctly when I home the Z axis.  After that it only moves in one direction.  Tried inverting the z motor and then it moves only in the down direction.

I've checked both sketches line for line against each other and no issues found.  

Some magic BLTouch tweak I don't know about?

Please advise...

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

xC0000005, 2018-06-23:
What does the endstop status report?

bfrakes256, 2018-06-23:
Did not look to see. 

If I disconnect then reconnect the Z up and down works once.  After that no matter whether I do Z+ or Z- it goes down.and will eventually crash into the bed.

I am in the process of taking the known good sketch using the phsical switch and cloning it, then adding/adjusting the items in the BLTouch Manual page 2 to that sketch.

Since I am running the 1.1.8 version and page 2 of the manual show the added items being version 1.1.6, I'm not sure this is going to work.

I've attached the configuration.h and configuration_adv.h to this email.

Since it has been years since I've done any c++ programming, I'm pretty out of practice on it.

Very frustrating...

 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * //==============================
 * // Wanhao IIIP As of 06/22/2018
 * //==============================
 * 
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H
#define CONFIGURATION_H_VERSION 010107

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
[... 1632 lines cut ...]

// SkeinForge sends the wrong arc g-codes when using Arc Point as fillet procedure
//#define SF_ARC_FIX

// Support for the BariCUDA Paste Extruder
//#define BARICUDA

// Support for BlinkM/CyzRgb
//#define BLINKM

// Support for PCA9632 PWM LED driver
//#define PCA9632

/**
 * RGB LED / LED Strip Control
 *
 * Enable support for an RGB LED connected to 5V digital pins, or
 * an RGB Strip connected to MOSFETs controlled by digital pins.
 *
 * Adds the M150 command to set the LED (or LED strip) color.
 * If pins are PWM capable (e.g., 4, 5, 6, 11) then a range of
 * luminance values can be set from 0 to 255.
 * For Neopixel LED an overall brightness parameter is also available.
 *
 * *** CAUTIO

bfrakes256, 2018-06-25:
Ok, I've gotten the latest version of Marlin and created uploaded to the board.

Here are the symptoms:

I activated Safe_Homing to where it would not try to home off the bed and it now works.

Now when I use the Marlin3d application to probe the bed, it goes to the center of the bed ok and move z up then down.  The probe drops and when it gets to the bed it senses the be and retracts, z moves up then down and crashes into the bed.  I think it is now pretty close but there must be something I've missed in telling it to reverse the motor and do the second sense.

I've attached the zip with both configs...

On 6/22/2018 10:01:02 PM, Bob Frakes <bfrakes256@gmail.com> wrote:
Did not look to see. 

If I disconnect then reconnect the Z up and down works once.  After that no matter whether I do Z+ or Z- it goes down.and will eventually crash into the bed.

I am in the process of taking the known good sketch using the phsical switch and cloning it, then adding/adjusting the items in the BLTouch Manual page 2 to that sketch.

Since I am running the 1.1.8 version and page 2 of the manual show the added items being version 1.1.6, I'm not sure this is going to work.

I've attached the configuration.h and configuration_adv.h to this email.

Since it has been years since I've done any c++ programming, I'm pretty out of practice on it.

Very frustrating...

thinkyhead, 2018-06-25:
In order to send configurations you'll have to go to https://github.com/MarlinFirmware/Marlin/issues/11086 and actually include them in a reply. It doesn't work to attach them to email replies.

bfrakes256, 2018-06-25:
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/2133796/Configuration_adv.zip)

bfrakes256, 2018-06-26:
Here are the .h files.  Any suggestions appreciated

I've tried many settings on the BLTouch and when it touched the bed, I activates but does not reverse the Z motors and continues down and crashes into the bed.  This should not be this hard to do...

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/2135099/Configuration_adv.zip)

thinkyhead, 2018-06-27:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch so we're all on the same page here.

bfrakes256, 2018-06-27:
I've updated to the 1.1.x version copying the settings from an existing (working configuration.h and configuration_adv.h).
I've downloaded the bugfix-2.0.x and looked at it.  Do I need to apply all the files in the bugfix to the 1.1.x subdirectories?  Do I need to copy the settings to the configuration.h and configuration_adv.h files and move them into the Marlin subdirectory in 1.1.x?

or

is the bugfix a complete set of updated Marlin files I can use instead?

Please advise...

bfrakes256, 2018-06-27:
Working through the bugfix2.0.x now.  

FYI

Found an issues with the literal MIN_PROBE_EDGE and where my probe is mounted.  The BLTouch is mounted 35 mm to the right and 50 mm in front of the extruder.

The error checking was having an issue with calculating the front and left side probe position

I created X_MIN_PROBE_EDGE and Y_MIN_PROBE_EDGE with appropriate values.  The modified the calculation routines that use the old MIN_PROBE_EDGE with the new literals...

The code I'm guessing assume the probe is at the same location as the extruder which can never be.

thinkyhead, 2018-06-30:
> The code I'm guessing assume the probe is at the same location as the extruder which can never be.

The code uses the given probe offsets to do its work. Probe offsets of 0,0 are allowed for cases where the nozzle is used as the probe.

bfrakes256, 2018-06-30:
I have the bugfix2.0.x code integrated into my sketches.  Using the z stop, it is working in the bugfix code.
I've switched over to the BLTouch on the z stop pins and enable the BLTouch and Binlinear auto leveling and disable manual leveling.

Everything works but when the probe contacts the bed on the initial rapid touch it reverses the z motors, moves up then back down at half speed (like it should) but never reverses on the second touch an crashes into the bed.  The only way to stop the motors from stalling is to disconnect and reconnect.

I have the x, y homing working on the corners and set the middle for the safe homing.

The .h files are zipped and attached here...

There has to be something I've missed but I can't find it.  Probably some stupid mistake or oversight...

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/2152217/Configuration_adv.zip)

thinkyhead, 2018-06-30:
To test your BLTouch and endstop settings…
- Raise the Z axis to a sufficient height.
- Send `G31` to deploy the probe.
- Send `M119` to get the probe state.
- Use your finger to trigger the BLTouch probe.
- Send `M119` to get the probe state.
- Send `G32` to stow the probe.

bfrakes256, 2018-07-01:
Does not appear that G31 and G32 are valid codes.
Did m280 p0 s10 to deploy probe
M119 showed x, y, z open
bumped probe up until it started flashing 
M119 showed z still open.

m280 p0 s60 and m280 p0 s120 are also valid for the BLTouch

thinkyhead, 2018-07-01:
Sorry, `M401` / `M402` instead of `G31` / `G32`.

thinkyhead, 2018-07-01:
If the probe doesn't show up as triggered then either the wiring is incorrect or the probe has issues.

bfrakes256, 2018-07-01:
Since the light comes on when the pin is retracted tells me that the z stop pin on the ramps should go high and low when the pin is extended.

I should be able to put my volt meter in series with one of the wires going to the z stop pins and tell if the BLTouch is working, right?

In the current setting, the Z_MIN_PLUG is enabled and the Z_MIN_ENDSTOP_INVERTING is false.

Should the Z_MIN_PROBE_ENDSTOP_INVERTING be set to false?

bfrakes256, 2018-07-01:
OK, so I put my volt meter in the circuit the black and white wire off the BLTouch is connected to the Ramps board.  Raised the bed and activated the probe with M280 P0 S10 which extended the probe.  Since the Z was 23mm above the bed, I placed a block under the probe.  The voltage read 3.2 mVolt with the probe down.    I then brought z down .1mm until it retracted at 19.6mm.  The volt meter momentarily went to 1 or so mVolt and came right back up to 3.2 mVolt.   

It looks like the probe and wiring is ok.  Since the probe activates like it should the sensing wire does not appear to be telling it to reverse z and do the send 1/2 speed probe.  

Not sure where to look in the configuration to see what is missing.

kAdonis, 2018-07-02:
Afaik the Bltouch is only sending a short pulse (5ms) to show its triggered so M119 is of no use
what you see at your Voltmeter is totally normal
You z homing  speed is low (4mm/s), it might be, that your z axis is moving **too slow** for the Bltouch
try this in Configuration_adv.h  line 384:

`#define HOMING_BUMP_DIVISOR { 2, 2, 2 }`

Test carefully with your finger or a block under the probe and be prepared to stop the printer

bfrakes256, 2018-07-02:
z seems to be moving pretty good.  I thought the BLT might be defective, that is the reason for the voltmeter.

I'll do this today and see what happens.  

I'll let you know and THANKS....

kAdonis, 2018-07-02:
reading the thread again, I am not sure, understand you correctly
Is the z axis reversing at first bump or not?
Is the black and white Endstop cable connected properly?
white - Zmin
black - Gnd
which Bltouch version you have classic or smart?

bfrakes256, 2018-07-02:
Changed the setting and uploaded.  Did not change.  Way above the bed, extended the probe with m280 p0 s10.  Moved down 10 mm and triggered the probe.  It retracted then extended.  z continued down.

I have the BLT on the ramps board with the jumper to supply the voltage to the BLT.  I noticed when it is in alarm mode, the LCD dims when the light is lit on the BLT.  Voltage stays at 3.3.  Probably the coil in the BLT current draw.

bfrakes256, 2018-07-02:
Just read your reply...

Smart BLT.  The black and white is on the z_min_endstop pins

kAdonis, 2018-07-02:
> The black and white is on the z_min_endstop pins

please check the polarity

bfrakes256, 2018-07-02:
Headed for my bike ride now, I'll check it in a couple of hours and let you know...

bfrakes256, 2018-07-02:
In my Caterpillar days, we used Hall Effect sensors everywhere.  I was treating the BLT like a switch which does not care about polarity.  It was referencing ground all the time but the sensing side.  Reversed the connector and it now works.

Should I set the HOMING_BUMP_DIVISOR back to 2, 2, 4 like it was?

Thank you...

kAdonis, 2018-07-02:
glad it works!
You may leave HOMING_BUMP_DIVISOR at 2,2,2 it works for me several month now

kAdonis, 2018-07-02:
please close the issue if it is solved, and the related issues #11107 and #11180

Sandman0815, 2019-04-11:
Hallo there, i know, this issue is old but i hope, you can help me.
i've an ender5 with Marlin 1.1.9 and the Ramps1.4 Board and the A4988 steppers.
all things went right but the Extruder only Turns in one direction, if i  extrude or retract.
following things i've checkt:
The Stepper changed, The motor changed, The plug changed ( Extruder0 / Extruder1)
Marlin new installed.
The problem is every the same. Have you a nice hint for me?

bfrakes256, 2019-04-11:
Not sure...

The motors basically have two windings that control the rotor.   I would compare the wiring of a known good motor (one that turns both ways) to the one that does not.  

I had many problems with wiring on these things.  I've spent many hours with these beasts.  I've also seen that the Ramps boards can be fried very easily with open circuits.  I have a drawer full of known good spares.

You also might try pulling the connector off on a known good motor and put it on the one that does not work right.

If you have a spare controller board with ramps, your could pull the motors off and run them with the spare setup.  This would save having to fiddle with all the wiring in a tight space.

Any of the ramps boards will control any of the motors.

I'm thinking that there is probably a wiring issue.

Sandman0815, 2019-04-11:
Thanks for the fast answer, for two days the printer went with the original Melziboard. everything was fine, but the Displayport was burnt. Furthermore i've switched to the Ramps 1.4. The wirings from the motorplugs would not changed X,Y,Z goes correct, onely the E goes everytime the same direction.
As I say, i had changed the E- with the Y- Motor with the same result.
Perhaps you are right and I have to order a new Board.

bfrakes256, 2019-04-11:
My Melzboard fried very quickly.  The current draw for the bed is way too high for the board to handle.  I went with an Arduino Mega2560.  

https://www.amazon.com/gp/product/B01HEQVQAK/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1 [https://www.amazon.com/gp/product/B01HEQVQAK/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&amp;psc=1]

It has a large MOSFET that will handle the current draw.  I put a fan on it to dissipate the heat.

The here is the kit with the sheild needed for the Ramps.  Use the above MOSFET instead of the one on the shield.

https://www.amazon.com/gp/product/B0111ZSS2O/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1 [https://www.amazon.com/gp/product/B0111ZSS2O/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&amp;psc=1]

Very straight forward install.  

I ordered a spare kit just to do testing with...

Here are my Marlin settings for my printer.

Life is much easier with the Arduino setup.  Here is my custom setup for my printer and printer enclosure...

bfrakes256, 2019-04-13:
So, did you figure out your problem?

On 4/11/2019 6:13:23 AM, Bob Frakes <bfrakes256@gmail.com> wrote:
My Melzboard fried very quickly.  The current draw for the bed is way too high for the board to handle.  I went with an Arduino Mega2560.  

https://www.amazon.com/gp/product/B01HEQVQAK/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1 [https://www.amazon.com/gp/product/B01HEQVQAK/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&amp;psc=1]

It has a large MOSFET that will handle the current draw.  I put a fan on it to dissipate the heat.

The here is the kit with the sheild needed for the Ramps.  Use the above MOSFET instead of the one on the shield.

https://www.amazon.com/gp/product/B0111ZSS2O/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1 [https://www.amazon.com/gp/product/B0111ZSS2O/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&amp;psc=1]

Very straight forward install.  

I ordered a spare kit just to do testing with...

Here are my Marlin settings for my printer.

Life is much easier with the Arduino setup.  Here is my custom setup for my printer and printer enclosure...

Sandman0815, 2019-04-16:
Yes, my Rampsboard is defective. I've Changed it with one from my good friend and now it works :-)
I've flashed the same Marlin sketch an it works. I'm so happy. Thhere are many more menues now than with the Original-Board. But the Rampsboard is very Fat, so i'm thinking about an MKS, that can be installed inside the Brainbox and it goes with 12 or 24 Volts without any electrical "customize-Hack".
Nevertheless, thanks for your Help.
Sandman

bfrakes256, 2019-04-16:
No problem.  Yes, I'm not too keen on the Ramps board, but it works and once you get it working, leave it alone.  

I also added the bed leveling probe...

https://www.amazon.com/BLTouch-Leveling-Sensor-Premium-Printer/dp/B01FFV2TOS/ref=pd_rhf_ee_s_rp_0_8?_encoding=UTF8&pd_rd_i=B01FFV2TOS&pd_rd_r=14fa554e-dcaf-41fd-8c3c-42863b42d385&pd_rd_w=yGwh2&pd_rd_wg=HhKH6&pf_rd_p=f4c63947-4e8b-4aeb-a925-1d4e1a211124&pf_rd_r=ZHNB3D22WBW1QZCY5J6P&psc=1&refRID=ZHNB3D22WBW1QZCY5J6P [https://www.amazon.com/BLTouch-Leveling-Sensor-Premium-Printer/dp/B01FFV2TOS/ref=pd_rhf_ee_s_rp_0_8?_encoding=UTF8&amp;pd_rd_i=B01FFV2TOS&amp;pd_rd_r=14fa554e-dcaf-41fd-8c3c-42863b42d385&amp;pd_rd_w=yGwh2&amp;pd_rd_wg=HhKH6&amp;pf_rd_p=f4c63947-4e8b-4aeb-a925-1d4e1a211124&amp;pf_rd_r=ZHNB3D22WBW1QZCY5J6P&amp;psc=1&amp;refRID=ZHNB3D22WBW1QZCY5J6P]

It is a huge pain to configure but it is really nice to run the bed leveling before a print and not worry about the prints being thin on one side and not sticking to the bed.

Alot of fiddling with mounting and configuring Marlin.  Took be about two weeks to get it working properly.  It comes with "instructions" (so called).

This probe is the only one that works with the glass bed.  The ultrasound and megnetics do not work on the glass bed.

Sandman0815, 2019-04-16:
Yes, you are right. I've installed the cheap-Copy of BL-Touch
[image]

and switched the Bilinear-Leveling on. That works absolute Fine ant Printing is very Funny now.
I hope it is not so difficould to switch to the MKS-Board later.

bfrakes256, 2019-04-16:
So are you in Germany or the Netherlands?   

Very similar languages...

[Bettleveling] [https://user-images.githubusercontent.com/23085485/56241827-1e86f500-6097-11e9-91b3-6d88124073b4.jpg]

bfrakes256, 2019-04-16:
So, where are you getting the graph?  I was using a program called the Marlin 3d Printer Tool.  It stopped working on my Windows 10 computer.

On 4/16/2019 6:24:04 PM, Bob Frakes <bfrakes256@gmail.com> wrote:
So are you in Germany or the Netherlands?   

Very similar languages...

Sandman0815, 2019-04-17:
Hi there, I'm from Germany. The graph comes from my Raspberry Pi, i'm using Octoprint as a Printserver.
And I have a nice Plugin, that makes this wonderfull graph. Bedleveling is very funny, since I've this fine Tool.

Greetings from Germany
Sandman