MarlinFirmware/Marlin #11087: UBL over and under compensating

ramireseverton, 2018-06-23:
Have been facing this issues since the first versions of ubl, have tried in a variet of machines now. Every time I probe the mesh, it gets too low on one side and too high on the other side.
The sequence I use:
G28
G29 p1
G29 p3 ( multiple times to fill the mesh)
G29 s0
G29 a
M500

I can handle this by changing values poit by point with m421, but this basicly invalidates AUTO leveling.

AcHub, 2018-06-23:
It reads like you are doing this sequence every time you do a print to level your bed - are you? Because this isn't necessary.
Do this once:
G28
G29 P1
G29 P3 ( multiple times to fill the mesh)
G29 S0
G29 A
M500

If your mesh needs fine tuning use M421 to adjust the points which need tuning, or use the G29 P4 command to fine tune (see this for all parameters: http://marlinfw.org/docs/gcode/G029-ubl.html). 
I find G29 P4 much easier to use: Usually I print a grid pattern to identify areas where the nozzle is too close/far away, and write down the X and Y position of those areas. Then I use G29 P4 Xxxx Yxxx H0.2 and a 0.2mm shim to finetune the meshpoint. If the area you need to tune is larger than just one point, you can add the R parameter which will probe additional points around the X/Y position in one sequence. After you are done with finetuning save again with G29 S0 and M500. Now you have a topographic model of your printbed, which normally wouldn't change (unless you deform it with raw force).
Before a print you now only need to validate that your printbed is aligned with your model, which you can do automatically with the G29 J command. You can add this to your startup gcode:
G28 ; Home all
G420 L0 S1; Load Mesh stored at slot 0 and activate bed leveling
G29 J ; calculate a plane and adjust the existing mesh to the bed tilt.
And you are ready to print.

By the way, you can export your mesh to gcode with G29 S-0 once you are satisfyed with it, so you can restore it from a gcode file at any time if you ever lose your stored mesh

gloomyandy, 2018-06-23:
Before you go too far down any investigation what version of Marlin are you using? There have been a lot of changes/bugs fixed in the latest bugfix release. In particular there were problems with using G29 J in previous releases.

Roxy-3D, 2018-06-23:
The G26 pattern is the one to trust.   The whole point of probing the mesh and auto filling it is to get to a point where you can use G26 to see what the printer really will do.

If you can print a G26 pattern...   Don't probe any more.    Use that pattern to edit mesh points so everything is perfect.

One problem we have been seeing with all the bed leveling systems is the X Axis will have some twist to it.   So the probe is more or less under the nozzle depending where you are measuring on the bed.   What that means is in that case the probed values are not 'real'.    But...   If they are good enough to do a G26, you don't worry about it.   You just use G26 to correct the mesh.

forkoz, 2018-06-23:
So the J isn't needed? I haven't needed to manually edit any points like op. The inconsistency is troubling though since the bed leveling is different every time its probed. Maybe something is up with the x axis but its 2020 + 3 wheels so I don't see it twisting easily.

gloomyandy, 2018-06-23:
You really shouldn't be getting different mesh results each time you run it (even with a twisted bed). How much do they differ? If your probe is not delivering consistent results then how can you trust it to correctly set the initial Z height as part of the G28 home operation (assuming you are using a probe to home Z). Without that being right you will not get very good prints no matter how good your mesh is (unless you are are going to manually adjust/babystep things every print). 

What sort of results do you get with a probe test M48. 

I used to use an inductive probe (18mm one) that gave inconsistent results, running M48 would give a lot of identical values but every now and again I'd get one that was 0.1mm+ different. This used to drive me mad as I would end up with the Z offset being wrong by this amount from time to time. In the end I replaced the probe with a 12mm one (the smaller probes sense at a closer distance and seem to offer better accuracy/repeat-ability). Now I get results that are always within +-0.01mm, usually better.

forkoz, 2018-06-23:
I posted the pastebin link as to what I get. The probe accuracy test gets like .0005. The homing is always consistent just the UBL meshes are not.. When I visualize them they do look similar just slightly different values, hence I dunno wtf is going on. Not much problems with the prints either, esp not how it was when I started and didn't know what I was doing.

gloomyandy, 2018-06-23:
That is very odd, the variation in your mesh is pretty big. What location are you homing at? What sort of probe are you using? Are you sure you are getting a consistent Z offset? Do a G28 then lower the nozzle until it touches the bed (or a slip of paper), is the Z reading the same if you do this say 10 times?

thinkyhead, 2018-06-28:
> its 2020 + 3 wheels so I don't see it twisting easily.

These setups actually exhibit this issue a lot. It's a regular epidemic, and this issue is a duplicate. Search for "twisted gantry" in the Issue Queue to see several, and to see how other users have solved it.

closed 2018-11-07 by thinkyhead