MarlinFirmware/Marlin #11088: Edit UBL mesh Without an LCD

ramireseverton, 2018-06-23:
UBL is great, but beeing able to run the whole process including assisted fine tuning over serial com would be very useful for hosts like octropi. We have a touch display, no need to a second oldschool one!

Roxy-3D, 2018-06-23:
You can also edit mesh points with M421.    The syntax varies depending on what bed leveling system you are using.

thinkyhead, 2018-06-28:
> run the whole process including assisted fine tuning over serial com 

Pretty complex, and basically would need a companion OctoPrint plugin that mates with the feature.

Roxy-3D, 2018-06-28:
> being able to run the whole process including assisted fine tuning over serial com would be very useful or hosts like octopi. 

If somebody is willing to code up the client side of things...   It would be straight forward to put the control elsewhere.   The client could do a G29 W to find out all the details of the mesh.   (The mesh points, how many there are, the distance between them, etc.)    And then a G29 T will give the actual values.

With a touch display you could outline an area and then pull it up or push it down.  The client could edit the values by sending the appropriate M421 commands to the various mesh points within an area.

But for now...   The LCD Panel works to edit the mesh.

sjasonsmith, 2020-09-11:
@GypsyRobot I believe this capable already exists. According to the [Marlin documentation for UBL](https://marlinfw.org/docs/gcode/G029-ubl.html) this can be done with a combination of G42 and M421 commands.

[image]

closed 2020-09-11 by sjasonsmith