MarlinFirmware/Marlin #11089: Filament runout sensor Without LCD

ramireseverton, 2018-06-23:
A lot of people runs octopi or other hosts that doesnt need the old fashioned lcd at all. We would be very glad to be able to handle advanced pause, filament runout and m600 completely over serial communication.

thinkyhead, 2018-06-28:
What kinds of sanity-check errors is Marlin currently throwing when you try to enable the features you want to use, without an LCD controller? (Please test with the current bugfix branches.)

Features that simply need to be able to continue after user confirmation may be able to use `EMERGENCY_PARSER` and the `M108` command where an LCD is otherwise used.

boelle, 2019-03-13:
@GypsyRobot what @thinkyhead said ^^^^

closed 2019-05-16 by thinkyhead